sinara-hw/meta #1: swap EEM connectors on two-port devices

jordens, 2018-06-04:
Due to the ribbon cable topology, the EEM connectors on a device should be right to left while keeping them on Kasli as left to right. Not doing so merely results in unnatural/unintuitive EEM slot numbers being used.

gkasprow, 2018-06-04:
this can be done on Kasli - only silk screen modification would be needed. On existing Kasli boards one can place stickers with slot numbering.

jordens, 2018-06-04:
I don't think that's better. Then the boards in the crate would be numbered/inserted from right to left. IMHO the Kasli numbering is correct. I'd just keep this as a "best practice" for future device designs and revisions.

hartytp, 2018-08-15:
AFAICT, this is now done. Thanks Greg!

hartytp, 2018-08-15:
(Would close this, but I don't have appropriate permissions for this project. @gkasprow can you either change my permissions or close this?)

jordens, 2018-08-15:
I'll leave the permission management to @gkasprow.

closed 2018-08-15 by jordens