sinara-hw/meta #2: [RFC] use EMC shielded front panels for EEMs

jordens, 2018-06-06:
_From @jordens on May 24, 2018 18:34_

They are a about 6-8 € each (including gaskets) and would help when putting EEMs into shielded crates (~130 €). Don't know about secondary effects on the assembly costs though. Needs double checking that the lip doesn't collide with the RJ45/SMA connectors.

_Copied from original issue: sinara-hw/sinara#549_

jordens, 2018-06-06:
_From @gkasprow on May 24, 2018 18:42_

Such panels from Schroff are already foreseen in the design so all should match.
Since we do not plan to change holes and cut-outs, we can finally switch to them.
Schroff has quite expensive tooling, but later on the panels are not that extensive, roughly 20EUR with machining and screen printing at quantities of 20+

jordens, 2018-06-06:
Pretty sure that the left lip (underside) will touch/collide with the SMAs on Urukul.

jordens, 2018-06-06:
_From @gkasprow on May 28, 2018 11:53_

They should fit - of course the THT pins will need to be trimmed.
[image]

jordens, 2018-06-06:
That's one solution.
Also that washer under the screw doesn't exist on the boards I have. Might not be needed but it affects clearance.

hartytp, 2018-11-17:
This is now done and the new FPs work nicely AFAICT.

closed 2018-11-17 by hartytp