sinara-hw/meta #3: EEM front panels washer cutout.

jordens, 2018-06-06:
_From @jordens on May 15, 2018 12:46_

The cutouts that receive the rectangular snap-in grey plastic "captive" nipples for the panel mounting screws were too narrow on the older boards. The washer was squeezing the screw and not allowing it to move freely.
More recent ones seem to be fixed. Has that fix been applied to all boards?

_Copied from original issue: sinara-hw/sinara#547_

jordens, 2018-06-06:
_From @gkasprow on May 22, 2018 21:39_

Did you observe it only with first revision or later as well?

jordens, 2018-06-06:
The cutout was too small for all board I received until a month ago. In the last batch there were still some panels with bad cutouts but those may have been from older runs. I just wanted to make sure all panels are updated.
As a sidenote: receiving 8HP panels with 4HP handles (like for the last Kasli batch) is disappointing.

hartytp, 2018-11-17:
@gkasprow I think this has been fixed on all boards now. Please confirm and close.

hartytp, 2019-02-20:
@gkasprow ping...is this now fixed on all designs?

gkasprow, 2019-02-20:
I think so.

closed 2019-02-20 by gkasprow