sinara-hw/meta #4: isolated SMA assembly

jordens, 2018-06-08:
Right now the angled SMAs get the metal teeth washer, then a plastic one, then the panel, then another plastic one and then the screw. AFAICT in many cases the metal+plastic washer stack on the inside of the panel is too think resulting in a slightly angled panel and the remaining thread being too short.
I'd either (a) drop the metal+plastic washer on the inside or (b) just use the plastic washer on the inside and ensure that the gap it fits in is not too large.

hartytp, 2018-11-17:
@gkasprow has something been done about this? If so, please can you close this issue?

gkasprow, 2018-11-17:
We cannot do much towith BOM. The SMA connectors are often not available so we have to use similar ones. And they differ slightly in mechanical details. So we must make sure that additional washers between SMA and panels are installed to not bend the panel

hartytp, 2018-11-17:
So, is "add suitable washers" something we need to add to the schematics annotations/assembly instructions? How do we ensure we don't have problems with this in the future?

gkasprow, 2018-11-17:
Just make sure that quality control is performed correctly. And notify the board producer about the issue.

jordens, 2018-11-18:
I don't have the hardware in front of me right now but I thing one issue was also that the washers collide with the PCB on urukul and are therefore angled. That will also bend the panel and might be something to fix in the layout. I'll check next time I have one on my hands.

jordens, 2018-11-18:
And I don't think that having the board houses isolate themselves from these issues is a good solution. Similarly I don't like that they do their own unknown, secret, undocumented, and untracked testing instead of contributing to and using the a available test suites.

closed 2020-02-18 by hartytp