sinara-hw/meta #5: [RFC] new EEM: humpback

jordens, 2018-06-21:
Despite the general purpose of the Sinara ecosystem hardware and the wide range of features, lot of development in labs is still based on random "toy"/"development"/"evaluation"/"hobbyist" boards with µCs and FPGAs. They are dirt cheap, fit for the many applications that lie outside the typical use cases for Sinara EEMs, get the job done just fine, and have a shallow learning curve suitable for physicists. That won't change and there seems no point in trying to replace those "boards". Instead let's try to integrate them.

I'd like to propose a simple modification to the [TTL tester EEM](https://github.com/sinara-hw/sinara/raw/master/ARTIQ_ALTIUM/EEMs/Tester/3U_TTL_Tester.PDF) to give it a much wider applicability and open it up for more use cases.

The basic idea is to

1. get connectivity between boards and the Sinara EEM connector
2. build a easy way of mounting those boards in a crate
3. to supply them with power.

There seem to be two approaches:

A. Build a "shield" that just implement the connectivity aspect (1).
  * RJ45 to be used outside the crate
  * Should use the same LVDS-to-CMOS converters and DIP switches as Tester
  * Should be powered from the board.

B. Build a EEM that achieves (1), (2), and (3).
  * Call it e.g. "Humpback"
  * Mount the board on the EEM
  * Since there will likely be the need for external connectivity, while the panel pattern for that connectivity is unknown, let's open up the panel and also build a gap into the board so that only a panel frame (or even just a panel-U) remains.
  * Make the EEM short so that small boards can be used and large boards just stick out of the crate.
  * One EEM connector
  * Power the board from the EEM
  * Maybe add a barrel connector so that the EEM can power the MP3V3, the board and an attached EEM (like Tester).
  * Maybe a 5V SMPS for VIN to feed the more [sensitive](https://www.arduino.cc/en/ArduinoCertified/IntelGalileo) boards.
  * Use the same LVDS-to-CMOS converters and dip switches as Tester. Don't bother with board-controlled LVDS direction signals. Maybe add an I2C direction shift register to achieve that.
  * Expose I2C to board.
  * Maybe expose MP3V3 to board.

As choice for the connector layout (power, i2c, digital), probably using a few pins in a single row of 100mil with the "Arduino Uno v3" pattern for the shield and the EEM plus a common denominator of mounting holes and standoffs for the EEM would cover the largest subset of boards. Position and orientation unclear. Also, these are typically stacked and EEM would probably the "top" of the stack while the board is the "bottom".

A small collection of boards trying to not discriminate any:

µC/CPU:
* [Arduino](https://store.arduino.cc/arduino-due)
* [Intel Galileo](https://www.arduino.cc/en/ArduinoCertified/IntelGalileoGen2)
* [ST Nucleo 144](https://www.digikey.com/product-detail/en/stmicroelectronics/NUCLEO-F429ZI/497-16280-ND/5806777), [ST Nucleo 64](https://www.digikey.com/product-detail/en/stmicroelectronics/NUCLEO-F446RE/497-15882-ND/5347712)

FPGA:
* [Arty](https://www.xilinx.com/products/boards-and-kits/arty.html#hardware)
* [XLR8](http://www.aloriumtech.com/xlr8-fpga-board/)
* [Papilio (Pro)](http://papilio.cc/index.php?n=Papilio.PapilioPro)

marmeladapk, 2018-06-21:
Just a quick note: Intel Galileo (Edison and Joule also for that matter) has been [EOLed](https://qdms.intel.com/dm/i.aspx/F79ED317-6EF3-4293-8752-9C4E7B3C11C5/PCN115581-00.pdf).

gkasprow, 2018-06-21:
There is new Arduino with FPGA, so this could be interesting idea.
If we want to keep the cost down, maybe ICE40 FPGA family could be an option? It is very low cost (~1$) and has completely open source toolchain.
Tiny FPGA between uPC and EEM could be benefit and help build relations between Arduino and EEM. ICE40 would replace LVDS drivers. FPGA is loaded from FLASH or processor.

sbourdeauducq, 2018-06-21:
> f we want to keep the cost down, maybe ICE40 FPGA family could be an option? It is very low cost (~1$) and has completely open source toolchain.
> Tiny FPGA between uPC and EEM could be benefit and help build relations between Arduino and EEM. ICE40 would replace LVDS drivers.

Sounds good.

sbourdeauducq, 2018-06-21:
If we have a FPGA flash, the boards can come shipped with a bitstream that turns the FPGA into a dumb LVDS/TTL converter.

gkasprow, 2018-06-21:
Exactly, and we would cut the cost. Arduino has nice feature - standarisaton of the connector. So we can plug variety of clones even with Linux machines on it. We can make double Arduino/RPI footprint.

gkasprow, 2018-06-21:
There are ICE40 shields for RPI with tools that configure the FPGA from Linux, this could be powerful tool.

dtcallcock, 2018-06-21:
Would this be a way of creating an Ethernet carrier, as discussed in [#473](https://github.com/sinara-hw/sinara/issues/473)?

gkasprow, 2018-06-21:
This is my understanding of this task.
Recently I [use](https://github.com/gkasprow/Sensor-Box/wiki/SensorBox-Hub#linux-hub) tiny Orange Pi board. It has essentially all what is needed, quad core CPU and several IOs.
The cost is 8$.
But RPI has better support and probably would be optimal choice for such purpose.
So ability to plug either Arduino Leonardo (with Ethernet) or RPI would satisfy most use cases.

gkasprow, 2018-06-21:
ICE40 FPGAs has nice features : hardened SPI and I2C cores :)

gkasprow, 2018-06-22:
@sbourdeauducq I didn't use them, can be useful in case of lack of resources when somebody desperately needs to get some logic cells:)

jordens, 2018-07-18:
@gkasprow For Humpback (but not Shield) that makes a lot of sense to me. The added cost of one more IDC header seems ok. And most µC/FPGA mezzanine boards seem to be able to handle 16 IO. Then for the FPGA on Humpback, 2*1 I2C pairs and 2*8 LVDS pairs, plus 2*8+2 CMOS IO to the mezzanine? That would be 54 pins. Are we talking something like the ICE40HX1K-TQ144?

gkasprow, 2018-07-19:
@jordens The ICE40 family has LVDS lines only in one bank.
With TQFP140 package you can connect only one EEM, BGA256 enables 3 EEMs.
I want to use Beaglebone as default with corresponding panel cutouts.

gkasprow, 2018-07-19:
Some sketch
[image]

gkasprow, 2018-07-19:
why beaglebone?
- has real Ethernet with low latency
- has fast IOs with PRU
- has external bus interface (GPMC) which gives fast IO channel to the FPGA

Of course place for Arduino Leonardo Ethernet will also be available

jordens, 2018-07-19:
@gkasprow Ack the ICE40 BGA256 and the three EEMs.

The Beagle bone is fine. But I'd like to support a simpler/cheaper µC board as well. The BB needs male headers on Humpback. It looks like we can add female headers for the [ST Nucleo 64 boards](https://www.st.com/content/ccc/resource/technical/document/user_manual/98/2e/fa/4b/e0/82/43/b7/DM00105823.pdf/files/DM00105823.pdf/jcr:content/translations/en.DM00105823.pdf#page=15). And I think they will not collide with the beaglebone. They are wider apart and the underside of the nucleo is empty.

Why Nucelo 64/STM32:
* STM32F4 are one of the dominating embedded chips with Rust support with the most active commity.
* Are used in labs, even have flight herritage.
* Are really cheap and have a wide range of chips/boards readily and easily available.
* Don't collide mechanically with the Beaglebone.
* Have boards with [Ethernet](https://www.st.com/content/ccc/resource/technical/document/user_manual/group0/26/49/90/2e/33/0d/4a/da/DM00244518/files/DM00244518.pdf/jcr:content/translations/en.DM00244518.pdf)

gkasprow, 2018-07-19:
OK, so maybe let's put Ethernet and USB couplers on the panel and provide several footprints on the EEM. 
In this way the panel is fixed and we can install Arduino, BB, STM32, etc.

jordens, 2018-07-19:
A couple open questions:

* Add DIP switches to the FPGA to configure the LVDS direction or is that only done by bitstream/I2C?
* Leave the panel cutout wide open so that different boards can be accessed through it? I think there are two problems with matching ethernet+usb cutouts in the panel. They are not mechanically compatible between boards and they are a mechanical constraint that conflicts with the pin headers (see Allaki). I'd rather solve that like FMC (or the next iteration of Allaki) with a wide cutout and a panel on the mezzanine/board.
* Maybe ship the pin headers for the different boards unsoldered so that the user can choose and solder the ones he wants.

gkasprow, 2018-07-19:
DIP switches will be routed to the FPGA so can be used for any purpose. Defualt configuration will provide LVTTL-> LVDS buffers.
I'd use low cost [Etherne](https://pl.aliexpress.com/item/20x-Cat5e-RJ45-Connector-Plug-Socket-RJ45-Network-Cable-Blank-Panel-Patch-DIY-RJ45-Extender-Plug/32805676437.html?spm=a2g17.10010108.1000016.1.29a07c19jVeueb&isOrigTitle=true)t and USB couplers on the panel. We can also place the USB and RJ45 connectors on the PCB and make such coupler ourselves.

hartytp, 2018-07-19:
> I'd use low cost Ethernet and USB couplers on the panel. We can also place the USB and RJ45 connectors on the PCB and make such couplers ourselves.

I'm never a fan of having USB/ethernet feedthroughs on the FP + internal cables. They tend to be bulky and annoying and the cables get caught on things. Plus, it's more parts that need to be sourced.

If we're thinking of going down the route of having the customer solder on pin header and supply the microprocessor board, then why not have front panels sold separately to the PCB? That way the user buys the right panel for their microprocessor board and installs it when fitting the microprocessor module.

jordens, 2018-07-19:
Yes. There is the pigtail mess with inflexible but short RJ45 cables and the added parts. Thinking about it, I'd also prefer either a big empty cutout in the panel large enough to accommodate the target boards or different panels for different boards.

jordens, 2018-07-19:
And I suspect this will need to be 6 HP or 8 HP, certainly for the Nucleo, but looking at the rendering also the BB.

jordens, 2018-09-21:
After the Banker design a few additional notes and opinions.

* Let's do 16 dip switches. To the fpga.
* Let's keep it simple and have the IOs all 3.3 V only. No need for translation or drivers.
* 3 EEM like Banker.
* I'd like a big cutout or maybe even a deep wide slot in the front panel. So that the remaining Frontpanel is just a U shape. So that the boards can get a matching rectangular (like FMC/allaki) or even T shape panel insert. And 4HP. The front panel on the boards can then be T shape and extend the width to 6/8/12 HP.
* A common set of mounting holes for the boards.
* Maybe consider mounting a solderless breadboard on Humpback so that there is prototyping flexibility. There are small ones around with mounting holes.
* I can't come up with a common set of header placement for the boards. This is difficult. Like the mounting holes.

gkasprow, 2018-11-18:
What about using 3D printed panel inserts? I will make and releases stl files so everybody will be able to print them at home. 3D printers are so popular these days. For 200$ one can get one with sufficient print quality. We have such one in the lab. Recently I bought Prusa MK3 for home use and it is decent print quality printer for less than 1k EUR.
Why 3D printed panels? Various board have different distance between connectors and board edge. Some have 5V supply connectors that it's safer to cover entirely. Some have mini USB that require very thin panel. 3D printed one would solve all the issues easily. Due to small area print time / cost would be low.