sinara-hw/meta #6: [RFC] LCD panel and user interface

gkasprow, 2018-07-25:
At the WUT we are working on  equipment for various applications and an idea of standardisation came to our mind recently.
Some of them includes multichannel HV and precise DC supplies, RF sources to count a few. ARTIQ and EEM approach is fine since we use 3U anyway. The only lacking piece is user interface.
The equipment needs to have local keyboard and some display.
Since we will build humpback which will be low cost alternative to Kasli for non time-critical applications, there is an idea to attach to the Beaglebone or other SBC the LCD display and simple keyboard in form of 3U front panel. In this way one can build laboratory-grade equipment that can be used without external computer as an user interface. For example for optogenetic experiment we need a few DDS channels running at 27MHz, a few [RFID receivers](https://github.com/flisadam/RFID_system/wiki), several TTL channels and it needs to be operated by biologists  which means it have to have simple user interface.
So the idea is to attach low cost touch panel 7" LCD (~20$) to the SBC and run some standard GUI. And of course provide nice looking panel that could be mounted on any side of the 3U chassis.
Maybe ARTIQ could do the job? Do you @jordens  think that it would work?

jordens, 2018-07-25:
I think an LCD is a fine idea for Humpback+SBC from a hardware perspective!
If we don't use an FPGA (Kasli/Sayma/KC705) to control the EEMs but something else, the interface and the abstraction needs to be reimplemented for that platform. But that's something that will need to be done anyway for Humpback.
Software-wise, the ARTIQ master, controller, controller manager, worker etc might be a bit heavy for a small SBC. And GUI tends to work best on large screens with plenty of real estate to arrange the docks.
I guess it depends mainly on the kind of and abstraction/programmability level you need to provide to the biologists.
Another idea would be WiFi on the SBC/µC/Humpback and ARTIQ on a tablet.

dhslichter, 2018-07-26:
I second the notion that the current ARTIQ GUI is not friendly to small screens.  What one could potentially do would be to write very simple GUIs for testing/debugging/simple applications the individual boards, with a small defined subset of parameters (e.g. for Urukul, just freq/phase/amplitude control and on/off; for Zotino, 32 voltages to set).  To be honest, though, I feel like Humpback is designed to allow user control via their own funky SBC, so the user needs to be in charge of the GUI stuff as well, not us.  Making provision for a screen to be mounted nicely would be good to do, but I think it makes sense to leave this to the user.  If you want to use ARTIQ, we support that with ARTIQ; if you want your own funky board to control it, then you are probably the kind of person who wants to roll your own software/GUI too.  We can just make it physically easy for you to put a screen on.

gkasprow, 2018-07-26:
Guys in my lab started playing with STM32 graphical library and it seems it's the way to go. There are powerful STM32 + LCD devkits for roughly 50$. We will add 2 rotary encoders + maybe a few buttons, but touch screen seems to be standard interface these days.

gkasprow, 2018-11-19:
There are solutions foreseen for LCD/OLED and GUI.
For Beaglebone, one will use low cost HDMI display with USB touch panel
For STM32, low cost STM+LCD HMI modules are available, and can be attached to the Humpback using IDC cable.

closed 2018-11-19 by gkasprow