sinara-hw/meta #7: [RFC] DC current meter for Zotino

gkasprow, 2018-08-05:
I visited quantum physics lab in Torun (Poland). Guys make ion-trap experiments where they additionally  treat Ca ions with electron beam.
They are interested in Sinara/Artiq ecosystem, and would like to have some additional features.
They supply trap electrodes with RF + DC and need to measure DC component (current from electron beam) without interfering with RF. So such current meter needs to be matched to the RF signal. The range is <<1uA. It means it probably needs to be placed very close to the target and could be made as extension board to Zotino where DC is added to RF and fed to the vacuum chamber. They need more DC than Zotino can deliver (30V), so addition amp would be used as well.
And the question is if the community would have a use case for such current measurement extension.

hartytp, 2018-08-05:
@gkasprow how many channels do they need? Sounds like this might be better off being a new design with DACs, current meter and everything else on a small PCB near the RF matching.

Depending on what BW/latency they need for the measurement, it might be a good PoE design. Otherwise, LVDS over RJ45 could be the way to go.

Anyway, this isn't something we'd use, but there are other experiments with similar requirements.

gkasprow, 2018-08-05:
They would need 12..16 channels. Since the use case would be limited, I want to reuse existing designs and add only small and simple module that connects Zotino, Urukul and Sampler and do the job.

hartytp, 2018-08-05:
ack

gkasprow, 2018-11-19:
no funding and no enough attention so closing

closed 2018-11-19 by gkasprow