sinara-hw/meta #8: QC issues

hartytp, 2018-08-15:
We had a few soldering / qc issues with the previous batch of EEMs. e.g. https://github.com/sinara-hw/Zotino/issues/3#issuecomment-413104662

@gkasprow what's the plan for resolving them going forwards? Please close this issue once you're satisfied that these issues will be resolved in future production rounds.

gkasprow, 2018-08-16:
Technosystem claim they had series of connectors with poor lead wetting. But this still should be detected by the QC anyway. AOI is performed only on SMT assembly. We have to force them to do inspection on THT as well.

jordens, 2018-08-22:
Another one. Who can spot the stray resistor? It's semi-soldered. Shouldn't AOI see that?
This is DIO_BNC/v1.2, TS 36/18
[image]

hartytp, 2018-11-17:
@gkasprow closing this as there doesn't seem to be any outstanding action here. We should continue to discuss these QC issues with TechnoSystem if we're going to keep using them as a supplier.

closed 2018-11-17 by hartytp

gkasprow, 2018-11-17:
In case of QC issues, notify the vendor directly. They are usually not following these threads.