sinara-hw/meta #11: [RFC] new EEM: zapper

dtcallcock, 2018-08-18:
High voltage supplies are commonly required in ion trap experiments.  These are widely commercially available but typical models have many of the following drawbacks:

- Noisy output
- Radiated RFI from the box
- Expensive
- Poor stability
- Low channel density
- High power dissipation when voltage output static
- Unnecessarily high current capability (safety issue as well as being the root cause of most of the above)
- Flaky USB connection with high latency

# Use cases

### Trap electrode voltages

Zotino only provides ±10V.  This is generally fine for small high-electrode-density surface traps.  However many groups operate larger surface traps or macroscoptic traps of the 'blade' or 'wafer' variety.  These typically need much higher voltages, from a few 10s of Volts up to a few hundred Volts.  Requirements are basically the same as for Zotino but scaled to higher voltages.  The [thesis](https://www2.physics.ox.ac.uk/sites/default/files/2011-08-15/cjbthesisfinal_pdf_33239.pdf) of Chris Ballance has a good discussion of the requirements and a circuit design that could be the basis of this EEM.  This type of trap generally has less electrodes so a <32-ch solution may be acceptable.  

### Slow piezos

We have several piezo-actuated mirrors on our experiment, such as the Thorlabs [Polaris K1S2P](https://www.thorlabs.com/newgrouppage9.cfm?objectgroup_ID=5035).  These need 0-150V and have a 0.35uF of capacitance.  The Thorlabs [MDT694B/MDT693B](https://www.thorlabs.com/newgrouppage9.cfm?objectgroup_id=1191) are ubiquitous in our lab but suffer many of the problems described above.  For this use case, the new [PiezoDrivePD32](https://www.piezodrive.com/product/pd32-multi-channel-piezo-driver/) may be a good commercial alternative when coupled with a Zotino.  I haven't tested one out but we have been happy with other products from PiezoDrive. 

# Implementation

### DAC

16-bit resolution.  

A few options:

- Use Zotino.  The card would sit next to Zotino and be connected to the IDC headers on it via a ribbon cable(s).  Lower risk, less development work, and the Artiq driver would be pretty much for free.  Frees up board real estate as no EEM connector etc. would be needed.

- Use an onboard DAC.  The 32-ch AD5372 in Zotino has 8-ch ([AD5362](http://www.analog.com/en/products/ad5362.html)) and 16-ch ([AD5360](http://www.analog.com/en/products/ad5360.html)) equivalents if that is likely to reduce risk/effort.  At 4-ch and 8-ch, more accurite, lower noise DACs are available.

### Channel Count

This will probably depend a lot on the power consumption, size, and cost of the output stage we decide on.  8 would be a reasonable starting point.

### Ranges

Traps generally require a mix of high positive voltages for endcaps (becasue our ions are positively charged) and lower bipolar voltages for other electrodes.  There seems to be 2 options for dealing with this:

- Per board range selection.  The board could be configured as either single ended or bipolar.  If different ranges are required, you just use two boards.  The cheaper the board is and the lower the number of the channels, the less of an issue this is.

- Per channel range selection.  Board has three supply rails (+V1, +V2 and -V2).  Each channel can be changed to run at 0 to +V1 or -V2 to V2 using jumpers or component selection.

### Voltages

It would be good to hear from potential users here.  As a starting point, how about:

0 to +V1 or -V2 to +V2 where V1= 200, 100, or 50V and V2 = 100, 50 or 25V

Historically some traps have used >1kV but my guess is that there isn't a sensible design that covers that and 10s of volts.

### Bandwidth

Zotino has 75kHz analog bandwidth.  This sounds like a reasonable starting point.

There should be the option to install components that drastically limit this to say, 100s of Hz.

### Stability

Zotino output stage has "Worst-case OpAmp self-heating is about 25C. At 0.2ppm/C, this is 4ppm, which is fine!"  This might be harder to achieve at high voltages but <10ppm would be nice.

### Connectors

- Individual Front Panel coax connectors - Often slightly less common connectors are used on HV supplies to prevent accidentaly connecting the wrong thing and destroying it.  I have seen SMB, SMC, LEMO OO and LEMO 0B used.

- High Density Connector - D-sub etc. 

- Both - Same idea as Zotino, where the high density connector can be installed or a ribbon cable run to a panel with individual connectors on it (ie. a [BNC-IDC](https://github.com/sinara-hw/BNC_IDC/)).  We could make a version of BNC-IDC with SMB etc. instead.

hartytp, 2018-08-20:
This would be greaT. Those yhorlabs boxes are also quite buggy firmware wise.

Nb Chris’s design uses rectified mains so is quite bulky. Iirc the jqi design based on laser printer ics with post regulation would be the way to go here.

Wonder if this could be a poe design with its own dac?

dtcallcock, 2018-08-20:
> Those yhorlabs boxes are also quite buggy firmware wise.

As far as I can tell, early Thorlabs USB stuff was shipped with the firmware/software still in alpha.  It looks like they are phoning it in a bit less these days but I remain to be convinced.

> Wonder if this could be a poe design with its own dac?

For trap voltages an EEM connection would be preferable as latency can be a big problem for things you want to change every few shots of an experiment (like a shim voltage).

I'm not against it having its own DAC so that it can be easily 'humpbacked' into a standalone POE device though.

dtcallcock, 2018-08-20:
> Nb Chris’s design uses rectified mains so is quite bulky. Iirc the jqi design based on laser printer ics with post regulation would be the way to go here.

It looks like the two circuits have fairly comparable noise and tempco specs so either would work.  A quick look at the board files for JQI design makes me think we'd struggle to squeeze more than 3-4ch onto a Eurocard (stripped it down to just the DAC and HV section). 

PiezoDrive also make a really nice 3-ch driver [PDu150](https://www.piezodrive.com/modules/pdu150/).  It's only $120/ch and and you could get 9 channels on a board.  On the downside you are limited to only one range (-30 to 150V) and we would have to retrofit a different heatsink solution if we wanted to avoid a 12HP board.  They also do custom OEM work so once we have a spec it might be worth asking them for a quote.

dhslichter, 2018-08-22:
My recollection is that the cheaper PiezoDrive boards used a [DRV2700](http://www.ti.com/product/drv2700), a bit noisy but a simple way to not have to deal with rolling your own high voltage rails and amplifiers...

gkasprow, 2018-08-23:
Long time ago I built piezo driver for AFM. The idea was similar to this [approach](https://www.electronicdesign.com/analog/specialized-circuit-drives-150-v-piezoelectric-motor-using-low-voltage-op-amp) but the output stage was fully symmetrical (bipolar). Do we care if piezo is driven in differential or single-ended way? Are there some cases where actuator must be grounded?
Single ended drive simplifies the circuit and enables doubling number of channels. But power supply needs to be symmetrical. If we want to use same supply for all channels than it's fine.

About HV supplies. Is good idea to have them as separate module or integrated with piezo driver?
Separate design gives ability to use it as HV source and more flexibility in connection with piezo driver. But is more error-prone. 

I've built series of galvanic isolated (10kV) low noise (fully resonant architecture) HV supplies for GEM detectors, but ranges were different (500 to 1.5kV). It can produce positive or negative voltage and has programmable current limit.
This [paper](https://link.springer.com/content/pdf/10.1007%2Fs10894-018-0167-0.pdf#page=10) describes both my project and HV supply. Here is module schematic.
[HVSUP.PDF](https://github.com/sinara-hw/meta/files/2313858/HVSUP.PDF)
This would be probably an overkill for this project. But I plan to move this design to 3U format and use Humpback as a controller. I'm just curious if there would be any application for it in quantum world.

dhslichter, 2018-08-23:
I think for most purposes, something like 150-200V is all one needs.  In general most of the piezos we deal with are unipolar, and so would want something like 0-150V or 0-200 V.  Similarly, for trap voltages one generally wants to have positive voltages on the end cap electrodes.  However, one could envision use cases where bipolar output is desired for trap voltages in a number of cases, so I think it would be good to be able to have a bipolar output option.

hartytp, 2018-08-24:
> 3-4ch onto a Eurocard (stripped it down to just the DAC and HV section).

IMO, 4 channels is probably the right number for this design anyway.

hartytp, 2018-11-17:
I've had decent experience with the Thorlabs KPZ modules (apart from the firmware being hilariously buggy). While they're not in a form-factor that's as convenient as an EEM, they achieve a decent density, cost and are fairly low power. It would be good to articulate a bit more clearly the reasons why we need to reinvent this wheel rather than using something like KPZs.

dtcallcock, 2018-11-18:
I think a simple Eurocard carrier for the [PiezoDrive PDu150](https://www.piezodrive.com/modules/pdu150/) would be my preferred non-wheel-reinventing solution.  

It's about 5x cheaper and 100x lower noise than the KPZ.  It would need to be driven from Zotino/Fastino/Stabilizer (or front panel pot.) but I won't miss the junky Thorlabs firmware.

gkasprow, 2018-11-18:
This would occupy at least 16HP. They have smaller moodules like PDu100B
What about other suppliers like [E836](https://www.physikinstrumente.com/en/products/controllers-and-drivers/piezo-drivers-for-open-loop-operation-of-piezo-actuators/e-836-compact-piezo-amplifier-oem-module-601955/).

hartytp, 2018-11-18:
@dtcallcock fair enough. That makes sense.

Conceptually I like the idea of doing something like the JQI solution (laser printer IC high voltage source with a feedback circuit to reduce the noise/improve the stability). But, while it would be a fun project, I'm put off by the amount of work it would be to get that working really reliably.

Something like a board with a potted module and a microprocessor + ethernet sounds like a decent compromise that could be done with an acceptable level of work. The firmware Greg's student wrote for Thermostat gives one a good base to work from when writing firmware for new projects like this (a lot of the time consuming boiler plate like hooking up ethernet, writing command handlers, etc is already done for one, which is great).

dtcallcock, 2018-11-20:
| |PDu150|E836|
|---|---|---|
|Voltage|-30 to +150V|-30 to +130V
|Gain|20|10|
|Channels|3|1|
|Cost/Channel|$120|TBC|
|Bandwidth)|180kHz (no load),1 kHz (1uF)|8kHz (no load)|
|Noise|26uV (0.3Hz-10MHz, 1uF load)|800uV(0-100kHz)|
|Power supply|24V|24V|

I asked PI for a price on the E836.

> This would occupy at least 16HP.

They make a PCB mount version.  I asked them for data on that and asked them how to go about switching out the heatsink.

jordens, 2018-11-21:
http://www.powerampdesign.net/pad138datasheet.html and the other models might also be interesting.

sbourdeauducq, 2018-11-21:
> Conceptually I like the idea of doing something like the JQI solution (laser printer IC high voltage source with a feedback circuit to reduce the noise/improve the stability).
> But, while it would be a fun project, I'm put off by the amount of work it would be to get that working really reliably.

Were there a lot of problems at JQI? I don't think it's an annoying circuit to get to work well - you don't even need to make your own transformer, and the chips can't have obscure "features".

dtcallcock, 2018-11-21:
> http://www.powerampdesign.net/pad138datasheet.html and the other models might also be interesting.

I asked for pricing.  The quiescent power dissipation (3.2W/ch vs 0.8W/ch for PDu150) and requirement for external power supply are less convenient.  The fact it is an op-amp makes it more flexible for driving trap voltages though.

dtcallcock, 2018-11-21:
> What about other suppliers like E836.

As $446/channel I don't think it's competitive with the PDu150.

dtcallcock, 2018-11-21:
I got Altium files for the PCB mount version of the PDU150 and confirmed they can sell it without a heatsink so that we can add our own to make a lower profile (8HP?) card.

gkasprow, 2018-11-21:
We would need dedicated 24V power entry. And it should be different than DC jack already used. Maybe we should switch to SMPS modules mounted to the rack rear wall?

hartytp, 2018-11-21:
> We would need dedicated 24V power entry.

Why not just a 12V->24V SMPS?

hartytp, 2018-11-21:
@gkasprow I'd be a bit surprised if a 3 channel unit used more than 60W. 20W per channel seems like a lot. The PDu150 has a peak current of 300mA, so 7.2W which is well below the 60W that our 12V supplies can deliver.

FWIW, I find it really nice to have a big box of 12V PSUs and to be able to power anything in the lab from them...

gkasprow, 2018-11-21:
OK, so no problem then. Conversion from 12 to 24V is not an issue either, there are high efficiency synchronous converters that do it nicely

gkasprow, 2018-11-21:
For example [this one](https://www.analog.com/en/design-center/reference-designs/circuit-collections/ltc3703-12v-to-24v-5a-synchronous-boost-converter.html#cc-overview)

hartytp, 2018-11-21:
Nice! Well, that would definitely be my preference then.

gkasprow, 2018-11-21:
Shall such Zapper be controlled over EEM or by Stabilizer?

dtcallcock, 2018-11-21:
>> Stabilizer
> Definitely not by Stabilizer IMO.

It will need to be controlled by Stabilizer in some applications (cavity/laser piezo).  However I think it would be easier to do this via an analog connection.  If so we just need to decide whether to supply a front panel connector or use a header and an internal connection.

hartytp, 2018-11-21:
@dtcallcock ack. I hadn't considered that use case.

> If so we just need to decide whether to supply a front panel connector or use a header and an internal connection.

Do both like Zotino? Just need to watch out since on the current stabilizer plan we mix both DACs and ADCs onto the same header.

dtcallcock, 2018-11-21:
>So you would have max 2A at 24V

From PDu150 datasheet: The maximum supply current is 0.9A at full power.

dhslichter, 2018-11-26:
For everyone considering PDu150, the outputs are differential only, NOT single ended, and you cannot ground-reference one output.  Thus it won't work for typical piezo actuators where the signal is ground-referenced on one lead.

dhslichter, 2018-11-26:
The PDu150 is designed with a differential push-pull with two +/- 75V op amps, one could basically adapt some of the design to make your own version that is a proper single-ended output.  The circuit design is not complicated.

gkasprow, 2018-11-26:
do you have good photo of the PDu150 layout?

dhslichter, 2018-11-27:
No, everything is potted in a thin layer of corona dope but you can at least scratch off and see what the components are.  I will have to go hunt down the one we have, but if @dtcallcock has the Altium files that might be easier?

gkasprow, 2018-11-28:
It depends if we want to make own amplifier that can drive GND referenced loads or use COTS modules.

dhslichter, 2018-11-28:
It seems to me that ground-referenced loads represent a substantial fraction of the applications (probably the majority of applications), so using a COTS module that can drive such loads (i.e. not PDu150) is how we'd want to go, if we don't want to design our own amplifier.

gkasprow, 2018-11-28:
I can try to simulate some stage if we won't find COTS one that fits all needs.
Years ago I designed some HV piezo drivers for AFM applications.  It was fun to design and a lot of sweat to make it running. Don't you have some broken or not needed driver on which I could do reverse engineering ? As usual devil is in the details.

dhslichter, 2018-11-29:
LTC6091 is the output amplifier stage; I can look a bit more about the components (there are some integrated boost regulators to make the high voltage rails as well).  Not sure if we have a unit we could loan you for testing but perhaps.

dhslichter, 2018-11-30:
Looks like the [PDm200B](https://www.piezodrive.com/modules/pdm200b/) can run unipolar or bipolar in various ranges, with reasonable noise (1 mV RMS @ 50 kHz bandwidth, 100 uV RMS @ 70 Hz bandwidth).  Might be a good compromise, cost is low as well.

gkasprow, 2019-02-06:
One of my students decided to build such a module. It will be his engineering thesis. 
So, moving this issue to a dedicated repository.