sinara-hw/meta #14: [RFC] Low cost FMC carrier

gkasprow, 2018-08-28:
Some team from US lab which I cannot disclose right now (their procurement would not like it), want us to build universal, low cost FMC carrier with specification:
- designed using KICAD, of course open source.
- same FPGA as Kasli, 484ball Artix 7
- 2x FMC LPC
- ext 12V supply, similar as Kasli
- additional monitoring of FMC voltages/currents
- 4 SFPs
- RJ45 Ethernet
- White rabbit support
- stand-alone operation aka "pizza box"
- simplest possible implementation, no unnecessary options. @sbourdeauducq would appreciate it :D
- the box dimensions would be sth like AMC board giving access to the FMC and SFPs.

I'm thinking about making it as AMC board with SFPs installed in place of RTM connector and GTPs routed to SFPs or to the AMC connector. Ethernet would be routed either to RJ45 or PORT0. In this way same design could be reused as stand-alone or MTCA board.
Would such board be attractive for quantum labs? There is large selection of open-source ADCs, DACs, IOs, etc.
We could make it running with PoE supply (module mounted as option), i.e. as fast feedback controller.

jordens, 2018-08-29:
Like a cross between Kasli and that Inrevium board (200T, 2 HPC, no SFP). I know one lab where that's used and another where it would be interesting. In both cases with 8 channel ADC and 8 channel DAC HPC FMCs. And yes: feedback/DSP.

hartytp, 2018-08-29:
@gkasprow I can see a lot of potential use-cases for a nice way of interfacing FMCs to ARTIQ.

gkasprow, 2018-08-29:
@hartytp @jordens what form factor would be most welcome? 6U Euro or AMC? For the lab that want to fund it stand-alone aluminium box is optimum. But I can choose any board size.

hartytp, 2018-08-29:
> @hartytp @jordens what form factor would be most welcome? 6U Euro or AMC? For the lab that want to fund it stand-alone aluminium box is optimum. But I can choose any board size.

You can't fit 2 FMCs on a Eurocard, can you? Also, a lot of FMC cards are quite power hungry, so I might worry about cooling when using this in a Euro-rack.

If you have a reasonably-priced standalone AMC box with a good cooling and power solution, then that would probably be my preference

sbourdeauducq, 2018-08-29:
> You can't fit 2 FMCs on a Eurocard, can you?

6U.

> I might worry about cooling when using this in a Euro-rack.

With microTCA and its IPMI bugs, the fans can turn off with the firmware thinking they are at 100% :)

hartytp, 2018-08-29:
> 6U

Right, sorry. I'm less keen on 6U Eurocards.

Personally, I'd go for AMC so long as there is a good single-unit box with simple fan + power control that already works robustly without more firmware development needed. But, I don't feel strongly about that

gkasprow, 2018-08-29:
with 6U one can fit all connectors on front side. There are also single 6U slot COTS boxes. 6U PCB is large, much larger than 2x 3U. But it is new form factor to our ecosystem
AMC is much more compact and PCB will be much cheaper. I think we will go for AMC + very simple box with cooling. 6U would have to be in VME format to gain some popularity.
Anyway I think the simple box I developed for AMC + RTM is still too big for some applications and there is place for completely passive box for single AMC only.
3U is 160x100 = 160cm2
6U is 160x233 = 372cm2
AMC is 147x180 = 265 cm2

michallgaska, 2018-08-29:
maybe we can use this type of case for a single AMC card:
https://www.tme.eu/gb/details/hm-1455t2201bk/enclosures-with-panel/hammond/1455t2201bk/
This one is a little too wide (160 cm for PCB) but we can add removable frame to PCB and thus keep compatibility with AMC slots. 
Of course this aluminum case requires machining for IO connectors and cooling gaps.

gkasprow, 2018-08-29:
It must be exactly 180mm, so we either cut it or make own.

jordens, 2018-08-29:
I would prefer AMC and alternatively a stand-alone box.

jordens, 2018-08-29:
4U crates could also be built. An AMC PCB would be just 4.05 mm too high.

gkasprow, 2018-08-29:
I have some experience with custom enclosures and they cost more less the same as COTS ones + engraving and milling. 
It's often easier to design own enclosure than modifying existing one.
Anyway, let's try to fit it to AMC form factor. There would be much more use cases than 6U or 4U ones.

jbqubit, 2018-08-29:
> I think AMC is a better choice here than introducing an entirely new form factor of our own devising.

I agree with @dhslichter. There's been a host of surprising and annoying problems with the uTCA form factor. AFAICT WUT looks to have it under control now. Provided past follies aren't repeated the AMC path forward to straightforward.

sbourdeauducq, 2018-08-30:
> There's been a host of surprising and annoying problems with the uTCA form factor.

They were not surprising, in fact I told you the uTCA gadgetry was going to be overly complex and buggy.

gkasprow, 2018-09-14:
@hartytp @sbourdeauducq @jordens @jbqubit @dhslichter We are  looking for a good name for this project.

dtcallcock, 2018-09-14:
FMCker?

gkasprow, 2018-11-18:
The specification is frozen, schematics are 90% done so closing. Will create another issue in dedicated repo.

closed 2018-11-18 by gkasprow

gkasprow, 2019-03-16:
The board schematics are finished, the layout is also nearly done.
https://github.com/gkasprow/Simple_AFC/releases
@sbourdeauducq guys will talk with you about Migen support for this board.