sinara-hw/meta #17: sinara-hw tidying up

dtcallcock, 2018-08-31:
It's really great that Sinara has its own organisation and sensible repository structure now.  Thanks to everyone who made that happen.  However, I think a few small changes could make it less intimidating and easy to navigate for people new to the project.  Here's a few suggestions:

- [ ] Make a website and landing page for the project using [Github Pages](https://pages.github.com).  This page could be fairly static, just giving an overview of the project and pointing people to the wiki etc.  In any case it should be no harder to maintain than the wiki as it can be written in markdown and live on GitHub. 
- [ ] Move relevant wiki pages from Sinara to meta, delink the depreciated pages for individual boards and link the new board wikis.
- [ ] Make it clear that the Sinara repository is mainly depreciated except as a place for the Sayma issue tracker.
- [ ] Make a readme file for meta pointing people to where the action is (Issues and Wiki).
- [ ] Address anything left in and/or close this issue: https://github.com/sinara-hw/sinara/issues/208

I am happy to pitch in with this.  Also happy to leave some of this if there's a reason for how it is right now.

gkasprow, 2018-09-01:
It's here
https://sinara-hw.github.io/

dtcallcock, 2018-09-01:
Thanks @gkasprow  

I was hoping we could use the [built-in Jekyll integration](https://help.github.com/articles/using-jekyll-as-a-static-site-generator-with-github-pages/) so that the website can be in markdown instead of html.  If you give me write access to the sinara-hw.github.io repository I will set it up and you can see if you prefer it to the html.

gkasprow, 2018-09-01:
Done.
I didn't know about JekyII. I did what was suggested on Github Pages :)

dtcallcock, 2018-09-02:
I made a simple Jekyll webpage with the information I thought would be useful to somebody coming across Sinara for the first time.  People should feel free to iterate.

@gkasprow can you make https://sinara-hw.github.io/ the website of the sinara-hw organization so that the link appears at the top of the page for people who go directly to 
https://github.com/sinara-hw

dtcallcock, 2018-09-02:
I rebuilt the wiki at https://github.com/sinara-hw/meta/wiki including moving all the images over so there aren't dependencies on the old Sinara repository.  Most of the links are fixed too but I don't have access to the dio_bnc, mirny and VHDCI_Carrier repositories so couldn't fix those (can you give me access @gkasprow).  I cleared out a bit of cruft but I didn't want to touch the stuff I'm not up to speed on (mostly uTCA stuff).

gkasprow, 2018-09-04:
done

dtcallcock, 2018-10-24:
@gkasprow As part of the effort to drive new people to the right places would you mind doing the following if you have a second:

- Pin the 'meta' repository to the top of the organisation's page.  See: https://twitter.com/github/status/791759883800571905?lang=en

- Change the description of 'sinara' repository from 'Hardware designs for the ARTIQ DRTIO ecosystem (Sayma, Metlino, Kasli)' to something like 'Sayma issue tracker'

- Move the tags from the 'sinara' to 'meta' repository (or just delete them).

- Add a description of the 'thermostat', 'mirny', and 'VHDCI_carrier' repositories and delete the empty 'booster-circuit' repository.

marmeladapk, 2018-11-07:
Just a heads-up: GitHub recently added an option to move issues to other repositories: https://help.github.com/articles/transferring-an-issue-to-another-repository/

gkasprow, 2018-11-17:
I know that level of documentation is poor. This was one of the reasons why we asked for money for commercialisation of the equipment.  Writing decent documentation is part of it.
Within this grant CTI will:
- do review of project, replace EOL components, prepare for mass production
- prepare automated test benches. This is already done for some modules.
- write documentation.
This grant is continuation of previous one which was funded to commercialise OHWR modules. Documentation example is [here](http://creotech.pl/wp-content/uploads/2015/08/CTI-AFC_datasheet.pdf)

Another thing is wiki. All documentation should start here and point to project resources and finally to web shop where products can be purchased. This process has already started but this will take time.
I update regularly page with status of modules, but this should have more interactive form.

hartytp, 2018-11-17:
That's really good new @gkasprow. Looking forward to seeing some of those docs come on line.

There is quite a lot of test data for these boards scattered through the issues. It would be great if some data like this could make it into the final data sheets.

gkasprow, 2018-11-21:
I added initial wiki pages for Banker, Stabilizer and Humpback

drewrisinger, 2018-11-29:
The Kasli link in this section (https://github.com/sinara-hw/meta/wiki#carriers) is broken. Points to VHDCI carrier instead. As a side note, if there's some way for the community to edit the wiki, that'd be great.

gkasprow, 2018-11-29:
fixed.
@drewrisinger I gave you write rights.

gkasprow, 2019-01-13:
I added wiki pages , updated status for all modules.

dtcallcock, 2019-08-09:
Related to the above, but spurred by comments in https://github.com/sinara-hw/Zotino/issues/24

> hunting down the BOM hidden under releases

Releases is the right place to put the BOMs and pdf output files.  However, nobody seems to expect that's where they would be (partially because GitHub hides the 'releases' tab pretty well).  I tried to make this clear on https://sinara-hw.github.io but I don't know if there is a better way to signpost it.  Like a sentence that states this at the top of every wiki page?

> Wiki editing is limited to members of the sinara-hw Github group

I know this is a can of worms but would some statement somewhere about the governance/ownership/purpose of sinara-hw (or at least how to get editing rights) be useful to new people?  People thinking it's some kind of closed shop or an internal M-Labs project isn't good for expending the user base or participation.

dtcallcock, 2019-08-09:
Also, can you take a look at this if you get a second @gkasprow ?

> Pin the 'meta' repository to the top of the organisation's page. See: https://twitter.com/github/status/791759883800571905?lang=en

drewrisinger, 2019-09-03:
> Releases is the right place to put the BOMs and pdf output files. However, nobody seems to expect that's where they would be (partially because GitHub hides the 'releases' tab pretty well). I tried [...]

We should be able to put a link to the project's release page in the base README.md on each project's GitHub page. See https://github.com/sinara-hw/Zotino/pull/25 for an example. The wikis would also be an appropriate place to put links to the design files.

dtcallcock, 2020-06-17:
> We should be able to put a link to the project's release page in the base README.md on each project's GitHub page.

I have now done this for most of the repositories, but lack write access to Phaser, Driver, CompactPCISerial_EEM_Adapter, Humpback, IDC_HD68_Adapter, Kasli-backplane-adapter.

Can you fix this for me @gkasprow ? 

The repository descriptions are also all over the shop or missing. I'm happy to sort this out but I'd need to be elevated to administrator on most of the repositories (I can currently only edit one or two of them).

gkasprow, 2020-06-17:
You have the Maintain rights now.

jordens, 2020-06-17:
thanks @dtcallcock