sinara-hw/meta #19: ARTIQ Sinara user poll 2018

jbqubit, 2018-09-13:
I'd like to conduct another user poll. I've based the current year's questions on those asked in 2017. This may permit a nice differential showing how the community has grown. Please comment in this Issue. 

https://docs.google.com/forms/d/e/1FAIpQLSe87HSZU7x1Z9CB9_40WdXsSTmez6p8Ln4-0DgbH3D5tpUlLA/viewform

Feel free to play around with the Poll. I'll reset its contents and more widely publicize once we've agreed on questions to ask.

marmeladapk, 2018-09-13:
I feel like adding ubuntu 16.04 (maybe even 18.04) to OS options would be useful. You could also add "(or derivative)" for linux options. Perhaps centOS options would also be helpful?

I think that adding booster to hardware questions would also be helpful.

ARTIQ experience - there's an "Option 12" option.

jbqubit, 2018-09-18:
> "For-Contract Features", and I would put the links in the actual questions where they are asked, rather than at the top of the page (kind of hard to read).

The Form tool doesn't accommodate URLs or carriage returns in individual questions. It looks even messier with the URL attached to the question. 

I've added several new questions.
- Sinara PCBs used by your labs (list of check boxes)
- Rate your interest in the EEM form factor for future hardware.
- Rate your interest in the microTCA form factor for future hardware.
- What missing capability or feature is impacting your use or adoption of ARTIQ/Sinara most?

jbqubit, 2018-09-21:
Thanks for the feedback everyone. I'll send out notice of the poll to the ARTIQ mailing list. @sbourdeauducq please also tweet it. :) -Joe

closed 2018-09-21 by jbqubit