sinara-hw/meta #20: [RFC] simple VHDCI carrier

gkasprow, 2018-09-23:
@jordens @hartytp @dhslichter I finished extension boards for Banker and an idea came to my mind. We can make simple LVDS VHDCI to EDGE connector converter. In this way we can use BNC/SMA/DSUB adapters directly with FMC-VHDCI or Metlino once it becomes available. In this way up to 32 simple IOs can be connected directly without VHDCI carrier + EEM IOs.
I don't know if it is worth the effort since it duplicates the EEM idea.

jordens, 2018-09-24:
The modules are really nice. I guess doing 4 edge connectors on each side would have been a bit crowded.
About the lvds vhdci to edge converter I personally am a bit hesitant. It's nice for really simple DIO from sayma/metlino/kc705 and similar. But probably people would primarily go for the existing solutions since it is extensible and more powerful.

dhslichter, 2018-09-28:
I think the existing EEM DIO breakout boards are fine from my point of view, although this is a nice idea.

closed 2018-11-17 by gkasprow