sinara-hw/meta #23: [RFC]: Current source

hartytp, 2018-10-16:
Something I've had a lot of questions about recently is an ARTIQ/Sinara current source:
- target is magnetic coils for ions/atoms
- currents from 1A to a few hundred A
- low noise/high-stability
- modulation BW of order 10kHz
- probably an EEM for control by Kasli

Suggestion:
- keep this modular, separate current source from modulation/stabilisation. 
- we have a [design](https://github.com/OxfordIonTrapGroup/CurrentStabilizer) for a current stabilisation module. [Paper](https://arxiv.org/pdf/1808.03310)
  - This measures the current with a 4-terminal resistor. 
  - low-noise differential front-end
  - error subtraction and amplification
  - loop filter
  - loop-filter output controls a transistor used as a current modulator (shunts current away from the coil to stabilise the current through the coil)
  - second actuator that's driven from a DAC used for feed-forwards to remove ambient AC fields (e.g. 50Hz noise). Key different between the two modulators is that the feed-forwards modulator is connected so that it does not change the current measured by the sense resistor
  - for low-ish currents the sense resistor measures the current directly. For higher currents one uses a transducer like a LEM Ultrastab 400 to reduce the current to a range that's easier to measure with a resistor
- works really well (see paper), however has a few issues:
  - not on a convenient Eurocard format
  - no EEM to provide convenient programming/modulation
  - loop factor is analog so inconvenient to tune. Would be much better to use an ADC after the error amplifier to digitise the error signal. Then do loop filter in DSP and program both DACs from microprocessor. This would also make modulation easier. Moreover, with an ADC one can log the error signal while stepping the set point to make loop filter optimisation easier. Note that due to inductance of coils, getting really good performance requires the loop filter to be individually optimised for each setup; a generic loop filter provides quite bad performance (bw limited to a few hz usually)
  - Best would be if the DACs weren't on the eurocard. Instead, it should have  a digital output and put the DACs + shunt/modulation transistors near the load
- Plan would be to combine this stabilisation module with a suitable programmable (but not very low-noise or high-stability) current driver. For higher currents, one uses an Agilent current source or similar, and programs the current over ethernet to be somewhat bigger than the required coil current. The modulator shunts away enough current to achieve the desired current
- for lower currents, it would be nice to have a eurocard with a 5-10A programmable source, as well as the stabilisation transistors. The current source is programmed via EEM to give an appropriate current.

gkasprow, 2018-10-16:
We can make it low cost with Kasli serving as FPGA resource or slightly higher cost with dedicated FPGA from Xilinx or Lattice. But this also means additional work on programming interface.
Shall this module control external current source with dedicated interface?
Aren't we trying to invent ADC + DAC module which we talked earlier?
With external shunt and external current modulator, we are essentially converging to genera purpose ADC + DAC module :)

hartytp, 2018-10-16:
> I would suggest we not try to make an EEM that provides an actual current source, since these will be challenging engineering to make low noise and keep within power and space constraints in a Eurocard rack, and existing programmable commercial ones with good performance and compact size can be had quite cheaply.

@dhslichter For the typical ions requirement -- producing a constant current of 100A or so -- I'd agree with you.

The EEM programmable current supply is more directed at neutral atom experiments. The key difference there being that they want to do things like ramp their current from 0A to 10A reasonably fast. I'm assuming that the current shunt/modulation transistors will only draw a few hundred mA or so, so large modulations require the main current source to be modulated as well. This in turn requires that the main current source can be modulated from Kasli with decent bandwidth. Hence the requirement to build an EEM current source as well. That EEM current source doesn't need to be particularly low-noise since it's cleaned up by the current stabiliser module, so the design is relatively straightforwards.

My thinking is that we have two modules: a current stabilisation module, containing the sense resistor, loop filter and ADC; and a current source module with a programmable current source, as well as the stabilisation DAC + transistor. The DAC on the current source module is controlled by the stabilisation board.

hartytp, 2018-10-16:
Anyway, it would be good to have some input from the neutrals guys on what they need/how they'd like to see it implemented...

hartytp, 2018-10-16:
> We can make it low cost with Kasli serving as FPGA resource or slightly higher cost with dedicated FPGA from Xilinx or Lattice. 

For 1MSPS ADCs + DAC, this can be done on a decent microprocessor.

> But this also means additional work on programming interface.

Yes, creating a decent programming interface (e.g. nice ways of recording traces for step response measurements, ways of programming complex loop filters etc) would be a large fraction of the work. If they're interested, this could be something that M-Labs could be funded to do.

> Shall this module control external current source with dedicated interface?

As I said, I currently imagine two modules: a general purpose current stabilisation module, which measures the current, and has a digital loop filter, with a digital output that controls a DAC + bypass transistor; and, a separate current supply, which could either be some COST box or a separate EEM, depending on the experiment. 

> Aren't we trying to invent ADC + DAC module which we talked earlier?
With external shunt and external current modulator, we are essentially converging to general purpose ADC + DAC module :)

Close, but there are some differences.

Firstly, the ADC + DAC module would probably be a general purpose module without an AFE. However, to achieve the noise/stability levels we need requires a carefully designed AFE (see our paper).

Secondly, and related, the resolution required for many experiments is better than 16 bits. That's why we chose a topology with two cascaded DACs with a gain stage in between. i.e. we have a "coarse" DAC that sets the range and a "fine" DAC that controls the set point. The 16 bit ADC would be placed after set-point subtraction and amplification; a general purpose board with simple AFE and a 16-bit ADC wouldn't be nearly good enough.

Thirdly, we need a specialist 4-terminal current sense resistor for this, while the ADC + DAC board would just measure voltages. The sense resistor could be placed on an external PCB, but then one has to be really careful about grounding/pickup/thermal management/etc.

Finally, the overall topology in terms of connections for the ADCs and DACs is a bit specialist here. It could be done with a general purpose board, but things would end up being a bit of a mess. This is a common enough use-case that it's worth having a dedicated design for it.

dhslichter, 2018-10-16:
@hartytp why are we worried about building current sources for neutral atom experiments?  I think they should make their own designs if and when they decide that ARTIQ is what they want to use, it just seems like this is a very low priority task for us right now.  Let them take the lead here.  

Current stabilization for ion trap quantization fields is certainly something that would be nice to make an EEM for.  I agree that it's something fancier than one might want to do with a "general purpose" DAC + ADC board.

dhslichter, 2018-10-17:
@hartytp ack.  I would encourage the relevant neutral atom groups to come and comment on this post -- can you tag appropriate people?  It would be good to involve more people in the hardware specification process, particularly for boards they find themselves interested in having.

hartytp, 2018-10-17:
Yes, that's the intention. Have a little patience while we coordinate, we're not in a rush for this design afaict.

we36sussex, 2018-10-23:
Hello,

Firstly, we just wanted to introduce ourselves as we’ve been in the shadows for a while. We’re a group of neutral atom trappers from the University of Sussex and have purchased a number of ARTIQ systems for controlling experiments in our lab. I am a PhD student in the group and am working with two postdocs, Mark Bason and Tom Barrett.

In terms of current sources, we are in the process of developing low-noise atom chip drivers, something which would be relevant to many neutral atom groups, and low-noise coil drivers for residual field compensation. There are a number of existing designs we have looked at, [Erickson](https://aip.scitation.org/doi/10.1063/1.2953597), [Libbrecht-Hall,](https://aip.scitation.org/doi/10.1063/1.1143949) [Dudzik,](https://ieeexplore.ieee.org/document/8114598) but we were interested to see if there are any plans for ARTIQ/Sinara designs.

For the atom chip drivers that we’re interested in:
	Current range 0 to 20 A
	Each channel independent PID control with < mA precision
	Ability to ramp 0 – max in < 1 ms
	Fast-switching, switch off in < 100 μs
	Integrated into EEM/Kasli hardware
	Modular approach - drive several channels from one supply
	Preferably using a sense resistor over [current sensors](https://www.lem.com/en/itn-12p) ~ £ 500 
	Explicit analogue control as an option

For low current field compensation coil drivers:
	Current range 0-100 mA and 0-3 A
	Each channel independent PID control with high precision
	Independent current sources for each channel
	Bandwidth DC – 10 KHz
	Integrated into EEM/Kasli hardware
	Feedback signal from a sense resistor

We would also be interested in the current stabilisation circuit for external power supplies that Tom has described above for higher currents.

From the comments above it looks as though it would be possible to design a range of current drivers in the Eurocard standard. We would be interested in financially contributing to such a project, could you give us some details about how that might work in practise and what sort of structures other projects have taken?

hartytp, 2018-10-23:
Hi @we36sussex! Thanks for joining the conversation. 

At a high-level, I think there are two approaches we can take here for lower currents (say, 20A or less):

1. design a few different ultra low-noise/drift current sources covering different I/V ranges. Each circuit would have to incorporate a sense element + feedback components. Call this the monolithic approach.
2. design a single ultra-low noise stabilisation circuit as one EEM and combine that with relatively cheap current source EEMs (e.g. IC SMPSs with some LC filtering) to give ultra-low noise performance at different I/V levels. Call this the modular approach.

The stabilisation circuit would probably have a 4-terminal resistor as its input allowing to either be used to measure Amp-level currents directly, or to be combined with a current transducer for higher currents. By choosing an appropriate sense resistor + optional transducer, the same stabilisation circuit (with different loop filter settings) could be used for essentially any current level.

For currents above about >20A the modular approach (probably combined with an off-the shelf high-current PSU) is probably the only sensible approach.

> We would be interested in financially contributing to such a project, 

Great!

> could you give us some details about how that might work in practise and what sort of structures other projects have taken?

Absolutely. @gkasprow is the person to speak to about hardware designs and @sbourdeauducq/@jordens about the software. I'm happy to talk about our experiences -- just email me.

jordens, 2018-10-24:
@we36sussex The way this can work depends a lot on what structure you are looking for in terms of project management. Let me know if you would like advice on that.
Independent from that you should start with the lobbying and the use case analysis to maximize the value of the specification by getting as many users behind it as possible while not making it too complicated.
AFAICS this won't be easy here because the the design will look different for each decade of full scale current. And there are users covering at least 3 decades (100 mA shim coils to 100 A neutral transport/trap/Feshbach).

dtcallcock, 2018-10-24:
> With external shunt and external current modulator, we are essentially converging to general purpose ADC + DAC module :)

>>It could be done with a general purpose board, but things would end up being a bit of a mess. This is a common enough use-case that it's worth having a dedicated design for it.

@hartytp  I disagree as this is how we implemented your board at NIST and it works really well (see below).  Therefore I would just focus resources on making a general purpose digital lock box ([which Sinara sorely lacks](https://github.com/sinara-hw/meta/issues/16)), and make the current sense board a modular add on.  This is how we use the Oxford board at NIST:

[image]

The Oxford board has everything other than the analog front end and error subtraction circuitry stripped off of it.    The only digital part of the sense board is the offset subtraction DAC.  We just talk to this over a USB-SPI adaptor as it only rarely needs changing.  SPI over RJ45 would be a convenient way of doing this for Kasli.  The 'lockbox' is the usual NIST FPGA-based unit.  We feedback directly to a voltage control input on the PSU.  This could also be a small current shunt board between the outputs of a commercial PSU  or a Eurocard PSU interfaced digitally with the lock box.  We inject our 60Hz feedforward directly into the lockbox.

Another reason to do it this way is that it avoids bringing the coil (or sense) current up to the Eurocard and back.  Long leads and electrical noise in the Eurocard chassis could be an issue.  For those who really want the current sense board in the Eurocard chassis though, it would be simple to make a bracket for it (it'll be a small board - 2"x2" or so).

hartytp, 2018-10-24:
@dtcallcock That's another way to go, which I would be happy with. If we go that way, I'd prefer to have a single EEM with, say, 2xADC and 2xDAC and a microprocessor/FPGA rather than bolting together something like Sampler+Fastino+Kasli.

Anything sensible along those lines would be a very welcome addition to Sinara as far as I'm concerned.

gkasprow, 2018-10-24:
Why we need another FPGA?:) Cannot the algorithm be implemented in Kasli?

hartytp, 2018-10-25:
> Why we need another FPGA?:) Cannot the algorithm be implemented in Kasli?

Maybe that's the right way to go. Some things to consider are:
- We probably want the sample rate for this to be as high as possible when using SPI parts. So, at least 1MSPS and ideally more like a few MSPS
- this can be problematic when the FPGA is not on the same card as the ADC due to meeting timing on the ADC SDO lines, which are have a round-trip delay + some delay from the ADC when compared with the SCK (generated by the FPGA).
- You then have to either limit the ribbon cable to quite short lengths or do something more fancy to allow you to take out the SDO->SCK delay. One option for this is to use an ADC which generates a delayed SCK signal and then latch SDO from that (or add a buffer to loop that signal back to the FPGA). It's not so hard, but it requires thought and also takes up more pins on the EEM connector
- Kasli has 12EEM connectors, which seems like a lot. But, they go fast when one starts using EEMs that take up multiple slots. For a 2-channel board (2 ADCs + 2DACs, so 4 BNCs on the front panel), we need to think about how many EEM connectors would be consumed and make sure it's reasonable
- maybe this is not a good argument, but I felt that it's nice to be able to update the servo card gateware without having to reflash the main Kasli (this could be what's running the experiment, so it's annoying to have to update it). I was thinking that we might want to move more in the direction of "distributed" local control units, rather than making Kasli do everything.
- We also need to think a little about timing etc when shoving too much logic on Kasli.  Probably okay with the higher speed grade, but I've certainly found that adding almost anything extra (on top of ARTIQ) to the -2 speed grade can cause timing failures. 
- finally do we need an FPGA for this, or can we implement a decent complexity feedback loop with a latency of ~2us using a microprocessor?

Anyway, just some thoughts, so long as this is sensibly implemented I don't mind how it's done.

niwotongzai, 2018-10-27:
Hello,  @we36sussex  has given a very detail explanation for the requirements of typical neutral atom experiments.  I have read the paper of Oxford ion group (arxiv 1808.03310),  and the excellent noise reduction and stability is enough for us.  At first, we want a current source in a single EEM card that can be controlled with Kasli. But @dtcallcock David's idea is also acceptable. And it'll be better to integrate the 60Hz feed forward and current setpoint into one board.

hartytp, 2018-10-27:
@dtcallcock one thing about this diagram: https://github.com/sinara-hw/meta/issues/23#issuecomment-432697334

Depending on how many mains harmonics one wants to take out, what loop filter bandwidth and order one has, etc one might prefer to have a separate modulator (i.e. one that modulates the coil current, but not the current seen by the current sensor) to do the FF rather than modulating the lock set-point.

we36sussex, 2018-11-02:
We would prefer to take up the monolithic approach, the very high precision circuit for high currents might be of interest further along the line.

We are trying to avoid using a large racks of commercial power supplies so would like to develop a set of standalone EEM current sources for 0-150 mA, 0-3 A and 0-15 A which can be programmed from a Kasli. An end atom-chip system may contain something like 10 x 15 A, 10 x 3 A, 4 x 0-150 mA.

This would look like the Erikson (mentioned above) circuit but with DACs for PID control and an option between a programmable digital set point and input analogue set point. Would it be enough to have a microprocessor/FPGA on board here with DAC values updated from the Kasli? 

To limit the number of EEM slots used would it be possible to have these addressed and the SPI bus daisy chained together in racks? It would also be helpful to have multiple channels on a single card (e.g. 1 x 15 A, 4 x 3 A, 8 x 150 mA) providing appropriate heat sinking fits.

We have a prototype Erikson PCB, a low-noise Twinleaf power supply, Agilent 6671A and a cheap SMPS which we which we will characterise over the coming weeks.

hartytp, 2018-11-17:
@we36sussex FYI: we've decided to move forward with the "modular approach" to this, meaning:
- We will use a COTS (TTI/Keysight) programmable PSU as our current source. The noise/stability of these PSUs will not be good enough for our application, so we will improve it using a feedback circuit
- @gkasprow is developing stabilizer feedback board (designs should be ready in the next couple of weeks)
- We will convert the Oxford current stabilisation circuit into a current sense AFE card for stabilizer (a small board with its own front panel that screws directly into stabilizer). This card will have a FP connection for current input and a current sense resistor, set-point subtractor and error amplifier. For the currents you're talking about, the current can be passed directly into the stabilizer AFE. For larger currents, the input will be from a LEM current transducer (we'll pick a sense resistor package with a decent range of resistance values available to allow users to match the resistor to their needs). This AFE board will also have the current shunts used for the feedback and for 50/60Hz (and harmonics) feedforward to remove ambient noise.
- If you combine this with a very simple 5A programmable current source EEM (e.g. a programmable SMPS + some basic passive filtering should do the trick) then it could provide a very nice solution to your needs without you having to do too much work.

hartytp, 2019-04-08:
Finished our version of this and posted [here](https://github.com/OxfordIonTrapGroup/stabilizer_current_sense)

we36sussex, 2019-04-11:
That looks great, thanks for the heads up. 

We're doing some final testing on our updated board too and would like to try interfacing that with the stabilizer once that's available . It would be a good opportunity for us to do a comparison so we will keep an eye out for any testing.

closed 2019-09-30 by hartytp