sinara-hw/meta #25: new EEM FPs

hartytp, 2018-10-23:
I've done some tests with the new EEM supplied by Creotech. As expected, they do a good job of connecting the PCB ground to the chassis. A grounding bolt on the chassis then ensures the rack is safe/doesn't float to 300V due to the mains PSU brick.

Are TechnoSystems currently shipping EEMs with these panels? If not, when will they start?

sbourdeauducq, 2018-10-23:
> A grounding bolt on the chassis then ensures the rack is safe/doesn't float to 300V due to the mains PSU brick.

Note: With the recommended adapter (AKM65US12) the ground of the output is connected to earth and this does not happen.

jordens, 2018-10-24:
They don't and last I heard yesterday they won't start with that for a while. Lead time issues.
@sbourdeauducq Who recommends that and why?

sbourdeauducq, 2018-10-24:
Technosystem; they also shipped them at some point with the Kasli. I don't know the detailed reasons, but they get the job done IME.

klickverbot, 2018-10-26:
I suspect the reason for this choice might just be that we (Oxford) have made good experiences with those PSUs in the past (and indeed are using them for our Kaslis now).

jordens, 2018-10-26:
I find the free galvanic isolation between the dirty mains earth and e.g. the zotino outputs very convenient. On top of getting the job done and having a smaller connector.

hartytp, 2018-11-17:
> I suspect the reason for this choice might just be that we (Oxford) have made good experiences with those PSUs in the past (and indeed are using them for our Kaslis now).

I think it was me who originally chose those PSUs. They were the first one I found on Farnell which looked like it would do the job. I've had no issues with using them in the lab, but that's not to say that there aren't other options that would work better in other circumstances.

Anyway that's off-topic for this thread, which was about FPs.

closed 2018-11-17 by hartytp