sinara-hw/meta #27: [RFC] New EEM: Slow Microcontroller Servo

dtcallcock, 2018-11-02:
A fresh issue for one of the two boards proposed in [[RFC] Sinara Servo #16](https://github.com/sinara-hw/meta/issues/16).

This would be a lower performance board with the feedback loop running on a microcontroller. 

- ~MSPS DACs/ADCs and ~us latency
- Ethernet for standalone operation
- Microcontroller/ethernet layout could be copied from proposed [Humpback](https://github.com/sinara-hw/meta/issues/5) target like [ST Nucleo 144](https://www.digikey.com/product-detail/en/stmicroelectronics/NUCLEO-F429ZI/497-16280-ND/5806777) to reduce risk and share development costs.
- One EEM to allow a Kasli to control lock parameters if desired _or_ to connect a DIO EEM.
- Low cost, of order $100/ch

This board could also be used as a general lab logger/controller to replace [LabJacks](https://labjack.com/products/t7).

hartytp, 2018-11-08:
@gkasprow FWIW (e.g. in case you have a masters' student looking for a project), other than thermostat, out of all the "simple EEMs" we're discussing, this is the one that would be the most useful for me.

A nice simple HW design with some 16-bit ADCs+DACs, minimal AFE and a decent microprocessor, with a decent software interface would be amazing.

hartytp, 2018-11-08:
Great! A simple but well-thought out hw design along with a decent SW interface that allows one to specify IIR/FIR coefficients in a sensible way, record ADC traces (step response measurements etc) would make this a wonderful tool.

hartytp, 2018-11-08:
Design sketch [WIP]
- *FP Connectors*: 12V power barrel connector; 4 x insulated BNC; USB micro; ethernet
- *Internal connectors*: JTAG (microprocessor programming); EEM connectro; 100mil header with some GPIO exposed for misc use
- *Inputs*: simple RFI filter + protection diodes (copy sampler); PGIA?; choice of 16-bit ADC. Input range is +-10V
- *Outputs*: choice of 16-bit DAC?; OpAmp buffer/driver; filter with ~300kHz ABW
- *microprocessor*: same TM4C1294NCPDTT3R as Thermostat

To do: fill in blanks

hartytp, 2018-11-08:
@gkasprow can I have permissions set to edit posts, please?

gkasprow, 2018-11-08:
done

hartytp, 2018-11-08:
hmm that didn't seem to work.

gkasprow, 2018-11-08:
[image]

hartytp, 2018-11-08:
Re ADC:
- There aren't any good options I'm aware of for this which take a +-10V input, so let's go for a +5V ADC
- Let's go for 2 single channel ADCs rather than worrying about a dual SPI bus
- A good choice looks to be LTC2310-16 there is also LTC2311-16 if we think it's worth going for a 5MSPS data rate (I don't)
  -  Relatively cheap
  - 16 bit
  - 2MSPS LVCMOS interface 
  - on-board 20ppm/K 4.096V reference
  - 1.7LSB RMS noise

hartytp, 2018-11-08:
Since that's the same ADC as sampler, let's keep the input AFE the same as sampler to minimize work).

The only change I'd make to the sampler AFE is looking for a lower-gain faster InAmp. IIRC there is a model in the same family as the Sampler ADC that does that.

hartytp, 2018-11-08:
There actually aren't many options for decent speed (>1MHz ABW), digitally programmable gain InAmps with decent ish noise. On Sampler we used [AD8253](https://www.analog.com/media/en/technical-documentation/data-sheets/AD8253.pdf). For this, I'd go for [AD8250](https://www.analog.com/media/en/technical-documentation/data-sheets/AD8250.pdf) Same part as sampler, but G=1,2,5,10 instead of 1,10,100,1000.

So, unless anyone objects, that's the ADC AFE done.

gkasprow, 2018-11-08:
This CPU has four high speed QSPI blocks. We will attach Sampler to the TI devkit to make sure it works.

hartytp, 2018-11-08:
> This CPU has four high speed QSPI blocks. We will attach Sampler to the TI devkit to make sure it works.

If we go for 2 x LTC2310-16 then we can just use standard SPI, which seems easier.

@gkasprow For the DAC I'd ideally like to use a bipolar DAC to avoid having to have an offset+gain stage. However, there really isn't much choice of: +-10V; >=1MSPS; >=16-bit. The only decent option I'm aware of is DAC8871. Do you know any others? If not, let's go for that part as the DAC.

All we need then is a simple 2-pole buffer/active filter (~300KHz ABW) after the DAC.

So, that's the "specification" part of the design more or less done...

hartytp, 2018-11-08:
> DAC8871

Well, that DAC is actually a bit annoying. So, maybe better to just use a standard 5V DAC and use the output OpAmp to do the gain + offset subtraction.

dtcallcock, 2018-11-08:
A lot of lock applications have one or two auxiliary voltages around that one would like to measure and log.  I therefore propose we add 2-4 channels of auxiliary ADC.  Bandwidth wouldn't need to be more than say 10Hz, so a cheap and simple I2C ADC could be used (eg. [LTC2493](https://www.analog.com/en/products/ltc2493.html)).

> microprocessor: same TM4C1294NCPDTT3R as Thermostat

This chip seems a little weedy.  Given microcontrollers are basically free, I advocate for the most powerful part that's convenient to use.  Something from the STM32 line perhaps?

> I'd go for AD8250 Same part as sampler, but G=1,2,5,10 instead of 1,10,100,1000.

The 100 & 1000 gain sound like they'd be more useful than 2 & 5.  The parts seem to have almost identical performance at 1 & 10.

hartytp, 2018-11-08:
> A lot of lock applications have one or two auxiliary voltages around that one would like to measure and log. I therefore propose we add 2-4 channels of auxiliary ADC

I'm wary of trying to cram too much functionality into this design. If you want a general-purpose multi-channel logger then there are easier ways.

If we do add the extra ADC channels, how do you want them exposed? There won't be room on the FP for that many extra BNCs. Would putting some extra ADC channels on an internal 100mil header along with some GPIO and supply voltages work for you?

> This chip seems a little weedy. Given microcontrollers are basically free, I advocate for the most powerful part that's convenient to use. Something from the STM32 line perhaps?

That's a good point. So long as no one has any objection to working with those chips, that sounds like a good idea.

> The 100 & 1000 gain sound like they'd be more useful than 2 & 5. The parts seem to have almost identical performance at 1 & 10.

I'm not too fussed either way. The G=2/5 are nice for getting the best possible signal to noise. For example, a uni-polar signal at around 3V on a 20V (+-10V) input span is a bit painful in terms of loss of resolution.

IIRC the G=1k option tends not to be that useful with a MSPS 16-bit ADC because one is limited by integrated noise from the InAmp (which is the same for G=100 and G=1k) rather than the ADC noise. Thus, switching from G=100 to G=1k basically makes the whole thing slower without giving a win in terms of S/N -- although I might be wrong about that, since it's been a while since I thought this through.

The other issue with the high gain settings is that the bandwidth is very low.

hartytp, 2018-11-08:
[A few more options for the DAC](https://www.analog.com/en/parametricsearch/10888#/p7=|6|13|16|17|18|20&p165=|1&p4129=2.2e-9|0.0000015&p4458=|Byte|DDR%20LVDS|DHSTL|I²C|I²C/Ser%202-wire|MicroWire|Nibble|Pulse%20Mode|PWM|QSPI|SERDES|Serial|Serial%20I2C|Serial%20SPI|SPI&p4527=1000000|12600000000&ps3=0.4|40&d=sel|7|4737|3899|1747|4952|4392|336|4458|4129|4527|4382|165|4101|s3|s5)

I'm beginning to think that the best option for the DAC might be to use an [ADA5542A](https://www.analog.com/media/en/technical-documentation/data-sheets/AD5512A_5542A.pdf) along with a dual OpAmp. The first OpAmp stage uses the RefFb pin to give a buffered bipolar output with +-5V for a 5V ref. The second OpAmp gives a gain of 2 and does some filtering.

gkasprow, 2018-11-08:
For logging we can expose the CPU analog input pins. Would 12bit resolution be enough? We can place header like Zotino has so when needed, one can attach BNC-IDC adapter. This will come for free in terms of HW.
In terms of CPU we should limit number of different families and chips within the SInara HW.
So far we use TM4C (Cortex M4F, 120MHz, embedded Ethernet PHY) and STM32F407ZGT6 (Cortex M4, 168MHz, external Ethernet Wiznet chip but can replace it with RMII PHY to make @sbourdeauducq happier)

sbourdeauducq, 2018-11-09:
Sure, software TCP/IP is much better than limited, proprietary/obscure, inflexible, non-upgradeable, difficult-to-use and clumsy Wiznet chips.

dtcallcock, 2018-11-09:
> In terms of CPU we should limit number of different families and chips within the SInara HW.

For sure.  The STM32F407ZGT6 doesn't sound too different to the TM4C though.  @sbourdeauducq @jordens What Nucleo board were you planning to use with Humpback?  The top-end NUCLEO-[H743ZI](https://www.st.com/en/evaluation-tools/nucleo-h743zi.html) has a 400MHz Cortex M7.  Perhaps that is overkill though.

> If you want a general-purpose multi-channel logger then there are easier ways.

Like what?  There doesn't seem to be a cheap off the shelf solution (eg. an arduino shield) with a decent AFE.  The Labjack somewhat fits the bill but is overloaded with features and expensive for simple tasks.   My thought is that a general-purpose logging board for Sinara would pretty much look like what we are designing here if it just had a few more AINs on it.  To me they seem worth adding in order to extend the board's utility.  Also, like I said a lot of locks have a few auxiliary voltages one needs to monitor.  Right now many of our NIST lock boxes have comparator boards hanging off them to turn digital inputs into crude 1-bit DACs!  Any solution that involves a second board is going to be way less convenient than just adding a few more AINs.

> For logging we can expose the CPU analog input pins. Would 12bit resolution be enough? We can place header like Zotino has so when needed, one can attach BNC-IDC adapter. This will come for free in terms of HW.

I think this is a great idea.  12-bits would probably be fine.  If we used the STM32 off the H743ZI, it has 16-bit ADCs.  Some rudimentary protection and gain selection might be nice but probably need to pick the MCU first to see what we'd be dealing with.

> IIRC the G=1k option tends not to be that useful with a MSPS 16-bit ADC

> The other issue with the high gain settings is that the bandwidth is very low.

But some locks are pretty slow anyway so couldn't we just oversample and average?  In any case the two chips are pin-compatible and easily hand-solderable so if you end up needing the other one it's not a big deal.

hartytp, 2018-11-09:
> For sure. The STM32F407ZGT6 doesn't sound too different to the TM4C though. @sbourdeauducq @jordens What Nucleo board were you planning to use with Humpback? The top-end NUCLEO-H743ZI has a 400MHz Cortex M7. Perhaps that is overkill though.

Let's go for something nice and fast for this design. It's nice to be able to do a 2-channel MSPS feedback loop with a reasonably complex loop filter without having to be too careful about code optimizations (MCUs are much cheaper than good code).

> For logging we can expose the CPU analog input pins. Would 12bit resolution be enough? We can place header like Zotino has so when needed, one can attach BNC-IDC adapter. This will come for free in terms of HW.

> I think this is a great idea. 12-bits would probably be fine. If we used the STM32 off the H743ZI, it has 16-bit ADCs. Some rudimentary protection and gain selection might be nice but probably need to pick the MCU first to see what we'd be dealing with.

Okay, exposing some MCU ADCs + GPIO + power on headers sounds like a good plan. Probably one "analog" header compatible with the Zotino/Sampler header and one digital header. Add a few protection diodes if we think the MCU ones aren't beefy enough. The BNC breakout board has room for RC filters/dividers to give a bit of extra flexibility.

Also expose some DACs if the MCU provides them, otherwise for really simple cases, PWM + a second-order RC filter on the BNC board may be good enough.

I'm less keen on adding a multi-channel AFE (e.g. programmable gain) for the DACs/ADCs, as that starts to get more complex (it's not "free").

hartytp, 2018-11-09:
> But some locks are pretty slow anyway so couldn't we just oversample and average?

Sure, that will work in some cases. But, with a decent ADC, even on G=100 you're still a long way above the ADC noise floor IIRC. So, the G=1k doesn't really help much.

> In any case the two chips are pin-compatible and easily hand-solderable so if you end up needing the other one it's not a big deal.

Yes, this isn't a particularly big deal for me. I'm happy either way (whichever we chose will be ideal for some applications and bad for others, so I'm not convinced that there is an overall "best" answer).

hartytp, 2018-11-09:
Other comments:
- let's make sure that the headers supply decent quality +-12V analog rails with a reasonable amount of drive capacity
- let's add some mount holes to allow daughter boards to mount off this

One use-case I have in mind for this is the high-current stabilization board. I'd like to do a similar thing to NIST and strip our design down to essentially just being an AFE for the servo board.

gkasprow, 2018-11-09:
what about such plan:
For both thermostats we keep TM4C CPU because design is already routed and student is working on it.
Thorlabs thermostat has also Wiznet option on it so even without CPU code it will simply work.
We can fund student's work on version with external ADC and DAC with same user interface as Thorlabs version.
For Slow Servo we will use the fastest STM32 CPU possible. There will always be some speed bottleneck anyway so let's have some MIPS margin.

hartytp, 2018-11-09:
Sounds good to me.

hartytp, 2018-11-11:
Couple of other thoughts about this:
- let's have some FP status LEDs hooked up to the microprocessor (e.g. railed, loop active, etc)
- we should have a real-time (TTL) method for putting the feedback loop into "hold" mode (e.g. pause integrator and freeze DAC value). This should be doable by setting one of the LVDS lines on the EEM connector. Ideally we also want a run/hold TTL on the FP as well. If there isn't room for a BNC/SMA/etc for this on the FP then let's expose it via a GPIO header instead (then users can break it out with a separate adapter board when needed).

hartytp, 2018-11-11:
@dtcallcock also presented some DAC options for Fastino [here](
https://github.com/sinara-hw/meta/issues/12#issuecomment-415591811). 

For the output, let's use:
- either AD5542A or LTC2642 as the DAC. There's not much difference between them. I'd probably go for the AD5542A, but am happy with either.
- Followed by dual OpAmp to convert to bipolar (+-5V), provide G=2 (might as well use resistor network to avoid degrading temp co -- probably overkill for most use cases, but doesn't add a significant amount to the BOM cost), and provide a 2-pole active filter with 300kHz of analog BW. OpAmp choice isn't crucial, just don't degrade the DAC's noise/offset/slew rate too much.

dtcallcock, 2018-11-11:
So in terms of internal headers, we're looking at:

- Microcontroller JTAG

- Analog header. Probably 6x ADC, 2x DAC, and power rails.  Compatible with BNC-IDC.

- Digital header. 8x DIO and power rails. Compatible with BNC-IDC.

- EEM to Kasli.  8x DIO.

- EEM to EEM Accessory.  This could be the same physical connector as the EEM above or seperate.  If this was wired to one of the microcontroller's SPI ports, it could in principle control many of the EEMs that have already been developed.  The obvious one would be a PDH lock board if we develop one.

dtcallcock, 2018-11-11:
In the [discussion about thermostat](https://github.com/sinara-hw/Thermostat/issues/4) @jordens said:

> As mentioned elsewhere, the STM32F4xx have flight heritage, we already have explored ADC/DAC mod/demod and servoing code for them, tweaked quite a bit out of it, they are used on the satellite platforms we are working with, there is a lot of code already existing, the rust community for them is the liveliest, and finally the students here know them and like them.

Based on this experience can you give a sense of what the performance of a 2-channel feedback loop with reasonably complex loop filter is likely to be (including Ethernet communication overhead)?

Do you all/some of these comments apply to the STM32F7xx/STM32H7xx or only to the STM32F4xx?

hartytp, 2018-11-11:
> Digital header. 8x DIO and power rails. Compatible with BNC-IDC.

- We might want to actually put 16 single-ended DIO lines on that header. This can be made compatible with the BNC-DIO board by setting half of them LOW in software. This allows other use cases that don't need the BNC-DIO board to access more DIO lines.
- If possible it might be nice to have a SPI port on this connector

> EEM to EEM Accessory. This could be the same physical connector as the EEM above or seperate. If this was wired to one of the microcontroller's SPI ports, it could in principle control many of the EEMs that have already been developed. The obvious one would be a PDH lock board if we develop one.

So your idea here is to allow the EEM connector to be used as a downstream port, to control daughter cards?

dtcallcock, 2018-11-11:
> So your idea here is to allow the EEM connector to be used as a downstream port, to control daughter cards?

If we made a PDH mod/demod card that had a SPI-controlled DDS, VGA etc you'd presumably want to be able to control it from the servo board instead of being forced to include a Kasli in the setup?

This does mean the servo board would overlap a lot with Humpback.  If they end up using the same STM32 (via a Nucleo board in Humpback) that probably isn't a bad thing though.

hartytp, 2018-11-11:
> If we made a PDH mod/demod card that had a SPI-controlled DDS, VGA etc you'd presumably want to be able to control it from the servo board instead of being forced to include a Kasli in the setup?

I definitely agree that having enough GPIO on Stabiliser to allow it to control simple daughter boards would be a good thing.

However, I'm not convinced that downstream EEMs would be a good way of achieving that. 
- the LVDS interface eats a lot of power
- one usually requires chips additional chips to translate LVDS to LVCMOs. So, it's more convenient to just have a header with some LVCMOS GPIO pins directly connected.
- I don't think it's likely that one will end up using Stabiliser as a general-purpose EEM controller. That's probably better left to Kasli or Humpback.
- you then have to start adding EEM drivers to the servo firmware along with some kind of remote (ethernet) interface, which seems like a lot of work.

Also, FWIW, I'd be tempted to make the PDH bit a standalone PoE device rather than tying it to Stabiliser.
a. because one will often use something like a DLPro for the feedback element instead of stabilizer, so it needs to have the option of standalone operation/configration and PoE seems like a good way of doing that.
b. because I don't think people will want to alter the PDH settings in real-time, so ethernet control is probably more convenient than EEM
c. because if the PDH board is configured from Stabiliser then one has to add that functionality to the Stabiliser firmware. If we go down that route, we start having to have different firmware revisions for each application, rather than having a single, well-maintained version that just does the basics.

dtcallcock, 2018-11-12:
Yes, a separate ethernet connection also sounds feasible.  For the initial lock setup you definitely need to talk to both boards at the same time.  However, we will presumably need some kind of GUI running on a PC to tune the lock anyway and that would be the place to do that.

One _could_ imagine some gain to be had by having a servo that can tune the PDH in real time or some esoteric lock that needs frequency hopping or something.  This could probably still be accomplished over Ethernet though.

In any case that's discussing a different board so I made an issue for it here: [Pounder (PDH lock signal generator)](https://github.com/sinara-hw/meta/issues/28).

sbourdeauducq, 2018-11-12:
> For both thermostats we keep TM4C CPU because design is already routed and student is working on it.
> For Slow Servo we will use the fastest STM32 CPU possible. There will always be some speed bottleneck anyway so let's have some MIPS margin.

Sounds good, on the software side I don't think it's a big issue to have different ARM microcontrollers in Sinara - with tools like cargo and openocd, it should be manageable without much hassle.
I'm not very attached to the TM4C, its main advantage is the integrated Ethernet PHY, and it has downsides - it's your usual trashy silicon with a long errata (should not be an issue for Thermostat AFAICT), and depending where you buy it from, you might have to deal with annoying export control red tape due to the integrated encryption features.

jordens, 2018-11-12:
Depends on a lot of factors. We do lock-in processing of 16 samples, demodulation, 2nd-4th order IIR filtering for two channels in 1-3 µs at ~170 MHz with about plenty of cycles to spare. The rest (sampling, conversion, modulation, DMA, output, accurate timing) is done automatically. I think the other chips of those families are also fine as long as the technology is the same. The higher CPU speed might not be worth the decreased level of support, smaller community etc.

hartytp, 2018-11-12:
> Yes, a separate ethernet connection also sounds feasible. For the initial lock setup you definitely need to talk to both boards at the same time. However, we will presumably need some kind of GUI running on a PC to tune the lock anyway and that would be the place to do that.

Exactly, I'd imagine a simple python driver for both modules and some scripts to configure/monitor them, plot step responses etc.

> One could imagine some gain to be had by having a servo that can tune the PDH in real time or some esoteric lock that needs frequency hopping or something. This could probably still be accomplished over Ethernet though.

In the first instance, I'd try to do this via ethernet, which might be fast enough, depending on the details of the application.

In any case, I still feel that modulation of the PDH params in real time is a bit of a niche case. Frequency hopping can generally be done by modulating Stabilizer's parameters, without touching the PDH settings. Before we worry about such a niche use, it would be good to hear a concrete example of when it would be needed.

gkasprow, 2018-11-12:
I wouln't affraid of connecting EEM to microprocessors. There are low cost, low power LVDS transmitters and recievers, 0.1..0.2$ / channel. Examples are [FIN1027](http://www.onsemi.com/pub/Collateral/FIN1027A-D.pdf) and [FIN1018](https://www.fairchildsemi.com/datasheets/FI/FIN1018.pdf).  If this makes sense, dual interface solution won't increase the cost much. Ethernet PHY costs <1$, connector with  PoE is another 3..4$.

hartytp, 2018-11-15:
Okay, summarizing the above, the draft specification is:
- Size: ideally 3U, but don't be afraid to go to 6U if we need to use stacked BNCs (I'd rather have a wider front panel with room for fingers than try to cram everything into an overly tight space!)
- 2 analog inputs
  - +-1V to +-10V input ranges, depending on PGIA gain, 450kHz BW
  - Connector should ideally be BNC. If there is not room on the front panel then use SMA. The separation between BNCs/SMAs should not be tighter than Sampler/DIO_BNC/Urukul otherwise there won't be room for fingers.
  - Same input path as Sampler except...
  - ADC is LTC2310-16
  - **PGIA is AD8250** (unless @dtcallcock objects strongly, in which case use the same PGIA as Sampler)
- 2 analog outputs
  - +-10V output, 450kHz BW
  - DAC ADA5542A
  - use a 4.096V reference (needs a low output impedance reference or a ref buffer to drive the DAC). NB DAC multiplying BW is high, so use a filter on the reference to remove high-frequency noise.
  - 2 x OpAmps after DAC (can be two singles or a dual). OpAmp must not degrade slew rate of bandwidth of DAC. I'd suggest that we go for an AD8620. ADA4625-1 or ADA4627-1 would also probably work well here.
  - first OpAmp does reference subtraction, using DAC's data sheet connections. NB DAC output impedance is high, so need an OpAmp with low current noise/bias current.
  - second OpAmp does fixed 2.5xgain to give +-10V outputs and 2-pole active filter. Use same filter topology as Zotino, but tweak component values to increase the bandwidth to ~300kHz.
 - use a resistor ladder for the gain setting resistors to avoid degrading the DAC temp co. Also add DNF pads for 0603 gain setting resistors so that users can easily select any other gain they need.
 - Do not have an RC filter after the OpAmp. Just have a ~30R output resistor to isolate the OpAmp from load reactance.
  - **If people want, we can increase the output impedance to 500Ohm, so that users can achieve a +-1V output by externally 50Ohm terminating the board**. Personally, I don't think this is necessary as a default population option, and it means that cable capacitance can start to affect the circuit bandwidth, which is not ideal. Better to just allow users to put an external attenuator somewhere to achieve the voltage range they need.
- **CPU: highest speed STM32 that's conveniently available and well supported (@jordens @sbourdeauducq any preferences for exact models?). Probably something like STM32H743ZI (400MHz)
- PoE for power and slow/non-real time communications**
- EEM connector as a secondary source of power and communications. EEM connector should hook up to SPI, some interrupts and some GPIO on the microprocessor. Also hook up to the I2C bus with the EEPROM
- If there is room (I don't think there is) then add a barrel connector on the FP for +12V power. If there isn't room for this then don't worry, it's not crucial (PoE injectors can be purchased cheaply and used as a power supply).
- Front panel LEDs controlled by the microprocessor for power, status (lock enabled, lock not railed, etc)
- Front panel micro-USB for JTAG, routed to a native port on the microprocessor. 
- **We need some way of reading out the MAC address via USB. This can either be done with a USB-serial interface to the microprocessor, or by hooking up the I2C bus to USB. Maybe go for the same arrangement as Kasli with the FTDI quad chip unless someone has a better idea?** (we should make sure that something similar is implemented on Thermostat).
- EEPROM for MAC address and storing configuration data
- Analog header: 8 channels and +-13V supplies on a 100mil header. Compatible with zotino header. How about 4 ADC and 4 DAC? ADCs are connected to the microprocessor ADCs via a very basic single-pole RC RFI filter and, if necessary, protection diodes. If the microprocessor has DACs then use those, otherwise use PWM pins and put a 2-pole RC (2x1K||100nF) filter on the board (no output buffer since 2K DC output impedance should be fine). If necessary, add some protection diodes. 
- Digital header: some GPIO and also the 3V3 microprocessor supply. 16 single-ended GPIO if there are enough spare on the microprocessor. Ideally some interrupt lines and a SPI bus. Also I2C?
- Need to watch power sequencing of the analog sections to avoid exceeding the ratings of any component during turn on/off

hartytp, 2018-11-15:
Comments @gkasprow @dtcallcock @cjbe @klickverbot @jordens @sbourdeauducq? I'd like to freeze the design specification in the next day or so.

gkasprow, 2018-11-15:
U mean 4HP and 8 HP, not 3U and 6U :)

cjbe, 2018-11-15:
* If we cannot fit a +12V barrel connector on the front panel, it would be useful to have one on the back of the board (for when one is using this stand-alone).
* The main analog inputs and outputs should also be connected to a header, ideally with +-13V power

In several use cases one needs a specialized analog front-end. The connectors on Stabilizer should be positioned so that this board can sit in an adjacent rack slot and easily connect to the "main analog IO", "digital IO", and "misc analog IO" headers.

hartytp, 2018-11-15:
Thanks @gkasprow @cjbe. Post above updated to include your feedback

gkasprow, 2018-11-15:
We can place Ethernet and barrel connector on other side of board, together with barrel connector. Will we use Ethernet during operation in the crate?

gkasprow, 2018-11-15:
OK, so there was no question.

gkasprow, 2018-11-15:
In DIO BNC we have 18 mm distance between BNC connectors. In Stabiliser we can fit 4BNC + Ethernet but we would have to decrease the distance between BNCs to 17.4mm. Would that work for you? There would be one row of connectors so more place for fingers.

hartytp, 2018-11-15:
> OK, so there was no question.

What do you mean? I'm not sure I follow you.

> In DIO BNC we have 18 mm distance between BNC connectors. In Stabiliser we can fit 4BNC + Ethernet but we would have to decrease the distance between BNCs to 17.4mm. Would that work for you? There would be one row of connectors so more place for fingers.

Okay, if it's only a 0.6mm reduction in the BNC separation then that's okay.

hartytp, 2018-11-15:
Aah, okay. Yes, I want ethernet on the FP. Power on the rear as a fallback plan is okay.