sinara-hw/meta #28: [RFC] New EEM: Pounder (PDH lock signal generator)

dtcallcock, 2018-11-12:
[Pound Drever Hall](https://en.wikipedia.org/wiki/Pound–Drever–Hall_technique) locks are pretty ubiquitous in laser labs.  It would be nice to have a Sinara board that does all the necessary modulation and demodulation.  

It would sit alongside one of the proposed servo cards such as [Stabilizer (Microcontroller Servo)](https://github.com/sinara-hw/meta/issues/27) or [Fast Kasli Lock](
https://github.com/sinara-hw/meta/issues/26).

There are commercial models available from [Toptica](https://www.toptica.com/products/tunable-diode-lasers/laser-locking-electronics/pdd-110f-pound-drever-hall/), [Also Toptica](https://www.toptica.com/products/tunable-diode-lasers/laser-locking-electronics/pdhdlc-pro-pound-drever-hall/), [Sacher](https://www.sacher-laser.com/home/lab-equipment/locking_equipment/servo_controller/top_of_line_modulator_demodulator-lb20.html), [Qubig](http://www.qubig.com/rf-drivers.htm), and [Stable Laser Systems](http://stablelasers.com/products/pound-drever-hall-signal-unit/).  I haven't used most of them but none look like they quite do what we want.  At NIST we usually just roll our own from a two-channel standalone DDS, a pile of Minicircuits bits and some custom passive rf filters and a custom active LPF on the output. 

Interface could be EEM and/or Ethernet.  Having no Ethernet will be obnoxious if we don't want to put an EEM master on Stabilizer.  Having ethernet could also make the board on option for people who aren't otherwise using Sinara - the fact there are quite a few commercial products out there demonstrates there is at least some market.

dtcallcock, 2018-11-12:
We use [Minicircuits SYPD-1+](https://www.minicircuits.com/pdfs/SYPD-1+.pdf) phase detectors btw.  Does anyone have experience with something else?

hartytp, 2018-11-12:
FWIW, here is the analog board we designed in house for this. [pdhBox.pdf](https://github.com/sinara-hw/meta/files/2571431/pdhBox.pdf)

If I were to do it again, I'd go for:
- Eurocard
- PoE as the only interface
- 2-channel DDS (or 2xAD9910 with hw SYNC implemented, but that seems overkill)
- DDS 1 drives the modulator. Signal chain is the same as Urkul, without the RF switch (just switch the DDS amplitude to 0): pre-amp to give >=10dBm output (e.g. to drive a Qubig EOM); step-attenuator; very basic reconstruction filter
- DDS 2 is the LO for a mixer to demodulate (may need a pre-amp to hit the requried drive level). Pick any mixer with decent isolation/LO leakage that covers the bandwidth of interest (something like 10MHz-200MHz)
- After the mixer we want an OpAmp active filter (to prevent any LO leakage from being rectified in subsequent stages) + amplifier. I probably wouldn't make this have adjustable gain or set-point subtraction since the following stages usually do that. 
- We want to be able to hit a lock bandwidth of around 3MHz, so this stage needs something like 10MHz of BW to avoid eating up too much phase margin (users can configure for lower BW by population options if they need to drive their EO at a lower frequency for some reason)

So, all in all, I think it's really a very simple design...

> We use Minicircuits SYPD-1+ phase detectors btw. Does anyone have experience with something else?

IIRC, there is nothing special about those phase detectors, they're just mixers. Any decent (good isolation/LO leakage) will do just fine.

dtcallcock, 2018-11-12:
> something like 10MHz-200MHz

For [modulation transfer spectroscopy](http://www.qubig.com/fm-spectroscopy.htm) you want a modulation frequency of something like the natural linewidth.  This is ~1MHz in the [Iodine cells](https://arxiv.org/abs/1603.07416) that Mg+ and Be+ lasers are typically locked to, therefore I think we should aim for 1-200 MHz.

> Signal chain is the same as Urkul, without the RF switch

The balun and digital attenuator will work down to 1MHz but the 10nF DC block caps might need beefing up a bit.

>  pre-amp to give >=10dBm output (e.g. to drive a Qubig EOM)

I would probably like the option of a bit more power before having to resort to an external power amp.  The Minicircuits [PHA-13LN+](https://www.minicircuits.com/pdfs/PHA-13LN+.pdf) would give us a bit over 20dBm.

dhslichter, 2018-11-13:
I like all this, as well as having a beefier amp option.  Agreed that any old mixer will serve here.  Is the idea with this board that it will just produce an error signal output that another board uses?  Or would you put a microcontroller or such onboard and implement the digital servo as well?  

Another point with all of these servos -- ARTIQ is not naturally suited to tasks like live tweak-up of servo loop parameters, for example.  Is setting up PDH locks really something you want/need to do with ARTIQ, or is this better done as a simple "standalone" board of some sort (e.g. an Arduino shield) where you write some quick and dirty GUI and then set-and-forget?  Doing it in a Eurocard format is certainly appealing, but I am less convinced of the notion that full-on ARTIQ needs to be involved here.

sbourdeauducq, 2018-11-13:
If the board has Ethernet, both the GUI and the ARTIQ experiments could easily connect to it. It can be a Eurocard with Ethernet and standalone power (or PoE) support.
I would prefer something nicely done and not an Arduino shield with a "quick-and-dirty" GUI.

dhslichter, 2018-11-13:
I basically just want it to be useful for standalone operation, which would include something like non-volatile memory of its settings after power cycling, and perhaps the ability to do some of its own (lower-performance) digital servoing.  This would be ideal for something like iodine locks, which are slow.  My concern is that if you need to hook up to other boards in order to get the servo aspects, then you need to control *those* boards with a GUI, etc.  

So in all, the point I am trying to make is that I think it would be useful for this board to have a processor of some sort on board, as well as an analog output for a digital servo feedback signal in addition to just the raw demodulated error signal.  Even with ARTIQ integration you are very unlikely to be changing parameters like DDS phase or amplitude in anything approaching real time; you just want a rackable board that can provide you with an error signal output into a Stabilizer or whatever it's called at the moment.

hartytp, 2018-11-13:
> Is the idea with this board that it will just produce an error signal output that another board uses?

That's my thinking, yes. The board should output filtered mixer IF signal.

Leave the loop filter to another card, a DLPro, etc. 

> Is setting up PDH locks really something you want/need to do with ARTIQ, or is this better done as a simple "standalone" board of some sort (e.g. an Arduino shield) where you write some quick and dirty GUI and then set-and-forget?

That's my thinking too. I don't mind having an EEM connector here if people really want it, but the primary interface should be simple PoE IMHO. As you say, a configuration EEPROM to store settings would be a very good idea.

hartytp, 2018-11-13:
> I basically just want it to be useful for standalone operation, ... and perhaps the ability to do some of its own (lower-performance) digital servoing. This would be ideal for something like iodine locks, which are slow. My concern is that if you need to hook up to other boards in order to get the servo aspects, then you need to control those boards with a GUI, etc.

> So in all, the point I am trying to make is that I think it would be useful for this board to have a processor of some sort on board, as well as an analog output for a digital servo feedback signal in addition to just the raw demodulated error signal.

I'd really prefer to avoid putting the servo on this board. Let's try to keep each design simple, modular and cheap and let users stack them together to achieve what they need in their application. I really don't see any benefit in replicating hadware and software features on this board to do a worse job than Stabilizer at the servoing application. Plus, then you'd need an extra bunch of BNC connectors on the FP, making the board consume twice as much rack space.

This should be a very simple modulation/demodulation board that produces an input to a following loop filter stage. 

>  My concern is that if you need to hook up to other boards in order to get the servo aspects, then you need to control those boards with a GUI, etc.

Yes, stabilizer will need a GUI. But, that's something that will be handled in that project (and, anyway, once the driver is written, wiritng a basic GUI is a very quick job).

hartytp, 2018-11-13:
Potential alternative names in case "pounder" seems a little silly: "locker", "modem".

klickverbot, 2018-11-13:
Agreed with @hartytp on keeping the design modular – at ETH, we had a relatively simple two-channel FPGA-based PID controller, which enjoyed (well, presumably still does) great popularity precisely because you could use it for various kinds of laser locks (iodine/PDH/…), intensity stabilisation, etc. There is quite an advantage to having only one system to "learn", and improvements to the servo (such as the network streaming interface I ended up writing) end up trickling down to all the use cases.

dtcallcock, 2018-11-13:
> This is based on the 4-ch AD9959 (500MHz clock, 10-bit DAC). They make a 2-ch version, the AD9958 which is ~$40.

Actually given there is a 4channel DDS chip available why don't we make this 2 channel?

- It isn't much extra work (just copy and paste the analog section) and will probably only add a some 10s of $ to the board cost.
- Stabilizer is 2ch
- A lot of atomic references also have an AOM in the pump beam to offset it.  You could use one of the two LO outputs for this instead of a second PDH lock.  My guess it that this is why Qubig's box has an 'aux LO' output.

dtcallcock, 2018-11-13:
> IIRC, there is nothing special about those phase detectors, they're just mixers. Any decent (good isolation/LO leakage) will do just fine.

Yep.  The main reason we use them is that they are specified foor phase detection so you know what performance you will get rather than having to infer or test - for example with a random mixer you may end up with some annoying drifty dc offsets.  Minicircuits have [a nice application note on the subject](https://www.minicircuits.com/app/AN41-001.pdf).

dhslichter, 2018-11-13:
I am OK putting the servo on another board, the problem is that this board needs some non-volatile configuration stuff (setting DDS frequencies/amplitudes/phases, as well as IF gain perhaps).  If you want the servo board to control these things, that's fine.

hartytp, 2018-11-15:
> Yep. The main reason we use them is that they are specified for phase detection so you know what performance you will get rather than having to infer or test - for example with a random mixer you may end up with some annoying drifty dc offset. Minicircuits have a nice application note on the subject.

IIRC, the phase detectors are mainly specified with the RF input saturated, which is not what one generally wants for an atomic lock. With the RF input unsaturated there is less difference between a phase detector and a decent DBM. There is a good Marki app note on mixers as well. IIRC their conclusion is that isolation is a very good proxy for general mixer performance and, having done some really careful tests that use mixers as phase detectors, I've never had an issue with large offsets in mixers that have decent isolation/LO leakage etc.

Anyway, if there is a phase detector mixer that covers the frequency band, power range etc that we need then sure, let's go for it. If not, then let's just use a decent mixer. I'm equally happy either way.

hartytp, 2018-11-15:
Thinking about this a bit more, variable gain might be nice given the range of signal outputs we're potentially going to get. However, it would need to be implemented carefully to avoid reducing the lock bandwidth/adding noise etc. If it can be done without too much complexity/cost then great, but if not then it's not a disaster if we have to leave it out.

One option would be to put a low-noise DC-precise PGA on the error output path after the active filter (to avoid rectification of RF leakage through the mixer). However, to avoid degrading the lock bandwidth, we'd need something that has ~10MHz 3dB bandwidth, and I'm not aware of any PGAs that do that with decent gain.

Maybe the best approach is to put a digitally programmable VGA (https://www.analog.com/en/products/amplifiers/variable-gain-amplifiers/digital-control-vgas.html) at the RF input. Or, equivalently, a digital step attenuator followed by a fixed gain block. That way we ensure that we always have a known RF power reaching the mixer.

hartytp, 2018-11-15:
Could also implement an output PGA using a circuit along the lines of figure 67 in this data sheet: https://www.analog.com/media/en/technical-documentation/data-sheets/AD8610_8620.pdf

hartytp, 2018-11-16:
One other comment: we should definitely isolate the RF output, and maybe the RF input. Let's do this properly with a transformer, rather than relying on capacitors (which can break down and cause issues).

The motivation for this is when the laser modulation is achieved by modulating the laser current (which we do quite a bit). We've found that introducing any kind of ground loop on the laser current modulation kills the laser noise.

dtcallcock, 2018-11-17:
> Maybe the best approach is to put a digitally programmable VGA ... at the RF input.

Yes, I think an input VGA is a must.  

It looks like the [LT5554](https://www.analog.com/media/en/technical-documentation/data-sheets/5554f.pdf) would do the job.  It actually wants an isolation transformer on the input which as @hartytp mentions above, might be a good idea anyway.

The max gain is only 18dB though.  For some of our Iodine locks we have something like 70dB of gain on the input.  This might be a bit much to shoot for.  However it would be good to be able to swich in another 20dB on top of the VGA.  Perhaps we could do this with a pair of ganged-together SPDT switches around a 20dB gain block (or better yet a pair of SP3T switches with a 0dB, 20dB, and 40dB path in between them)?

One other feature request would be a directional coupler and log power detector just before the phase detector input.  This would be read out by whatever's easiest - probably the ethernet microcontroller's ADC.  As well as for initial optical alignment and VGA gain setting it'll be useful to be able to log this to see if it drifts or sags.

dtcallcock, 2018-11-17:
A filter between the input amplifier and the mixer can be useful for eliminating harmonics that cause dc offsets.

Out of the minicircuits range, the [HF1139 package](https://www.minicircuits.com/case_style/HF1139.pdf) has very good LPF and BPF coverage in the 1-200 MHz range.  Should we include this pad in the design (with default population being jumpers that bypass it)?

If we did that, it would be cool to make a little castelated PCB that matches the HF1139 footprint. It would look like these [pulse labs kits](https://www.pulseresearchlab.com/collections/signal-conditioning-kits) and would allow people to prototype and build their own filters.

dtcallcock, 2018-11-17:
> Could also implement an output PGA using a circuit along the lines of figure 67 in this data sheet...

That's a cool circuit.  We use something out of Horovitz  & Hill that's very similar (but with DIP switches instead of digital control).  It has two stages though and the other stage switches in and out capacitors to change the lowpass cutoff frequency.  I wonder if we could do something similar here unitor if the parasitics would kill performance out at the high frequency end?

hartytp, 2018-11-17:
@dtcallcock good points.

At this stage, it looks like the functionality we want from this board is pretty well mapped out. Now it's a matter of sorting out the details. Someone needs to sketch out a block diagram and start selecting components. We should also pin down things like the range of input signal levels we want to design for etc.

Do you want to have a go at that?

hartytp, 2018-11-17:
> Yes, I think an input VGA is a must.
> It looks like the LT5554 would do the job...The max gain is only 18dB though ... it would be good to be able to switch in another 20dB on top of the VGA. Perhaps we could do this with a pair of ganged-together SPDT switches around a 20dB gain block (or better yet a pair of SP3T switches with a 0dB, 20dB, and 40dB path in between them)?

Be careful using gain blocks or attenuators in switches. As soon as the gain/attenuation gets anywhere near the isolation of the switch (combined with any cross-talk on the PCB/supply lines) you'll start getting gnarly issues due to the amplified switch leakage coherently interfering with the other path.

Another way to do is to a 30dB digital step attenuator followed by a 30dB gain block. That gives you a 0-30dB PGA.

Anyway, conceptually this is all fine, someone just needs to think through the details and find a nice way of implementing it. Many ways of skinning a cat, just pick something that gets the job done without too much complexity. 

> For some of our Iodine locks we have something like 70dB of gain on the input.

Is this a niche use-case, or something that you think will be a common requirement for this board? If the former then it's probably easiest to just use an external MCL pre-packaged amp (to make life easier, we could expose a +5V supply on the FP via mini banana plugs to use to power that kind of amp). If the latter then we should try to support it on pounder -- although, let's make sure that we don't over-specify the requirements for this design, otherwise we'll end up with something that's massively over designed and over-complicated for the actual requirements.

hartytp, 2018-11-17:
> One other feature request would be a directional coupler and log power detector just before the phase detector input. This would be read out by whatever's easiest - probably the ethernet [...]

:+1: that can be implemented cheaply and easily, and is a nice feature to have.

hartytp, 2018-11-17:
> A filter between the input amplifier and the mixer can be useful for eliminating harmonics that cause dc offsets.

> Out of the minicircuits range, the HF1139 package has very good LPF and BPF coverage in the 1-200 MHz range. Should we include this pad in the design (with default population being jumpers that bypass it)?

> If we did that, it would be cool to make a little castelated PCB that matches the HF1139 footprint. It would look like these pulse labs kits and would allow people to prototype and build their own filters.

Pads for filters is always a good thing. Castellated modules might be a bit overkill for this. I'd just put a bare footprint for the filter on the board.

A few options here are:
- try to pick a default filter value that will work for "most cases" (if such a thing exists). Let users change the filter where required
- have a solder jumper to select between no filter (default population option) and a filter (DNPd by default). Let users add the filter when they need it
- slightly fancier, but...have a few different LPFs populated by default and use some RF switches to select between them. It's always nice not to have to solder anything.

Is it 2nd or 3rd harmonics you're worried about. If it's only 3rd and higher then the filtering is easier, so there is more scope to try to find a sensible default filter that will work for most users...

hartytp, 2018-11-17:
> That's a cool circuit. We use something out of Horovitz & Hill that's very similar (but with DIP switches instead of digital control). It has two stages though and the other stage switches in and [...]

Yes, if it can be done without killing the bandwidth then an amplifier after the mixer IF with switchable gain and switchable bandwidth would be really cool. This needs to be simulated (spice etc) to see if it will actually work. NB the filter needs to be quite fast since its main purpose is filtering out any LO leakage from the mixer to avoid rectification in subsequent stages -- the output of pounder goes into a loop filter box which should provide excellent tuneable filtering, so we don't need to recreate that here...

As always, we need to watch the complexity of the circuit and make sure that we don't overload it with low-priority features that are complex to implement.

dtcallcock, 2018-11-19:
> Be careful using gain blocks or attenuators in switches. As soon as the gain/attenuation gets anywhere near the isolation of the switch (combined with any cross-talk on the PCB/supply lines) you'll [...]

Good point.  You would need to use a gain block with a power down pin that you can connect to the switch line like a [TRF37D73](http://www.ti.com/lit/ds/symlink/trf37d73.pdf).

> Another way to do this is to a 30dB digital step attenuator followed by a 30dB gain block. That gives you a 0-30dB PGA.

Though you end up with a 30dB NF at 0dB gain.

> Is this a niche use-case, or something that you think will be a common requirement for this board?  If the former then it's probably easiest to just use an external MCL pre-packaged amp (to make [...]

I think it's reasonable to expect the first 20 or 30dB could come from the photodiode module or a minicircuits amp.  It would just be annoying if the board could only do 18dB and you had to add like 3 amplifiers.

hartytp, 2019-05-30:
@gkasprow if you have a masters student looking for a project at any point, this could be a really nice one. I was looking again at the COTS PDH boxes they use and just how bad they are. Slow, noisy, etc, but also lovely features like low-level parasitic oscillations at a few GHz that become very large if the amplitude knob is turned above 80% (the manufacturer suggested we place a LPF after the PDH box and forget about this issue, which about sums up their approach to engineering).

It wouldn't be hard for a student to build something significantly better than the units used in most labs. If done properly, I think it would be a popular product and a great Sinara addition...

gkasprow, 2019-05-30:
It seems we need a really good master student.
Do you think this should be something that is based on Stabilizer board, not an AFE but same PoE, CPU, etc, with EEM interface for operation in the crate?
What bandwidth? 10 or 200MHz?

dnadlinger, 2019-05-31:
> What bandwidth? 10 or 200MHz?

The 10 MHz – 200 MHz quoted by Tom above would be the range of LO frequencies of interest (possibly down to 1 MHz to cover saturation spectroscopy use cases).

If we want this to be a general-purpose PDH box rather than a dedicated Stabilizer add-on, the bandwidth of the signal chain after downconversion should be roughly 10 MHz to not completely take up the phase margin of a 2–3-ish MHz feedback loop (this with a faster controller than Stabilizer, of course, and at a suitably high LO frequency).

hartytp, 2019-05-31:
A couple of thoughts about this.

1. One usually has a band-stop filter (LPF costs too much phase margin) after the mixer to remove any LO leakage, which can otherwise get rectified by following stages, degrading the DC stability. Since we would like to support a wide range of LO frequencies, this will either need to be a fixed filter selected by users (probably the easiest option) or widely tunable. A SDR approach might be fun here. ie. a minimal fast AFE after the mixer and then rely on a fast ADC + DSP to do the filtering as well as the loop-filter
2. What about making this an AFE for Stabilizer? This simplifies the design a lot, allowing the student to just focus on the PDH modulation/demodulation path in the circuit design. When someone wants to lock faster than ~100kHz we can design the "fast servo" card we've been discussing as a mechanically/electrically compatible Stabilizer replacement. Basically, replace the microprocessor with an Artix-7 and use faster ADCs + DACs.

gkasprow, 2019-05-31:
I'm building a few circuits with SDR chips from ADI (AD9364). Would there be a use case for it?
I would rather build faster PHD box than use 2 versions.
It would be probably cheaper then AFE +stabilizer. 
We would maintain one code and one HW.

hartytp, 2019-05-31:
> I'm building a few circuits with SDR chips from ADI (AD9364). Would there be a use case for it?

That device is just the ADCs and DACs, right, it doesn't include the DSP/loop filter/etc, right?

> I would rather build faster PHD box than use 2 versions.

The suggestion was to have one version of the PDH circuit, but allow it to plug into different feedback controllers (microprocessor/FPGA/whatever).

gkasprow, 2019-05-31:
It's also up/down converter with quadrature modulators, gain control. Filtering is done in FPGA.
So it means we would have 2 versions of Stabilizer - slow and fast. This makes sense. 
But Kasli can also be equipped with EEM with fast ADC and DAC using just 2 or 3 EEM connectors do serve such purpose. Or, do you want to run multiple such modules from single Kasli and dedicated FPGA + fast DAC + fast ADC would make more sense?
It looks like there are a few approaches:
- we make mezzanine for Stabilizer 
- and we make FPGA version of Stabilizer with fast ADC/DAC and EEM & Ethernet + PoE. The same mezzanine could work with existing Stabilizer or fast stabilizer. Multiple such modules could be connected to Kasli using single EEM. We would use STM+small FPGA or even the smallest ZynQ.

Another approach is to use Kasli and EEM module with fast ADC and DAC that utilize 3 EEM connectors. 
The AFE form factor could be the same as in the case of Stabilizer. In this way, we could run up to 4 such cards from same Kasli.  But we lose the ability to run stand-alone and PoE supply. 

Yet another approach is a fully dedicated solution either as a stand-alone or Kasli extension.

 I don't know which approach is more useful.

hartytp, 2019-05-31:
Right: we need a fast (~MHz ABW) servo and this can be implemented either as a special-purpose board (e.g. fast Stabilizer allowing us to reuse AFEs) or by adding a FAST ADC and DAC EEM to Kasli. I'd prefer to go for the former approach with a special-purpose design. e.g. routing enough IO for 2 ADC and 2 DAC channels over EEM connectors will be a pain (presumably has to be a parallel bus, as SPI won't work at that speed).

So, I'd make this a Stabilizer AFE with the plan of moving to fast Stabilizer once that's available. But, maybe other differ?

hartytp, 2019-05-31:
> But Kasli can also be equipped with EEM with fast ADC and DAC using just 2 or 3 EEM connectors do serve such purpose

How do you do that?

gkasprow, 2019-05-31:
look [here](https://github.com/sinara-hw/meta/issues/26#issuecomment-435270657)

hartytp, 2019-05-31:
> But Kasli can also be equipped with EEM with fast ADC and DAC using just 2 or 3 EEM connectors do serve such purpose

Oh, I see, you mean using that SDR chip. 12-bit TX (ADC) parallel bus, 12-bit RX (DAC) parallel bus, clock and SPI. For 2xADC and 2XDAC you'd still need 50 LVDS lines for the parallel bus (6 ribbon cables) + SPI. And, you'd now have to split your parallel buses across different EEM connectors so have to get the delay matching right there, which seems a bit of a mess.

gkasprow, 2019-05-31:
Yes, but SDR chip is another option.
I'm talking about general purpose 12bit ADCs.
Recently I'm using fast ADCs that require 1 LVDS per channel + clock + sync
The ADS5296ARGC is an interesting example of ADC that needs 10LVDS to run 8 channels at 80MS/s or 4 channels at 160MHz. So we can run 4-channel ADC @80MS/s with just one LVDS.

hartytp, 2019-05-31:
Is there a DAC equivalent of that?

Anyway, my preference would be to put the DACs and FPGAs on the same PCB as the FPGA and just make a simple Stabilizer replacement.

hartytp, 2019-05-31:
So, to be clear, the proposal would be:
- make a PDH modulator/demodulator as a Stabilizer AFE
- ensure that the AFE can work at >1MHz lock bandwidth
- for now run with loops at ~100kHz lock bandwidth using Stabilizer
- when someone wants they make a faster version of stabilizer with FPGA, ADCs and DAC

gkasprow, 2019-05-31:
So this would be equivalent of [fast FPGA servo](https://github.com/sinara-hw/meta/issues/26). Would you like to keep the same mezzanine layout also for a faster version of it?

hartytp, 2019-05-31:
Right. My thinking is that the PDH card becomes and AFE for the Stabilizer, and we adjust the fast servo proposal to be Stabilizer-compatible.

I'd like to keep the same mezzanine layout if possible, so that we can swap mezzanines between the two designs. In an ideal world, both servo cards would have the same FP, PoE, etc and ideally Similar software interfaces etc

dtcallcock, 2019-05-31:
The AFE solution will be slightly more complicated and expensive for applications where you don't need the Stabilizer (ie. when you are using the build-in servo in the Toptica boxes).  However I think this is offset by the fact the reverse will be true when you do need Stabilizer or Fast Servo (probably the more common use case).

If this becomes an AFE:
- How much current is available on the +/-12 and 3.3V? 
- Is the full board area still available or do AFE's have to be smaller?

hartytp, 2019-05-31:
> The AFE solution will be slightly more complicated and expensive for applications where you don't need the Stabilizer

FWIW we'll likely make a break-out adapter for the AFEs e.g. to allow the SPI to be driven from an RJ45 etc. (i.e. a quick hack if you want to use them without Stabilizer).

> How much current is available on the +/-12 and 3.3V?

See the power budget, but there are a good few W IIRC. Otherwise, you can copy the current sense AFE I designed and stick a higher current SMPS on the AFE (I needed that to supply >10W to a flux-gate sensor).

> Is the full board area still available or do AFE's have to be smaller?

IIRC they can be the full size, although the stabilizer connectors are a mild PITA. The current sense release I posted has a STEP file which shows how the current sense AFE mounts onto Stabilizer, including all mechanics.

dtcallcock, 2019-05-31:
Ok, here's a first go at a block diagram.  If we are happy with this then I can break the components that need further discussion into separate issues.

[image]

gkasprow, 2019-06-01:
Let's focus on critical components - I need to order Altium libraries first.
I assume we need quad DDS DDS like AD9959. Or maybe four lower cost DDS chips?
Can we re-use preamps from Booster? ADL5536, PHA-1+
power amps: ERA-3SM+, ERA-4XSM+, NPTB00004A, MMWX1
Power detectors we use are : ADL5904 and AD8363ACPZ-WP
Attenuators: HMC542BLP4E
RF switch: HMC349ALP4CE
Low pass filter series: LFCN-3000+
Fixed attenuators: YAT-12+
modulators: ADL5375-05ACPZ-R7, ADL5380ACPZ-R7
Low noise generators: CCHD-950-25-100.000, Si570, Si549

The cost of RF components would be quite high so adding some control logic won't change the price much and we can consider it.

dtcallcock, 2019-06-01:
> I assume we need quad DDS DDS like AD9959. Or maybe four lower cost DDS chips?

The AD9959 is $39 in quantity.  The cheapest single-channel chip with comparable clock frequency is already $12.  So it think both on price and circuit complexity the quad chip is the way to go.

> Can we re-use preamps from Booster? ADL5536, PHA-1+

Not if we want the circuit to work down to 1MHz (these have 20 and 50 MHz low end respectively).

> power amps: ERA-3SM+, ERA-4XSM+,

These are a bit weedy for the power amp, but the ERA-3SM+ [Datasheet](https://www.minicircuits.com/pdfs/ERA-3SM+.pdf) should be fine for the pre-amp if the fixed attenuator before it was adjusted so that there was no way of accidentally damaging the power amp input.

> NPTB00004A

This seems a bit overkill.  I was thinking about the [Minicircuits TSS-13HLN+](https://www.minicircuits.com/pdfs/TSS-13HLN+.pdf) - 1MHz to 1 GHz and about 1/2 W output power at out frequencies.  It also has a shutdown pin so it can be connected to the control lines of the switches that switch it in and out of the circuit. This saves power and saves having to worry about having enough switch isolation.

> MMWX1

I'm not able to find this chip.

gkasprow, 2019-06-01:
> MMWX1
> 
> I'm not able to find this chip.

oh, copy-paste went wrong. I meant  NPA1003QA but it has min frequency of 20MHz so won't work.

That's true with the DDS chip.

dtcallcock, 2019-06-01:
> Power detectors we use are : ADL5904 and AD8363ACPZ-WP

Both seem reasonable but the AD8363 goes down to -58dBm whereas the ADL5904 only goes down to -30dBm so I'd probably go for the AD8363.

>Attenuators: HMC542BLP4E

Looks good.  This part is pretty battle-tested at this point right?

> RF switch: HMC349ALP4CE

This switch's power handling totally tanks below 100MHz- we need a switch that can handle what comes out of the power amp.  Will have to do a bit of research here.

hartytp, 2019-06-01:
1 what’s the use-case for the ext ref? Synchronised locks? If so is it worth having a ref out port?
2 what are the second two dds channels for?
3 is it really worth putting the PA here? Will require some thought about thermal management, biasing (cpu dac?) etc
4 Make the filters dnp by default but allow users to populate as required? Eg the input bpf totally depends on lo freq, the out filter probably needs to be band-stop rather than lpf to maintain phase margin
5 I’d like to make sure the output rf is transformer coupled and can be floated. Ground loops are death for diode current mod
6 fixed gain baseband amp after mixer?
7 expose baseband out on fp as well?

Not sure about 0.5W switch after pa. May need to use (solder?) jumpers

Should we move to a new repo?

hartytp, 2019-06-01:
>  It also has a shutdown pin so it can be connected to the control lines of the switches that switch it in and out of the circuit.

Unless you're starved for IO, I'd put the two switches, amp enable, SMPS enable ? on separate IO so you can ensure proper sequencing. Otherwise preventing races can bit a PITA in these designs.