sinara-hw/meta #31: RJ45 breakout

hartytp, 2019-01-10:
One thing that's come up a few times is the need for a really simple cheap RJ45 LVDS -> 4 TTL breakout. No need for isolation or anything fancy.

The use case I have in mind for this is putting an eval board in a box and talking to it from Kasli over SPI. Currently I often use 4xSMAs as a quick hack, but it would be much nicer to use the RJ45_DIO card + breakout.

dtcallcock, 2019-01-11:
I made something that does exactly this a while back: https://github.com/OregonIons/RJ45LVDS_SPI

Works pretty well.  We had one project where someone was trying to drive several eval boards in parallel over long ribbon cables with it and the level translators on it weren't beefy enough.  Also, the layout is a bit rubbish - should have used a 4-layer board.

gkasprow, 2019-01-11:
we can improve @dtcallcock idea and add USB supply for the driver as I did for https://github.com/sinara-hw/SATA_SFP/wiki
So the spec would be:
- shielded RJ45 
- 4 bi-dir LVDS drivers/receivers
- direction selected with DIP switch
- 4 SMA (or BNC?)
- fits some standard plastic box
- micro USB supply
- do we need programmable inversion?

gkasprow, 2019-01-11:
do we need support for 3.3 and 5V signaling?

dtcallcock, 2019-01-11:
> do we need programmable inversion?

I added a single inverter circuit so that one can make an inverted CS locally and drive two SPI boards from one RJ45.

> do we need support for 3.3 and 5V signalling?

It seemed easy enough to add given unknown future uses.  The translators I used also do 1.8 & 2.5V.

> - direction selected with DIP switch
> - 4 SMA (or BNC?)
> - fits some standard plastic box
> - micro USB supply

This idea of my little board was to build it into prototype boxes etc, as @hartytp describes.  That's why it's as small as possible, had pin headers, and hangs off of its panel mount RJ45 instead of having a case.  I think what you're describing here is quite a different use case - a general lab box.   This is separately a very useful thing and we have a NIST design that's basically what you are describing.  It fits into a case, has a 5V barrel connector for power, SMA in/out, and DIP switches for direction control.  @dhslichter did the design but he can't access it or post whilst the shutdown is on.  He'll post it once he's back online.  

> shielded RJ45 

We actually use unshielded RJ45 for these remote boxes in order to break ground loops.  Even if this isn't the default population, it'd be good to be able to break the ground on the board with a jumper.

gkasprow, 2019-01-11:
I'm not sure that LVDS is good choice to break the ground loop. The margin is very low - generally below 1V. LVDS is not designed to operate without ground reliably. Such operation can trigger logic circuit latchup. RS422/RS485 is a much better choice for such use case and was designed to work with large common voltage signal - at least 10V and without ground.
That's why I recommended to operate the LVDS DIO only with shielded STP cables.

gkasprow, 2019-01-11:
For such use case we can design DIO module with RS422/RS485 transceivers. Some models work up to 100Mbit and would be much better choice in breaking ground loops.

gkasprow, 2019-01-11:
With a bunch of [MAX22502](https://www.maximintegrated.com/en/products/interface/transceivers/MAX22502E.html) we could make simple DIO EEM module and matching remote BNC/SMA converter. No galvanic isolation would be necessary since RS485 alreasy has >10k impedance and +/- 15V common mode input with 5V supply.

hartytp, 2019-01-11:
> This idea of my little board was to build it into prototype boxes etc, as @hartytp describes. That's why it's as small as possible, had pin headers, and hangs off of its panel mount RJ45 instead of [...]

Exactly, there are lots of things one could do here, but the use-case I'm thinking of is a tiny, minimally complex RJ45->LVCMOS breakout that can panel mount.

I'd go for:
- pin header with 4-5 LVCMOS, some grounds and power
- as @dtcallcock suggests, the optional 5th LVCMOS could be an inverted CLK (LVDS0) but I don't really mind if this is left off
- connect all grounds together (easy to add an isolator on the LVCMOS if required)
- direction control via DIP switches (could even fix the directions if that makes thing easier)
- design to panel mount with the RJ45 protruding through the panel

dtcallcock, 2019-01-11:
> direction control via DIP switches (could even fix the directions if that makes thing easier)

I used solder jumpers as a half-way between these options.

dtcallcock, 2019-01-11:
> I'm not sure that LVDS is good choice to break the ground loop. The margin is very low

These boxes were designed to be used in a fairly compact experimental setup where the power supplies have a common ground.  We also needed compatibility with other LVDS signalling in our experimental control hardware.  

I can see that a more general purpose box designed to be floating on a USB wall wart would need the kind of ruggedisation you describe though.

dhslichter, 2019-02-16:
@hartytp @gkasprow returning to this issue post-shutdown, I have two variants of the board which are described above, which I will call the "pin header" and the "SMA" versions, for lack of better terminology.  

Both boards have:
- 4x LVDS via RJ45 converted to/from 3.3V LVCMOS-compatible logic
- DIP switch selectable line direction
- Buffered copies of signals on LVCMOS side (either inputs or outputs) available on pin header for logic probe use
- LVCMOS side is 5V tolerant 
- Runs from 5V supply on screw-locking barrel connector

The LVCMOS outputs are driven by a SN74BCT25244 (http://www.ti.com/product/SN74BCT25244) which has very high current source/sink capability capable of driving long, terminated 50-ohm coax.  This is perhaps overkill for some applications, but the chip is not expensive and we have found that even in applications where one is just running logic signals over a ribbon cable inside a box, some drivers with less "oomph" sometimes don't get the job done.  The outputs have an optional series termination resistor to eliminate multiple reflections on unterminated coax lines.  

Regarding using LVDS to break ground loops, it's not really the right way to do it (for that you want opto-isolators), but as long as you can guarantee that your ground is not varying by >1V across the lab (usually this is a pretty safe assumption; if it's not, you have bigger problems to fix!), it is within spec on the LVDS chips, and there will be several hundred kohms between the grounds of the LVDS driver and LVDS receiver, which is suitable for reducing ground loop currents.  Certainly much better than if you had a BNC or SMA connecting these grounds together.  

Schematic for the pin header version is here, SMA version is basically the same but with SMA connectors.  
[artiq_ttl_breakout_v2.2.pdf](https://github.com/sinara-hw/meta/files/2871364/artiq_ttl_breakout_v2.2.pdf)

gkasprow, 2019-02-18:
So, what shall we do here? Shall I port it to Altium and place it in some nice, low cost plastic box?

dhslichter, 2019-02-19:
Do you want the SMA version or the pin header version?

gkasprow, 2019-02-19:
So maybe we should make a dedicated repo, and if there is enough interest, make a small production run ?