sinara-hw/meta #33: MCX adapter for Sampler, BNC-DIO, IDC-BNC

gkasprow, 2019-03-07:
Technosystem volunteered to design a simple adapter that can be mounted instead of BNC connectors and utilize existing boards with 4HP panels. In this way, all modules already available with BNC connectors can be easily modified with MCX or other connectivity option.
Moreover, existing modules can be easily transformed by the users to 4HP by ordering kits containing adapter and 4 HP panel. In this way, it will be much easier to manage the stock of boards and offer various configuration options.
What do you think? @jordens @hartytp

hartytp, 2019-03-07:
Personally I'm not overly fussed about the BNC/MCX issue. I wouldn't expect us to bother modifying any of our BNC boards to use MCXs, and I'd be happy to keep buying BCN versions until stocks are exhausted.

jordens, 2019-03-07:
Same here. Bnc is fine until exhausted.
Do I get that right that this would be a reverse mezzanine with mcx and chokes and the bncs on sampler would not be pupulated?

gkasprow, 2019-03-19:
This will be tiny mezzanine soldered to the BNC pads,  with MCX connectors on it. This would fit Sampler, DIO, IDC-BNC. In this way, the modules could be ordered with MCX or BNC or converted later on.

gkasprow, 2019-06-22:
Here is the design
[image]
Sampler with MCX option

[image]
[image]

DIO with MCX
[image]

gkasprow, 2019-06-22:
and IDC version
[image]

dtcallcock, 2019-06-22:
It looks great.  What makes the connection between the main PCB and the adaptor though? 

How is isolation from the front panel achieved?  Just a gap around the MCX?

gkasprow, 2019-06-22:
> It looks great. What makes the connection between the main PCB and the adaptor though?
> 

pieces of wire soldered on both sides

> How is isolation from the front panel achieved? Just a gap around the MCX?

yes

gkasprow, 2019-06-22:
The boards are connected using 11 mm standoffs. The MCX would probably need the shink tube rings around.

hartytp, 2019-06-22:
Cool. Is the plan to release a version of sampler etc with he mcx on directly soon, or do you want to keep using these adapters for a while?

gkasprow, 2019-06-23:
I want to keep for a while. Clients still want BNC version.

gkasprow, 2019-06-23:
Such solution won't have an influence on the cost.

gkasprow, 2019-09-17:
I've sent it for fabrication.

closed 2019-09-30 by hartytp

jordens, 2021-05-17:
@gkasprow could you spin a repository for this and publish design files + release PDFs? Did the prototype work?

gkasprow, 2021-05-17:
Yes, it worked. I don't have photos with me but did not see any issues.