sinara-hw/meta #34: Clean up SFP wiki page

dtcallcock, 2019-03-19:
I want to buy a known good SFP 1000BASE-T GBIC for Kasli.  However the [SFP wiki page](https://github.com/sinara-hw/meta/wiki/SFP) is confusing.  There is one listed (fs.com p/n 34976) under the slightly cryptic heading of 'other configurations' - but that section seems to be referring only to Sayma v1 though (the section above implies it applies to all Sinara hardware).   There is also a part listed that specifically doesn't work but it's not clear which Sinara hardware it doesn't work with.  

Could people with working hardware take a moment to update this, or give me some part number and Sinara board combos I can add myself.

hartytp, 2019-03-19:
The part we've used with Kasli is FS-GB-GE-T which we've never had any issues with https://www.fs.com/uk/products/12626.html

hartytp, 2019-03-19:
@dtcallcock I think some of that SFP stuff relates to the SATA to SFP adapter that @gkasprow made. There were some problems with an early version not configuring things correctly so it only worked with certain SFP PHYs. I think that's all fixed now so any PHY should work. Kasli isn't fussy AFAICT.

hartytp, 2019-03-19:
I'd suggest we just scrap all the Sayma related stuff there and add a list of parts that people have used with Kasli, as that's currently the best place to test things.

dtcallcock, 2019-03-19:
Thanks.  I added a new table with that part.  Then I added a disclaimer about the old stuff but left it up there for now.  If someone (@jbqubit ?) remembers the details of that then they can move the relevant bits to the table and delete the rest.

jordens, 2019-06-13:
See also https://github.com/sinara-hw/Kasli/wiki#standalone After having testing this with many different modules, pretty much everything that is not an autonegotiating GBIC worked. At fs.com many parts are in fact the exact same device including the EEPROM contents. Others (like the GBICs and the 1000Base-X-U/D) only differ in EEPROM content.
I haven't done this with Sayma.