sinara-hw/meta #36: tunable microwave oscilaltors/filters/antennas

gkasprow, 2019-05-30:
Today I had a meeting with a nice lady from the Department of Chemistry WUT.
They are working with various ceramic materials and recently are developing voltage tunable flexible ceramic substrates. They are looking for interesting applications of these materials.
The substrate has interesting features
- one can tune its Er by more than 100% by applying HV - usually -150..+150V
- one can print traces and structures forming antennas, feed lines. They can be soldered later on.
- one can glue it to the Rogers PCB with pads that selectively tune Er of the circuit regions
- the tuning is very fast, they claim that faster than can measure. I want to characterize it at some point.
- it works from 0 to over 100GHz
- the losses tangent is < 0.0001 so one can build high-Q circuits.
I got samples of the material and they can print whatever we want if you have interesting applications. For me, obvious applications are tunable microwave patch antennas, filters and phase shifters for beamforming applications.
 Maybe in quantum applications, there are also potentially interesting use cases?

hartytp, 2019-05-30:
Cool! Not sure I have a use for it, but sounds like a fun technology to play with.

dhslichter, 2019-05-30:
Does sound very cool indeed!  Tunable filters and tunable impedance-matching circuits is what I think of primarily being of relevance for our types of applications.  One has to be a little careful because any tuning knob is also a way for noise to enter :)  But I'd love to hear more if you get a chance to characterize these things.  What is the material?  Do they have some papers on their work that you could share links to?

gkasprow, 2019-05-30:
[Here](http://sci-hub.tw/10.23919/EuMC.2013.6686746) is a very early paper with initial parameters they achieved 6 years ago. Now they offer it commercially.

gkasprow, 2019-05-30:
I will post more info once I get it from these people.

sbourdeauducq, 2019-05-31:
>     * the tuning is very fast, they claim that faster than can measure

Make a parametric amplifier?

jordens, 2019-06-13:
Tunable, vacuum compatible (or not), GHz resonators with tailored magnetic fields. That might be nice for anything from atom chips to ion traps and superconducting qubits.

gkasprow, 2019-06-13:
Do you have some specific requirements? Guys can make such one and ship for testing, for free.