sinara-hw/meta #37: [RFC] Driver: current driver for laser diodes

jordens, 2019-06-12:
We may be looking at designing a diode laser current driver in the future. Comments and secondments welcome.

* Form factor TBD. May end up being a Stabilizer mezzanine or a separate board that interfaces with Stabilizer or a separate EEM
* Either two versions for 200-250mA and 1.5-2A range or variants or one design that fits both
* Specifications and performance goals similar to Libbrecht-Hall//Libbrecht-Hall-Seck/Vescent/ILX Lightwave/Toptica etc
* A:
  * 250 mA
  * 2V compliance
  * single quadrant
  * assumptions/requirements w.r.t cathode grounded, anode grounded TBD
  * 0.5 µA resolution, digitally controlled
  * 10 kHz small signal bandwidth
  * RMS noise 10Hz-10MHz: 0.3 µA
  * RMS noise 10Hz-10kHz: 0.05 µA
  * Stability (<10Hz) TBD
  * Linearity/absolute accuracy: 5% TBC
  * tempco 5 µA/K
  * could be some coarse+fine additive or multiplicative hybrid solution to achieve resolution TBD
  * has a modulation/error signal input that is both sampled by an ADC (~100 kHz) and a high frequency bypass (10 kHz to 100 MHz) that goes onto a bias tee on the output. Or alternative approaches TBD
* B same but different:
  * 1.5A
  * 3 V compliance
  * bandwidth and resolution much less critical
  * noise requirement at a similar relative level as A

dtcallcock, 2019-06-18:
Out of interest will you make this compatible with Toptica laser heads (either pin compatible or via an appropriate adaptor cable)?

>  a high frequency bypass (10 kHz to 100 MHz) that goes onto a bias tee on the output. Or alternative approaches TBD

The Toptica lasers do the fast modulation and bias tee on a little PCB right in the laser head.  This lets them provide modulation ports as coax connectors on the laser head so that you don't need the main cable to be that high performance.  Seems a reasonable approach.

dtcallcock, 2019-06-18:
> Form factor TBD. May end up being a Stabilizer mezzanine or a separate board that interfaces with Stabilizer or a separate EEM

If we make this as a mezzanine and we also make the [piezo driver mezzanine](https://github.com/sinara-hw/Stabilizer/issues/20), should enough signals be passed through so they can be stacked?  This would presumably require finding suitable surface mount headers so they can be placed on both sides of the board without interfering.

jordens, 2019-06-19:
The target here is a bit more complicated than the toptica lasers but I don't see an issue with making a wiring harness for that assuming they didn't deviate from the old LMU style too much.
Whether you bias-tee in the driver or at the head is a choice. What is important is that this is in general not just a tee but a four-port element (wide band error signal in from e.g. a phase lock split into lowpass out to the controller, highpass out to the laser plus lowpass path from the controller to the laser). That makes this element as cumbersome to have in the head as in the driver.
Stacking boards more than two high is something I would avoid at great cost. The mechanics, the electrical issues and the pain to debug it easily are so annoying. I would prefer to have the piezo driver a separate two-port amplifier device, especially since Stabilizer already has a suitable output on the front panel. But we won't have a piezo anyway.

jbqubit, 2019-07-08:
@restelli may have comments based on his experience with the JQI roll.

gkasprow, 2019-07-08:
A long time ago I designed some laser driver. It had an interesting feature - I used depleted MOSFET to ensure the laser diode is shorted all the time during on/off transients, even with power off. The MOSFET was disabled when safe operation of the diode was ensured. We can use the same mechanism here.

gkasprow, 2019-07-08:
@jordens can you sketch how the coupling of the diode with modulation input and high-frequency bypass input could look like? Does it have to support grounded cathode and grounded anode? If so, maybe the design should be bipolar...

jordens, 2019-07-09:
Generally speaking it's only about feeding a modulation signal (some "error" signal) to the diode where the low frequency part (<100 kHz) is filtered by the diode current controller. I.e. the high frequencies go through unmodified (we might talk about adding just a VGA/VVA and a relay/switch in that path if possible) while the low frequencies go to the controller (ADC), which then has a low frequency output (DAC) that is combined with the high frequency part and feeds the diode. This analog bypass allows fast feedback and fast modulation of the diode while the current controller can both sense and steer out the low frequency error signal (potentially onto some other channels like diode temperature or VHBG) and at the same time have extremely good low frequency control of the current.

```
       --[LP]--<ADC-DAC>--[LP]--
      /                         \
MOD --                           -- LASER DIODE
      \                         /
       --[HP]--[VVA/VGA]--[SW]--
```

The modulation input is only HF coupled to the diode. It's grounding arrangement doesn't matter.
The diodes we are looking at are floating. I.e. the current controller can be any type. There might still be unequal capacitive impedance to ground so the feedpoint of the HF modulation component should probably still be selectable (anode or cathode).
For a general application we'd need the DC part to be configurable as AG or CG but in my experience that comes with quite some complexity. Similar complexity to making it bipolar.

sbourdeauducq, 2019-07-09:
> The diodes we are looking at are floating.

On what lasers?

jordens, 2019-07-09:
Micro-integrated ECDL/MOPAs

sbourdeauducq, 2019-07-09:
It would be more interesting if the current driver would also work with less expensive and niche options. Do more common designs also have floating diodes? The newfocus tunable lasers do, but I don't know about others.

jordens, 2019-07-09:
If the "less expensive and niche options" contribute to the development then they should step forward and do so. They need to convince users that they are better. If you are referring hobbyists, then this is probably not interesting for them. It's simply not what's funded.

gkasprow, 2019-07-09:
If we want bandwidth of 100MHz then the signal needs to be injected to the diode very close to it. It's hard to make a broadband matching circuit for such non-linear device. The diode resistance depends strongly on the bias current.

jordens, 2019-07-09:
Sure. That RF signal only needs a reasonable wideband match from 0.1-10 MHz. Between 10 and 100 MHz there would be "actual" narrowband synthetic modulation signals that don't need a particular match as they can be tuned in power and frequency. Just that above 10 MHz the filter should not attenuate intentionally.

dtcallcock, 2019-07-09:
With a simple bias-T you should be able to get ~GHz of bandwidth.  In the Toptica lasers this is a separate input to a well-matched DC-20MHz input.  I like this approach as you can use one input for your servo and the other for your modulation.

sbourdeauducq, 2019-07-09:
I know hobbyists are boring and broke, but there are proper academics with proper grants who still make their own lasers (or use commercial laser heads) and could buy the device. It would expand its market.

jordens, 2019-07-09:
It's certainly not nearly a flat response up to GHz. As mentioned, the stuff above 10 MHz (and arguable above 100 kHz) is not really within the scope of the current servo here. The laser head people need to account for that.

This is just that: a proper academic project with a proper grant that makes its own laser. Anybody (also without academic job and without money) is welcome to speak up and contribute good arguments, analysis, insight and work.

jordens, 2019-07-09:
To clarify the AG/CG question: The driver needs to be of one such type. Either driver type is compatible with floating diodes and the respective grounded type of diode. For the other diode type, we'd need a bipolar driver or change the design for the other polarity. That added complexity and design cost/risk/time (for bipolar or second design) would need to be offset by the users of the "less common" diode connection scheme. TA/PAs are typically common anode, while the MO diodes are more likely common cathode. In our case both are floating but the anode has larger ground capacitance but that will be checked later.

gkasprow, 2019-07-09:
Take into account EMI. If you drive a diode assembly with the anode connected with housing and grounded cathode, your device will act as an antenna. So, for RF we need to make sure that housing is always  grounded for RF signals.

jordens, 2019-07-09:
That's what I said in https://github.com/sinara-hw/meta/issues/37#issuecomment-509635714 as well and it applies to some extent to floating diodes as well.
But it's not that complex to add a mechanical switch to feed the HF into either cathode or anode. It is however complicated to design the driver for bipolar or switchable polarity.

gkasprow, 2019-07-09:
True, the bipolar driver is difficult. It's much easier to design positive and negative drivers separately.

jordens, 2019-07-09:
If this ends up being a single design that can do either grounding scheme, then I would put both drivers (or their final stages) into the same device (the positive on the anode and the negative on the cathode) and deactivate/bypass/ground the unused driver side. That way it's just pseudo-bipolar which actually matches the way the diode itself operates (the diode is not bipolar just the grounding scheme is).

gkasprow, 2019-07-09:
We can actually make a module with separate outputs for both types of lasers.

gkasprow, 2019-07-09:
Is Libbrecht-Hall-Durfree design sufficient for this application?
There is a room for improvement, one can replace the output MOSFET with FET + BJT output stage to further limit the noise. I'm not sure if we can really benefit from Stabilizer design. It's easier to place the control DAC directly at the current source opamp input. Stabilizer generates GND referenced signal which needs further conversion since we want supply-rail referenced voltage. I'd rather use 16bit DAC as in original design. It's easier than using the output of the Stabilizer. Do we need complex AFE as in Stabilizer?

restelli, 2019-07-09:
Neil Pisenti and I a few years ago designed a low noise laser driver for the JQI strontium lab. The novelty of the architecture is that is designed to be ground referenced and to use extremely low power yet maintaining a relatively high control bandwidth in closed loop configuration. Noise characterization and overall design are available in our RSI paper: https://arxiv.org/abs/1609.03607

The project is on Github: https://github.com/JQIamo/hv-piezo-driver

Several of these current controller are in use in the JQI strontium lab, and I will be happy to show them in action to @jqubit.

restelli, 2019-07-09:
The main limitation of the [JQI](https://arxiv.org/abs/1609.03607) architecture is that for high amplitude sweeps it has a small slew rate.

gkasprow, 2019-07-09:
@restelli it seems you posted to wrong issue. You probably meant [Zapper issue](https://github.com/sinara-hw/zapper/issues/1)

dnadlinger, 2019-07-09:
> one can replace the output MOSFET with FET + BJT output stage

I was about to mention this, having toyed around with a few current source topology ideas recently (no finished designs). Cascoding the FET with a fast BJT should improve the output resistance at higher frequencies, and hence reduce the wideband noise. Polarity-wise, this appears considerably easier when sinking current into ground (fast PNPs are disappearing rapidly, while you can e.g. get very fast and linear Si:Ge NPNs).

Might also be worth considering using a BJT after the op-amp instead of the FET for the lower 1/f noise (with bias current cancellation if necessary).

jordens, 2019-07-10:
@gkasprow The design is sufficient **if** it meets the spec ;)
I'd also go with a BJT. 16 bit current resolution won't be sufficient here. 0.5µA/250mA is 2e-6.

gkasprow, 2019-07-10:
If we need more than 16 bits, the Stabilizer is not sufficient anyway.
@jordens what about ADC front end spec?

gkasprow, 2019-07-10:
@jordens 
0.5µA/250mA is 500k which means we need 19bit DAC. There is no much choice and we would have to use expensive [AD5791](https://www.analog.com/en/products/ad5791.html) DAC

gkasprow, 2019-07-10:
[MAX5719](https://www.maximintegrated.com/en/products/analog/data-converters/digital-to-analog-converters/MAX5717.html) is much lower cost alternative, compatible with MAX542 pinout

jordens, 2019-07-10:
Maybe it is possible/easier to go for a hybrid solution with a coarse DAC and a fine one. As mentioned the requirements in terms of INL are extremely lenient. Only DNL and noise are important. It would depend on what the laser current needs to do exactly and whether the very high effective DNL when the coarse DAC switches would become a problem.

gkasprow, 2019-07-10:
The hybrid solution works only if both DACs have a true 20bit resolution (for example PWM). MAX5719 is 15$ so it is not worth to make hybrids.

jordens, 2019-07-10:
The hybrid solution would work just fine since there is no need for large dynamic range i.e. **iff** the coarse DAC can be held constant whenever high resolution is required. There are other constraints (drift, tempco etc) but they seem manageable.
Maybe any of the old audio DACs (24 bit, 192 kHz) would be interesting here.

gkasprow, 2019-07-10:
Audio DACs have large, uncompensated DC offset which drifts with temperature. They are usually not good for DC applications. But here it might work.

gkasprow, 2019-07-10:
Hybrid DACs may cause issues when you need to switch between sub-bands of the lower LSB DAC. Such a solution can be not monotonical.

hartytp, 2019-07-10:
We take the 2-DAC solution in our [B-field servo](https://github.com/OxfordIonTrapGroup/stabilizer_current_sense/releases/download/v1.0/stabilizer_current_sense.PDF) where it works really nicely.

jordens, 2019-07-10:
As I said, whether the hybrid DAC will work here depends on dynamic range over which DNL needs to be good. For a constant B-field servo that's trivial. But if the current is expected to change (i.e. locking the frequency on DFB/DBR diodes or faster-than-piezo ECDLs) then the range tends to be larger.
If the audio DAC is in a digital loop, the DC offset and tempco would not matter. Don't know whether that's the right approach though. Without a high resolution ADC (which would otherwise not be needed here) this probably wouldn't have the required passive stability.
"More research needed"...

gkasprow, 2019-07-10:
The cost of an additional circuit that would be needed to use such audio DAC or make hybrids + low-cost DAC will be similar or close to the cost of 20bit DAC.

gkasprow, 2019-07-10:
Next thing - I assume we will reuse supply, ADC, PoE, CPU from Stabilizer. From what kind of sources do you want to drive it?
What about the name for the board?

gkasprow, 2019-07-10:
We can make it compatible with Stabilizer mezzanines. In this way, the laser driver could be paired with the piezo driver.

gkasprow, 2019-07-10:
I sketched an idea of how the power stage could look like. By adding a few components we can make it common anode and common cathode compatible. PB1 is used to choose the right polarity. Of course, CPU needs to know that and invert the DAC data for negative polarity. When somebody connects the laser to wrong output nothing will happen because it will always have 0V potential.
1k:200R divider is used to limit losses on the current sense resistor.
Of course, an additional monitoring circuit needs to be added to measure real diode current.
The relay is used to short outputs to GND until the valid current setting is confirmed by the measurement. Once the 12V disappears, the relay immediately shorts the outputs before transients occur. It's hard to find depleted FETs these days which have low RDSon.
This is an initial idea and still, some work is needed.
[image]

sbourdeauducq, 2019-07-11:
Is a mechanical relay going to short-circuit the diode fast enough before a possible power-off current spike can do damage? What about solid state relays?
Note that SSRs - and MOSFETs - have leakage current which is often specified at "<1µA", so this complicates selection further. The LCB710 and the CPC1705Y look OK if it weren't for this leakage problem.

sbourdeauducq, 2019-07-11:
The mechanical relay does have advantages (including being a very common component)... for maximum safety, maybe it can be connected to the 12V input directly, then there is a diode/MOSFET + electrolytic that powers the rest of the circuit long enough (> response time of the relay) when power is removed?
A 470µF capacitor loaded at 250mA would drop 2.7V during the 5ms response time of a relay, so the numbers for this approach sound doable.

gkasprow, 2019-07-11:
The relay is shorted by default. So there won't be an issue during power on. I connected it to the primary 12V source, so when power disappears, the relay turns off first. Other rails have some capacitances that will keep the circuits powered for a while. I can also connect it to the power good signal - once it disappears, the relay will be off. Note the Zener diode in series with a standard diode. This circuit ensures quick off action of the relay. I can also add a circuit that will switch off relay once the main power drops to 11V or so.

gkasprow, 2019-07-11:
I wanted depleted MOSFETs because they are shorted by default. One can also use standard MOSFET and make sure they are on before the power stage supply is enabled. But relay seems a simpler approach.

gkasprow, 2019-07-11:
Oh, I didn't know that there are NC OptoMOS relays :) So we can use them, too. The relay would be much cheaper.

sbourdeauducq, 2019-07-11:
> Oh, I didn't know that there are NC OptoMOS relays :) So we can use them, too.

What about their "<1µA" leakage in off state?

> Note the Zener diode in series with a standard diode. This circuit ensures quick off action of the relay.

Noted, but in this configuration relays still take milliseconds to switch AFAIK.

gkasprow, 2019-07-11:
1uA leakage should be not a big issue. We don't care so much about current setting accuracy. I'm worried more about the noise distribution of this current.
It's not a big problem to ensure that during the off state the critical rails will be still there for tens of ms.

jordens, 2019-07-11:
* That relay is not the final safety element I had in mind. It's fine and it doesn't hurt. But this one doesn't protect the diode when there is no driver connected. The usual solution for safety is to have a board married to the laser head that is good and quick at arresting surges, ESD, protects against reverse polarity, and has a relay to short the diode when the driver is off or when nothing is connected. That board at the head would also do the filtering of the modulation. And since in our case there are a bunch of other signals coming out and into the head on micro-SMP connectprs (multiple TECs, photodetectors, digital signals, RTDs) that breakout board would include the functionality of (a) filtering modulation, (b) safety, (c) adapting from micro-SMP to whatever goes to Driver/Themostat/Stabilizer/Sampler etc. That board is also very head-specific and we could design it so that the current driver is compatible with other laser heads as well.
* I would also be careful with noise of the MOS in the diode path. The relays are fine to me. Just needs sequencing.
* There will need to be decent monitoring of actual laser diode current and voltage.
* As for the name, let's stick with the original proposal "Driver". There hasn't been any opposition to that. It is simple enough.
* ADC, PoE, CPU from stabilizer is fine (modulo the issues raised there). Maybe seize the opportunity to clean up the IDC connector layout. I don't think the HV amplifier as a mezzanine is the best solution. I would keep that separate.
* With the 1k:200 divider you are seeing a lot more of the noise of the IC22 opamps because you are running them at very small input voltages. Does that not hurt?
* Maybe it is less error prone to have independent relays on anode and cathode. That simplifies the question of how to connect the diode a bit and gets rid of the "third terminal" for the return current.
* I am really interested in the startup and shutdown behavior of that circuit. What do the glitches during relay open/close do?
* What would change between the low current and high current version? Just the references? Note that the high resolution is only required for the low current version.
* What do you mean with "what kind of sources"? For power sources the same options as for Stabilizer.

jordens, 2019-07-11:
@gkasprow Want to break it out into a repository? Then we can split the issue by topic.

Also make a note on the schematic that the polarity switch is not hot-switchable.

jordens, 2019-07-11:
A few more things:

* You are way ahead of the schedule. I think that's fine as long as you keep in mind that the actual specifications won't be settled within the next X months. That then may require another revision. But it's good to have this for the testbed.
* I don't get why you bias the references from the (switched) DAC rails and not from P5V0/N5V0 directly.
* Also I think the unused output stage (e.g. T10) will be driving the in-use reference buffer (IC43A) pretty hard if something is accidentally connected to the unused output (e.g. iout+ to iout-).

dnadlinger, 2019-07-11:
A few notes regarding noise: Let's take the "Variant A" spec from above; 250 mA, 10 kHz adjustment bandwidth, and a classic op-amp + transistor design.

For the low-frequency region, 50 nA rms in 10 kHz is, assuming a flat spectrum, 0.5 nA / rtHz. Since this is within the control bandwidth, this will be dominated by op-amp noise. Let's assume a total average voltage noise density of 2 nV / rtHz. (This is non-trivial; something like the LT6018 maybe.) We then need at least 4 Ω of sense resistance; let's say 4.7 Ω (burning 0.3 W in the process).

Next, for the wideband noise, 300 nA up to 10 MHz is 94 pA / rtHz. This is quote low too – for reference, the shot noise of 250 mA is 283 pA / rtHz, so we are targeting almost 10 dB below shot noise. With the above 4.7 Ω, this is 0.44 nV / rtHz – doable with a low-noise BJT, but not much besides that. (I wish those GaAs pHEMTs had lower 1/f noise.)

Thus, we are left with a low-noise op-amp steering a low-noise BJT current source, with control bandwidth aggressively limited to make use of the lower BJT noise. Something like this, perhaps?

[image]

This only addresses noise, not stability/temp co, of course; the latter really depends on how tight the spec is. To use FETs, we'd need maybe 5 or 10 times more sense resistance to get the noise down.

How does the current-to-linewidth transfer function look like for your lasers? There might be a simpler design…

jordens, 2019-07-11:
You are an avid student of both style, phraseology, and methods of sci.electronics.design... ;)
Thanks for the analysis!
Yes. This is challenging. For wide band you definitely want to be below shotnoise (i.e. not limited by the light). Also the existing designs do achieve that. E.g. sub 100 pA/rtHz for the Vescent and 280 pA/rtHz for Toptica. Maybe revise this once we have better specs from the laser guys but it'll definitely be challenging. Maybe we'll find out the 1/f corner can be a as high as 10 kHz but I'll have to check. Vescent and Toptica give you 50 nA in 10 Hz-100 kHz and 0.1 Hz-10 Hz each respectively.
And note the high current version is not nearly that sensitive to current noise.
If we are input voltage noise limited, there are opamps like the ADA4898.

jordens, 2019-07-11:
Typical sensitivity is 200 Mhz/mA.

jordens, 2019-07-11:
For the problem of converting between linewidth ("short term", "lorentzian", "wideband" and what not) and more meaningful noise spectrum, see [Domenico 2010](https://www.osapublishing.org/ao/abstract.cfm?uri=ao-49-25-4801).
For some good ECDL frequency noise spectrum data see [Toptica](https://www.toptica.com/products/tunable-diode-lasers/ecdl-dfb-lasers/dl-pro/), for some rather old and to-be-improved-upon versions of the modules we are looking at see [FBH](https://www.osapublishing.org/oe/abstract.cfm?uri=oe-22-7-7790).
For good laser designs frequency noise is typically just current driver noise limited between 10 Hz and 1 MHz. The better the driver, the better the laser.

dnadlinger, 2019-07-11:
> You are an avid student of style, phraseology, and methods of sci.electronics.design... ;)

Darn – I was hanging around there when designing that photodiode module indeed, but I didn't realise quite how badly that has apparently rubbed off. Phraseology, really? ;)

> Also the existing designs do achieve that.

Sure, but not all of them, and not with much headroom.

> E.g. sub 100 pA/rtHz for the Vescent (but that may or may not be true)

They are below SPICE gives 62 pA/rtHz above 100 kHz for the above sketch, which may or may not be realistic as well. 

> If we are input voltage noise limited, there are opamps like the ADA4898.

Sure – 60Ω at 300K is already 1 nV / rtHz, though, so as I'm sure you are well aware, things become a bit tricky there (hence the reference to total noise). We can always use higher rails and more emitter resistance for more leeway, of course, at the expense of more heat to deal with. A big inductor might help too, I suppose (given that the bandwidth is limited by the RF bypass anyway)?

Anyway, my point was mostly that the current source placeholders in Greg's sketch seemed an order of magnitude or so off the required noise floor. Thanks for the references as well – I'd found the FBH paper, but wasn't aware of the Domenico et al. one.

jordens, 2019-07-11:
> I wish those GaAs pHEMTs had lower 1/f noise.

That sounded like a direct quote from s.e.d.

I fully agree, the frontend needs a really detailed design and quantitative prediction.

restelli, 2019-07-11:
> @restelli it seems you posted to wrong issue. You probably meant [Zapper issue](https://github.com/sinara-hw/zapper/issues/1)

Thanks, sorry for the mistake.

gkasprow, 2019-07-11:
The current source is just a placeholder. It needs careful optimization. I wanted to discuss only the general concept of switched polarity. There is also decoupling missing, I wouldn't also connect reference with digital supply directly.
Of course, the switch cannot be used during run-time.
I added 1k:200 divider to limit the shot noise in the current sense resistor and also dissipated power. Of course, the balance between opamp noise and resistor shot noise is needed.
I asked about modulation signal source - do we need to implement complex high-performance AFE as in the Stabilizer?
Since we are going to severely limit the current source bandwidth, what about 1kHz small-signal bandwidth? Did you mean BW of ADC - CPU -DAC circuit?
I know that I'm ahead of schedule, I treat it as a challenge.

[30 more replies]