sinara-hw/meta #38: sinara-hw EEPROM

TheCakeIsAPi, 2019-07-01:
Most of the sinara-hw contains I2C EEPROM chips (eg 24AA02E48T-I on Urukul). Some questions. 

- Do Creotech & Technosystem program board hardware number and serial number into the EEPROM? 
- Is read/write for EEPROM via I2C supported by ARTIQ yet?

@TheCakeIsAPi and @jbqubit

gkasprow, 2019-07-01:
Technosystem does not program the chips.
Creotech is reading the unique ID during the tests, but AFAIK they don't program their either.  @kaolpr am I right?

jordens, 2019-07-01:
QUARTIQ programs the one on Kasli and has defined a data format that is also used by the persistent data storage on urukul.

gkasprow, 2019-07-02:
Is there easy to use tool that can be used in a production environment to write EEPROMs?

jordens, 2019-07-02:
Sure. We published our deployment tooling long ago. https://github.com/quartiq/kasli-i2c . And I guess Pawel has been using it. It would be great if creotech and Technosystem would contribute to test and production tooling as well.

kaolpr, 2019-07-03:
@gkasprow you're right. Creotech is only reading and storing ID during test procedure. We're not writing anything to the EEPROM.

jmatyas, 2020-05-13:
#  SINARA EEPROM contents guide

I took the liberty to prepare a short guide for Sinara hardware vendros on what should be put into modules' EEPROM. It's mainly a description of deployment tools @jordens published a while ago and it is backwards compatible with what has been flashed onto many Sinara devices so far. 

Each board (apart from Kasli) is equipped with one EEPROM per EEM connector. It might be a good idea to agree a common standard for its contents. 

## EEPROM Layout

| Bytes | Field | Description |
| ----: | ----: | ----------- |
| 4 | CRC32 |control sum|
| 2 | magic | 0x391E (control value) |
| 10 | name | human readable board name |
| 2 | board | board ID |
| 1 | data_rev | EEPROM data format rev. |
| 1 | major | board major rev. |
| 1 | minor | board minor rev. |
| 1 | variant | board variant (e.g. Urukul variant) |
| 1 | port | port number the board is connected to |
| 1 | vendor | vendor ID |
| 8 | vendor_data | e.g. board serial number |
| 16 | project_data ||
| 16 | user_data ||
| 64 | board_data ||
| 122 | padding | default value: b'\xff' each byte |
| 6 | EUI-48 | EEPROM EUI-48 |

EEPROM used in Sinara HW is 24AA02E48 - 2 Kbit memory with I2C interface. Because the upper half of the array is write-protected, the 122-bytes-wide padding is needed. The EEPROM contents should be interpreted as bytes following layout from the above table. Bytes are big-endian and unsigned int is of standard size of 4 (Python ">I" format identifier when converting Python values to C structs).

CRC32 is computed for values [4:end] (including padding and EUI-48). EEPROM EUI-48 has to be read from EEPROM and put into the data struct first, otherwise EEPROM readout won't result with valid values after flashing.

 ## BOARD IDs

| ID | BOARD NAME |
| --- | --- |
| 0 | invalid |
| 1 | VHDCI_Carrier |
| 2 | Sayma_RTM |
| 4 | Sayma_AMC |
| 5 | Metlino |
| 6 | Kasli |
| 7 | DIO_BNC |
| 8 | DIO_SMA |
| 9 | DIO_RJ45 |
| 10 | Urukul |
| 11 | Zotino |
| 12 | Novogorny |
| 13 | Sampler |
| 14 | Grabber |
| 15 | Mirny |
| 16 | Banker |
| 17 | Humpback |
| 18 | Stabilizer |
| 19 | Fastino |
| 20 | Phaser |

## VENDOR IDs

| ID  | VENDOR |
| --- | --- |
| 0 | invalid |
| 1 | Technosystem |
| 2 | Creotech |
| 3 | QUARTIQ |
| #... | ... |

## BOARD VARIANTS

| BOARD | VARIANTS |
| ----- | -------- |
| Urukul | AD9910 |
| | AD9912 |
| Phaser | Baseband |
| | Upconverter |

----
I wasn't able  to retrive information on what to place in the fileds _project_data, user_data or board_data_, co @jordens  could you specify if you had some specific information in mind for these fields, or perhaps you just reserved it for the 'future feature'?

kaolpr, 2020-05-13:
I'll add a few words of comment here. At WUT we'd like to work out an agreement with manufacturers to have uniform EEPROM contents. But before it happens we'd like to make sure that we all agree on the contents format. 
That's why @jmatyas was asked to go through @jordens codes and make a format description. When the format is agreed, he will make guidelines for manufacturers to apply.

jordens, 2020-05-13:
Let's not hijack closed issues. The top post is answered and resolved.
Since the place where the work was done is the repository I mentioned, please file the issue there.

closed 2020-05-13 by jordens