sinara-hw/meta #39: WR oscillator Si549 characterisation

WeiDaZhang, 2019-07-29:
The thermal character of the DCXO Si549 we use in WR implementation has drawn our attention.
It has been noticed the open-loop phase noise of the DCXO changes over different (thermal) environments.
More precisely, the phase noise slopes and interception points between slopes changes in the very close-in region 1Hz and sub 1Hz, when the thermal environment changes. These frequency regions are not specified in its datasheet.
 
Oven tests suggest pure temperature does not make a big difference even in 20C changes.
However, changes of airflow rate make obvious differences.
Heat sinks attached to the DCXO make close-in noise less relative to the airflow (or thermal environment), presumably because of added thermal mass.
[image]

The test PCB is not designed to strengthen the thermal grounding of the DCXO particularly.
The result suggests heavy coppering the area and vias are necessary, and potentially heat-sink mounting.
[image]

In general, PLL removes the close-in noise of a VCO. However, if the slope is steeper than f^-3,
careful design of loop filters is required.

hartytp, 2019-07-29:
Thanks @WeiDaZhang!

Adding a few details...

The original observation here was that the DCXO open-loop close-in phase noise on Weida's test FMC was a lot worse than previous measurements -- sufficiently bad that the 3rd-order loop filter couldn't bring the DCXO down to the reference (Wenzel oscillator) noise floor.

This appears to be due to thermal fluctuations -- static changes in the DCXO case temp don't change the noise floor (as one might suspect), but air currents do. Adding a heat sink also improves the noise performance (presumably since it increases the thermal mass).

On Weida's test FMC the thermal grounding of the DCXO wasn't good.

So, the main recommendations from this are:
- use a nice high-order loop filter, with the widest BW that one is able to without degrading the broad-band noise floor too far. This is why we use a 3rd order filter, compared with 2nd for standard wr (we could even go to higher-order filters eventually if necessary)
- be careful about thermal grounding of the DCXO (and also placement w.r.t. other thermal noise sources)
- ideally screen the DCXO from air currents as far as possible

hartytp, 2019-07-29:
One other comment: on the KC705, Weida found that an external DFF improved the broad-band noise floor by 6dB (we haven't measured the close-in performance yet).

This suggests that with an external DFF we can use a wider loop bandwidth, since the DDMTD noise is lower. That would make us significantly more robust to close-in noise.

We have to make a decision about how much we care about noise v complexity in designs like Kasli. But, if we want to get the most robust/lowest-noise performance, we should consider adding the external DFF.

dnadlinger, 2019-07-29:
Would a shielding can covering the clock recovery section have the side effect of addressing this too? (Extra heat sinking through vias or via a glued-on gap pad if necessary.)

hartytp, 2019-07-29:
> Would a shielding can covering the clock recovery section have the side effect of addressing this too? (Extra heat sinking through vias or via a glued-on gap pad if necessary.)

Most likely, yes. Although it's hard to find screening cans that fit over these DCXOs without being excessively bulky. Also, soldered-on cans have the disadvantage of being difficult to remove when rework needs doing, while the clip-on ones can be more bulky.

The external DFF has the advantage that it somewhat decouples the CDR from the FPGA since the phase detector is now external -- although, it's not clear if that actually helps much, since the transceiver CDR could still limit (we're planning to get some data on this relatively soon).

But, yes, anything that increases the DCXO thermal time-constant by a factor of a few makes this a non-issue.

hartytp, 2019-07-29:
@gkasprow I can only see the 549CBAC000112ABG available with a MOQ of 50 (https://octopart.com/search?q=%20549CBAC000112ABG). That's kind of annoying. Did it used to be available in smaller batches, or did Creotech buy a batch of 50 for Sayma? For now, should we consider switching to the LVDS part, which is available in singles (https://octopart.com/search?q=549BACB001937ABG%20)?

gkasprow, 2019-08-05:
@hartytp They are in [Mouser](https://pl.mouser.com/ProductDetail/Silicon-Labs/549CBAC000112ABG?qs=%2Fha2pyFadugDIWrN8b%252BECtIhB8O0ABK9SZMzqktbG3SX6igcaM4aaA%3D%3D). Octopart sometimes do not show correct stock.

hartytp, 2019-08-05:
@gkasprow 
1. That mouser link was one of the results on Octopart
2. The mouser link you posts shows a MOQ of 50

hartytp, 2019-08-05:
NB the spurs in that measurement are real (a product of the fractional-N PLLs inside the DCXO). However, when the feedback is enabled they get whitened out pretty effectively, giving a cleaner spectrum.

mattiarizzi, 2019-08-14:
Hello,
I did a lot of phase noise characterization on WR hardware, especially WR Switch.
On the following link you can find lot of data about DDMTD & Gigabit transceiver performance.
Document [1] is a more elegant document, discussing part of the data that you will find in [2] and [3]. 
Another suggestion on improving the local oscillator: use silicone resin. It fixed the issue on the WR switch (dow corning rtv 3140), or use enclosed vctcxo like dot050f.

cheers

[1] https://ieeexplore.ieee.org/document/8400550
[2] https://www.ohwr.org/project/wr-low-jitter/wikis/Documents/DDMTD-report
[3] https://www.ohwr.org/project/wr-low-jitter/wikis/Documents/Xilinx-Virtex-6-transceivers:-phase-noise-and-stability
[4] https://www.ohwr.org/project/white-rabbit/uploads/2fa1a438446fc6c85b4540faecf1017a/ISPCS2016-WRClockCharacteristics.pdf

hartytp, 2019-08-14:
Thanks for the links to the lovely work. I’d already seen most of that :)

Do you have any data on performance of the vco before/after potting? That matches up with our experience as well that thermal fluctuations can really kill close in noise. Did you try using a more aggressive loop filter?

hartytp, 2019-08-15:
@mattiarizzi Did you ever look at the phase noise of the CDR on an ultra-scale FPGA? I assume the flicker noise will be lower since one can get the recovered clock out of an ECL pin directly, but would be good to have some data.

Have you had a look at our proposed implementation for WR on Kasli/Sayma? Any advice/comments/feedback?

Like you, we've noticed that the temperature stability of the VCO is critical for close-in noise. We're hoping to fix that with a relatively high loop BW and an aggressive (>=3rd order) loop filter. Here, using an external DFF will help a little as the lower noise allows for a wide loop BW. But, if that doesn't work, then we may put some thermal shielding over the VCO for users who care about this. Any other feedback, or tips for the gateware/software implementation very welcome!

mattiarizzi, 2019-08-16:
Hello @hartytp , thanks for the feedback!
Yes, I tried with a bigger bandwidth, see document [4], Section VI.A VCO Noise Leaking.
Witha  bandwidth of 200Hz, you avoid the low-frequency phase noise induced by the VCTCXO. This is a value that is fine for the WR Switch but may vary with different designs.
This bad behavior is know by VCTCXO manufacturer. When I asked to some tech guy working from one of these manufacturers, he told me that they to phase noise plot with a cup over the vctcxo (ha!).
The reason is that the temperature sensor is attached to the metallic enclosure, and the feedback system is designed with the goal of keeping the frequency drift vs fast temperature change low. This impair Allan deviation as it amplify the thermal noise.
Recently (2017), Epson started to seal vctcxo with the so-called "DoubleSeal", specifically to avoid this behavior.
Below the picture with the VM53S3 (WR switch vctcxo) potted.
[http://oi68.tinypic.com/ri5b9w.jpg](url)

cheers

mattiarizzi, 2019-08-16:
>Did you ever look at the phase noise of the CDR on an ultra-scale FPGA? I assume the flicker noise will be lower since one can get the recovered clock out of an ECL pin directly, but would be good to have some data.

I measured the flicker phase noise of Virtex-6 from the general (LVDS) I/O pins and from the GTX pins, it was the same. They claim ECL but seems ECL-**compatible**, however it is made with CMOS process, not bipolar/hybrid.
But in document [1] I measured the flicker noise from the I/O pins, and it is equal/slightly better than in Virtex-6

hartytp, 2019-08-16:
> This bad behavior is know by VCTCXO manufacturer. When I asked to some tech guy working for one of these manufacturers, he told me that they do phase noise plots with a cup over the vctcxo (ha!).

Reminds me of that Jim Williams app note about measuring the noise on the LTC6655.

> The reason is that the temperature sensor is attached to the metallic enclosure, and the feedback system is designed with the goal of keeping the frequency drift vs fast temperature change as low as possible. This impair Allan deviation as it amplify the thermal noise.
Recently (2017), Epson started to seal vctcxo with the so-called "DoubleSeal", specifically to avoid this behavior.

FWIW, this doesn't seem to be necessary with the Si549. AFAICT the temperature sensitivity is relatively low. In our initial tests with eval boards we didn't see any excess noise due to temperature even with all the boards on a desk in normal office space. The issues started for us on @WeiDaZhang's new test FMC, which has much worse thermal grounding than the eval boards. Some lessons we learn the hard way :)

hartytp, 2019-08-16:
Did you think about using the Si549 for the new WR implementation. There are a few reasons I'm really keen on it...

1) For a given price point XOs tend to be much lower noise and higher stability than VCXOs. The architecture we use (similar to the SI5324 and kin) allows us to use an XO as our reference, with a microwave VCO locked to it with a high-bandwidth fractional-N loop. By tuning the loop parameters we can lock the output RF to our reference. This lets us inherit the phase noise/stability from the XO while still getting tenability. The result is a much better close-in phase noise performance for the price range. I suspect that this is also why we don't see such bad temperature effects -- XOs have much lower temp cos than VCOs typically, so the SI549 is better than even relatively expensive VCXOs.

2) As the Si549 can put out any frequency, it decouples the choice of (VC)XO from the output frequency we want. i.e. we don't need a 25MHz VCXO and then a separate PLL to generate our RF.

3) I always worry that the DAC + VCXO solution is potentially very noise sensitive as the tuning voltage is a DC signal. I worry that on something like Kasli even breathing on the grounding could cause nasty phase changes in the recovered clock. It seems much easier to use something like the Si549 where everything is digital and all the hard work is done by Si Labs (the package even has LDOs etc in so it's robust to supply noise, it's screened so quite EMI tolerant, etc). All-digital systems seem quite nice to me!

4) Overall component count is quite low for us. We replace DAC + VCO + synth with only the Si549. Just need 2xSi549 one fanout and a 3V3 LDO. AFAICT that's quite a bit more compact/cheaper than the WR implementations I've seen.

5) Not related to the SI549 but on the topic of our implementation: we use a 3rd order loop filter, which seems to make us more robust to low-frequency noise. Did you try doing that? Cheaper/easier than potting things in epoxy!

6) IIRC the tuning resolution of the Si549 is quite a bit better than the 16-bit DAC and VCO, so lower noise from that.

hartytp, 2019-08-16:
> I measured the flicker phase noise of Virtex-6 from the general (LVDS) I/O pins and from the GTX pins, it was the same. They claim ECL but seems ECL-compatible, made with CMOS process, not bipolar/hybrid.
In document [1] I measured the flicker noise from the I/O pins of Kintex US, and it is equal/slightly better than in Virtex-6. No data on the CDR, yet.

Interesting, so maybe the US MGT clock output pins aren't lower noise than the old-fashioned methods. That would be a shame. Will be good to see some data on this

WeiDaZhang, 2019-08-27:
A few tests have been performed on our WR implementation (KC705 + FMC with DCXOs, CLK fanout buffers, and D-FFs). The timing-temperature coefficient is measured.
A Sanyo OMT Oven is used for controlling the environment temperature.
Its specification suggests the temperature fluctuation is +/-0.3 degree C (with time).
The timing error is measured with an RF mixer, low-pass filter, and a 6-1/2 digit Multimeter.

Test performed at 50-55-45 degree C each one for 24 hours continuously.
All the components of the DDMTD are implemented on the FPGA, the DCXOs are on the FMC.
[image]

Another test performed at 50-45-55 degree C each one for 24 hours continuously.
The difference from the above set-up is that the "Sampling" registers in DDMTD are implemented with a pair of standalone D-FF which are outside FPGA, on the FMC.
[image]

The results suggest:
- Temperature coefficient is about 1.5ps/K without ext DFF, and
- 1.2ps/K with ext DFF
- with external DFF the WR is less sensitive to temperature, but not really much change.
-  The long term stability of the system is limited by the oven's stability, and is currently better than 1ps over hours without careful thermal control.
- These tests are excluding the CDR in the FPGA (REF signal is fed to non-MGT pins of the FPGA), temperature coefficients of the CDR will be measured in future tests.

hartytp, 2019-08-27:
That’s pretty cool data! Nice to see you getting well sub-ps stability over long periods even without particularly good thermal control! At that point the stability is starting to get towards the ~100fs stability of or measurement apparatus (and I suspect we can probably hit that floor with decent thermal control).

In practice it looks like the wr pll stability will should not be the limitation in any non-trivial setup (at that level even cable drift can quickly become an issue).

gkasprow, 2019-08-27:
That' really interesting data. Worth writing a dedicated article!
@WeiDaZhang Did you look at the RF mixer thermal stability?

hartytp, 2019-08-27:
> That' really interesting data. Worth writing a dedicated article!

Yes, the plan is to write this all up (ideally with data from Kasli v2.0). More on that later.

> @WeiDaZhang Did you look at the RF mixer thermal stability?

The configuration here is a Wenzel oscillator as the frequency reference, buffered/split using an ADCLK eval board. The mixer and reference/buffer are outside the oven, but not actively stabilised (@weidazhang correct me if I'm wrong about that).

Weida has some nice baseline data on the mixer stability, but IIRC it's around a couple of hundred fs pk-pk over 24 hours, probably limited as much by cabling as anything else. This baseline drift doesn't change much with temperature stabilisation.

WeiDaZhang, 2019-08-28:
The stability analysis of the temperature-controlled data is done in the form of modified Allan Deviation and plotted on sigma-delta graph.
[image]

- (Circle Mark) The two None-DUT Instrument curves show the measurement limit (the difference in between is about a factor of 2, we believe it is contributed by cables, fanout, power splitters).

- Other traces are the stability of the WR output clock measured against the reference signal

- (Dagger Mark) The reference signal inputs via MGTREFCLK pins and went through CPLL results the lowest stability

- (Star Mark) The reference signal inputs via non-MGT pins provides better stability, the environment temperature makes small difference

- (Diamond Mark) When the reference signal and the main DCXO are both sampled with external D-FF outside FPGA gives the best stability, and the result is limited by measurement set-up. 

- None of the measurement involves CDR

WeiDaZhang, 2019-08-28:
> @WeiDaZhang Did you look at the RF mixer thermal stability?
The RF mixer and the Wenzel are outside the oven, but the fanout is inside the oven (I was not able to squeeze too many cables through the hole on the top of the oven)
We will get the MGT link on and perform the tests again referencing CDR output.

mattiarizzi, 2019-08-28:
Hello,

@hartyp
>FWIW, this doesn't seem to be necessary with the Si549. AFAICT the temperature sensitivity is relatively low. 

Si549 is not temperature compensated, which is a good idea for a slave WR oscillator.

>Did you think about using the Si549 for the new WR implementation. There are a few reasons I'm really keen on it...

I can a have a look. I'm just worried about the behaviur of the phase when you change the frequency, on a classic VC(TC)XO it's perfectly smooth, but Si549 has a PLL. I could do a test around October.

>Not related to the SI549 but on the topic of our implementation: we use a 3rd order loop filter, which seems to make us more robust to low-frequency noise. Did you try doing that? Cheaper/easier than potting things in epoxy!

Interesting! Do you have a repository with the code? Thanks

hartytp, 2019-08-28:
> Si549 is not temperature compensated, which is a good idea for a slave WR oscillator.

It's extremely stable -- look at the data we've posted! What's the temp co of the VCXO you're using, and how does it compare to the Si549 anyway?

> I can a have a look. I'm just worried about the behaviur of the phase when you change the frequency, on a classic VC(TC)XO it's perfectly smooth, but Si549 has a PLL. I could do a test around October.

It's glitch free and designed to be used in these kinds of PLLs. Again, see all the closed-loop phase noise plots we've posted, which would show any phase glitches up.

> Interesting! Do you have a repository with the code? Thanks

Weida has some code he's written for our testing (I'm sure he can send it to you if you ask). M-Labs are writing cleaned up code to integrate into ARTIQ. See here https://github.com/m-labs/artiq/tree/wrpll/artiq/gateware/wrpll

WeiDaZhang, 2019-08-29:
[image]
Phase noise measured on WR implementations with different reference path are plotted. The Carrier frequency is 125MHz.
The reference paths are:

1. MGTREFCLK pins -> QPLL -> TXPLLREFCLK_DIV1 -> TXOUTCLK -> "USRCLK_SOURCE (MMCM)" -> DDMTD Register in FPGA (see ug476 Figure 3-28 on p150)

2. MGTREFCLK pins -> QPLL -> RXOUTCLKPMA -> RXOUTCLK -> "USRCLK_SOURCE (MMCM)" -> DDMTD Register in FPGA (see ug476 Figure 4-23 on p210)

3. CC pins -> DDMTD Register in FPGA

4. DDMTD Register (D-FF) outside FPGA -> SelectIO pins -> following DDMTD logics

The difference between 1), 2) may be contributed by "USRCLK_SOURCE (MMCM)".
In 1) the MMCM converts 100M to 125M, but in 2) it is more or less a 125M fanout.
The performance of the WR-Receiver (I think it is called Node officially) is generally limited by the C/QPLL and CDR facility in the FPGA.
Curve 3) is similar to the performance of the WR-Distributor (Grandmaster) reported by other groups.
For the WR-Distributor (Grandmaster), the limited seems to be the input stage. Therefore, with external D-FF curve 4) gives a better result than 3).

hartytp, 2019-08-29:
@weidazhang thanks for posting that.

Let me check I understand you:
- the purple curve (4) shows the noise floor of the ECL DFF you're using for the DDMTD. The noise floor there is around -100dBc/Hz. This is the noise floor we could achieve in "grand-master" mode with an external DFF
- the yellow curve shows the phase noise of the internal DFFs used for the DDMTD. This is around 10dB worse than the external DFF. The difference is presumably the LVDS buffers in the FPGA or something like that. This data is roughly the same as the "Grandmaster" data published by CERN
- (1) and (2) (blue and red) are noise plots that should correspond to satellite mode, which is what we mainly care about here. These show that the limitation in the phase noise for the receivers/satellites will be limited by the CDR/transceiver PLLs. Is this consistent with the Cern data? 
- I don't understand fully the difference between (1) and (2). Is the difference here that you only have a 100MHz reference for your tests, so in (1) you're using an extra PLL (MMCM?) inside the FPGA to convert that to 125MHz, and that PLL is degrading the phase noise?
- does the red curve correspond to the noise level we expect to measure on Kasli satellites in the final system?

- am I right in thinking that in satellites/receivers you don't see any improvement from the external DFF since the CDR/QPLL dominates the noise?

hartytp, 2019-08-29:
So, the final measurements are the phase stability + temp co in satellite mode (i.e. when the QPLL/CDR are used)

WeiDaZhang, 2019-08-29:
@hartytp 
> * the purple curve (4) shows the noise floor of the ECL DFF you're using for the DDMTD. The noise floor there is around -100dBc/Hz. This is the noise floor we could achieve in "grand-master" mode with an external DFF

Yes, ECL D-FF -100dBc/Hz at 125MHz
> * the yellow curve shows the phase noise of the internal DFFs used for the DDMTD. This is around 10dB worse than the external DFF. The difference is presumably the LVDS buffers in the FPGA or something like that. This data is roughly the same as the "Grandmaster" data published by CERN

Yes, Fig. 10 in [1], Fig. 8 and Fig. 18 in [2] suggests the additive noise or the DDMTD floor is ~-105dBc/Hz @10MHz, which is similar to our ~-90dBc/Hz @125MHz

> * (1) and (2) (blue and red) are noise plots that should correspond to satellite mode, which is what we mainly care about here. These show that the limitation in the phase noise for the receivers/satellites will be limited by the CDR/transceiver PLLs. Is this consistent with the Cern data?

(2) corresponding to satellite mode noise. The CERN results seem not very consistent (probably because I didn't understand fully), in Fig. 18 in [2] it is around -85dBc/Hz, but in Fig. 3.10 in [3], it seems about -105dBc/Hz. I need to carefully read again to tell.

> * I don't understand fully the difference between (1) and (2). Is the difference here that you only have a 100MHz reference for your tests, so in (1) you're using an extra PLL (MMCM?) inside the FPGA to convert that to 125MHz, and that PLL is degrading the phase noise?

We did a few tests before, without involving CDR, but just let the reference go through the MGT for simplicity. That is (1). I noticed there is an MMCM (Xilinx wizard adds it automatically) when I was tracing the noise source.
> * does the red curve correspond to the noise level we expect to measure on Kasli satellites in the final system?

Yes, more or less. I'll look into the MGT a bit to see which clock to lock gives better result.
> * am I right in thinking that in satellites/receivers you don't see any improvement from the external DFF since the CDR/QPLL dominates the noise?

You are right. Not a single dB.

[1] https://www.ohwr.org/project/wr-low-jitter/wikis/Documents/White-Rabbit-Switch-performance-in-Grandmaster-mode
[2] https://ieeexplore.ieee.org/stamp/stamp.jsp?tp=&arnumber=8400550
[3] https://www.ohwr.org/project/wr-low-jitter/wikis/Documents/Xilinx-Virtex-6-transceivers:-phase-noise-and-stability

hartytp, 2019-08-29:
Okay, thanks for explaining. If you find low noise ways of doing the clocking, make sure to discuss then with sebastien

hartytp, 2019-08-29:
Once you’re done with that, will you measure the long term stability with the CDR? After that I think we have all the data we need

WeiDaZhang, 2019-08-29:
> Once you’re done with that, will you measure the long term stability and temp co with the CDR? After that I think we have all the data we need

Sure.

WeiDaZhang, 2019-08-29:
> Okay, thanks for explaining. If you find low noise ways of doing the clocking, make sure to discuss then with sebastien

Bypassing the "USRCLK_SOURCE (MMCM)" resulting 3dB improvement to the red curve (curve 2).

hartytp, 2019-08-29:
okay, so the PLLs are all about the same noise contribution, rule is use as few PLLs as possible?

WeiDaZhang, 2019-09-02:
> okay, so the PLLs are all about the same noise contribution, rule is use as few PLLs as possible?

hartytp, 2019-09-02:
?

WeiDaZhang, 2019-09-02:
A "remote" link WR recovered clock has been tested. The phase noise measurement is plotted in the green curve.
[image]
The "remote" link is set up, though in the same FPGA, with Tx path of the transmitter driven with Wenzel, and the Rx is driven with Main DCXO. The MGT set up the link first, and the WR locks the Main DCXO to the RXOUTCLK without losing the MGT link.
The setup is sketched as follows.
[image]
@hartytp Sorry, I clicked the wrong button earlier ....

hartytp, 2019-09-02:
Thanks @weidazhang! Can you summarise how this relates to your previous findings? What are the recommendations about setting up the PLLs? How does this compare to the Cern results?

WeiDaZhang, 2019-09-03:
The stability of the "remote" link has been analysed and shown follows.
[image]
[image]
@hartytp I'll summarise them soon.

hartytp, 2019-09-03:
> @hartytp I'll summarise them soon.

:)

hartytp, 2019-09-04:
NB one fun thing down the line might be to play with the new fan controller on Kasli to do temperature feedback based on the FPGA's internal thermometer. If we keep the whole WR loop inside the FPGA (so no external DFF) we can probably take out almost all of the temperature coefficient that way.

WeiDaZhang, 2019-09-13:
> > A reason can be directly measuring phase noise of an internal signal in FPGA is not possible.
> 
> You mean they measure the phase noise inside the FPGA, while we measure outside and the buffers etc could add some noise?

[Rizzi 2018] measures phase noise in two ways - meter and DDMTD - on three different setups.
There is one measured inside FPGA with DDMTD. This one suggests flicker noise of the Rx CLK by WR should intercept 1Hz at -95dBc.
The other two (fig 14 and 18) are on WRPLL output with the meter.
The former shows the flicker intercept 1Hz at about -100dBc without obvious floor observed.
The latter shows a FLOOR at ~ -80dBc without an obvious flicker.
The latter also shows by adding the LJD the floor reduced to ~-95dBc without an obvious flicker.

If comparing the fig 14 and 18 without LJD, assume there is a flicker intercept 1Hz at -75dBc but the slope is distorted by the loop filter, there is a 20dB difference. I didn't find a clear explanation for this apart from the fiber distance is increased from 3m to 10km. Note the grandmaster output also downgraded by more than 20dB which can not be contributed by the fiber.

These differences are recovered by using LJD. So basically at the end, the result with LJD gets close to fig 14, and similar to our result.

WeiDaZhang, 2019-09-13:
> With reference to the stability plot here [#39 (comment)](https://github.com/sinara-hw/meta/issues/39#issuecomment-527547429) can you remind me which curve is which, and what the differences are. e.g. IIRC the worse curves are with the CDR and are presumably limited by the QPLL noise. etc

The worst one is not actually CDR, but Wenzel -> MGTREF -> QPLL -> TxCLKOUT -> DDMTD -> MainDCXO.
The second worst is with CDR. As shown in [BD](https://github.com/sinara-hw/meta/issues/39#issuecomment-527175568), it involves QPLL -> Tx -> 2.5Gbps on 3-inch SMA cables pair -> Rx -> CDR ->RxCLKOUT -> DDMTD -> MainDCXO.
Apart from the Rx and Tx are physically located in the same MGT channel in one FPGA, this is very close to the really use case.

WeiDaZhang, 2019-09-13:
> Also, IIRC you tried both the CPLL and QPLL. Conclusion was that both add about the same level of noise, so it's good to minimize the number of PLLs, but no particular recommendation for optimizing the noise was found.

The conclusion of QPLL and CPLL are more or less the same was not involving CDR, not a real use case, not a remote link. I'll have to check this.
The recommendation is, however, Xilinx automatically add an MMCM PLL after Rx CLK, this one is not good for reducing phase noise or instability.

WeiDaZhang, 2019-09-13:
> > Interestingly in the second article, a DFF is used at the output of FPGA generated 10MHz to improve the performance.
> 
> Can you clarify what this means? IIRC you had a nice BD of this.

[image]
[White rabbit switch performance in grandmaster mode](https://www.ohwr.org/project/wr-low-jitter/wikis/Documents/White-Rabbit-Switch-performance-in-Grandmaster-mode)

WeiDaZhang, 2019-09-16:
@mattiarizzi 
Hello
We have been analyzing our results against a few papers and reports of yours. 
They have been very helpful.
There is a bit unclear to us, and I'm wondering if you could provide some hints

In [[White Rabbit Clock Synchronization: Ultimate Limits on Close-In Phase Noise and Short-Term Stability Due to FPGA Implementation]](https://ieeexplore.ieee.org/stamp/stamp.jsp?tp=&arnumber=8400550), Chapter VII Fig. 18, 20dB improvement is gained by the LJD from the "Standard Setup". The LJD results basically reached GTX limits in Fig. 14 and DDMTD limits in Fig. 8.

Is this 20dB mainly benefited by "using an external board to perform a clock synthesis currently done inside the FPGA" as mentioned in [[White Rabbit Switch performance in Grandmaster mode]](https://www.ohwr.org/project/wr-low-jitter/wikis/Documents/White-Rabbit-Switch-performance-in-Grandmaster-mode)?

Generally speaking, avoiding using MMCMs or PLLs in FPGA, and replacing the frequency synthesis module by external PLLs provide the improvement. Is this what you mean?
Thanks.

mattiarizzi, 2019-09-19:
Hello,
sorry for the delay, tomorrow morning I will reply to @hartyp and @WeiDaZhang

WeiDaZhang, 2019-09-19:
@mattiarizzi no problem

mattiarizzi, 2019-09-20:
>(Si549) It's extremely stable -- look at the data we've posted! What's the temp co of the VCXO you're using, and how does it compare to the Si549 anyway?

Yes, it is. That's a good news. The VCTCXO mounted on WRS are more unstable (at ADEV = 1s) in free-running mode, due to the temperature compensation circuit.

>Yes, Fig. 10 in [1], Fig. 8 and Fig. 18 in [2] suggests the additive noise or the DDMTD floor is ~-105dBc/Hz @10MHz, which is similar to our ~-90dBc/Hz @125MHz

Yes, it is -105 dBc/Hz from 10Hz to a few kHz (10 MHz carrier).

>(2) corresponding to satellite mode noise. The CERN results seem not very consistent (probably because I didn't understand fully), in Fig. 18 in [2] it is around -85dBc/Hz, but in Fig. 3.10 in [3], it seems about -105dBc/Hz. I need to carefully read again to tell.

It's consistent. The -85dBc/hz is the DDMTD noise floor with a STANDARD WRS.
The STANDARD one use a MMCM to multiply the external 10MHz input clock, this MMCM has a lot of phase noise between 100kHz-1MHz. This phase noise is aliased into the DDMTD baseband because DDMTD is a sampled system.
See https://white-rabbit.web.cern.ch/documents/White_Rabbit_Clock_Characteristics.pdf and https://www.ohwr.org/project/wr-low-jitter/wikis/Documents/White-Rabbit-Switch-performance-in-Grandmaster-mode for more info

>Generally speaking, avoiding using MMCMs or PLLs in FPGA, and replacing the frequency synthesis module by external PLLs provide the improvement. Is this what you mean?
Thanks.

Yes. Avoid MMCM at all cost, in Virtex-6 they are implemented as delay locked loops, very noisy at high frequency offset

>Interestingly in the second article, a DFF is used at the output of FPGA generated 10MHz to improve the performance.

The DFF is mounted in any WRS, the DFF does not introduce more phase noise than what you find inside the board (this statement is valid from sub-hz to 100kHz).

About the DDMTD white PM noise floor: you can improve it with multiple DDMTDs inside the FPGA. Check https://www.ohwr.org/project/wr-low-jitter/wikis/Documents/DDMTD-report page 36, figure 36
However you cannot improve the flicker noise because it is introduced by the input clock buffers.

About Si549: the closed loop results are very interesting. Are you updating it at 7 kHz or with a downsampled signal?

jordens, 2019-09-20:
The aliasing in DMTD (both digital and analog) can be really nasty. It makes masers at NIST look worse than they are because it doesn't properly reject minor high frequency "issues".

WeiDaZhang, 2019-09-20:
@mattiarizzi Thanks.
I see. When DDMTD samples the MMCM output significant noise is aliased back.

WeiDaZhang, 2019-09-20:
A few more characterizing tests were done on "Remote" link with temperature-controlled environment.
[image]
Result suggests working temperature has an observable impact on MGT-WR-PLL implementation stability. 45~55 degrees C oven temperature change causes 6e-14s ~ 9e-14s @ 1s variation on modified Allan deviation (curves with square marks).

Another experiment confirms external D-FF (DDMTD input register) doesn't help to improve stability in the "remote" link setup. We are limited by CDR noise floor (white) and CDR flicker noise (f^-1).

hartytp, 2019-09-30:
We now have all the data I think we could possibly want for the prototypes.

closed 2019-09-30 by hartytp