sinara-hw/meta #41: Wiki: EEM Board Front Photos

drewrisinger, 2019-08-06:
It would be helpful to have photos of the front of each EEM in the Wiki (seen from outside the rack). The wiki seems to be the best place for speccing out a new Kasli system, but it's hard to get a good sense of all card IOs without also seeing the front panel (for power, clocking, etc).

When building a system based on these modules, images of the circuit board are a secondary concern, and primary concern is the IOs (obviously minus any analog switches or IOs residing on the board itself). Not saying that circuit board photos should be removed, just that it's just as important to have the front panels to understand each EEM's IO.

gkasprow, 2019-08-08:
Good idea!
@jordens don't you have HQ photos of the panels by chance?

jordens, 2019-08-08:
@gkasprow you can cut the frontal photo of the Sinara rack on the ARTIQ/Sinara poster that I gave you. It's not nearly a complete collection though. This should be done properly for all devices. Otherwise old low-quality and incomplete stuff just sits there and nobody takes care of it.
@drewrisinger Things like these will require some investment of time and money. If you are interested, contact me.

gkasprow, 2019-08-08:
@jordens can me send an original photo?

jordens, 2019-08-08:
Done. But again, it's going to be incomplete and outdated unless someone steps up.

sbourdeauducq, 2019-08-20:
[graphic.pdf](https://github.com/sinara-hw/meta/files/3521540/graphic.pdf)
Work in progress and will be part of a new ordering system on m-labs.hk.

gkasprow, 2019-08-20:
I extracted panel photos and they are available [here](https://www.dropbox.com/sh/ibpkmvff5kuahs2/AACwiob3X5PPDderlNJB3qdia?dl=0)