sinara-hw/meta #42: summer releases

hartytp, 2019-08-07:
@gkasprow several of the sinara designs have small issues (some of which have been open for quite a while). Do you think we can do a bit of bug fixing, with the aim of a new set of releases in the next couple of months?

gkasprow, 2019-08-08:
Yes, I plan to release new versions of most of the boards once I come back from the US.
What is the priority list?
I assume:
- Urukul
- Thermostat
- Stabilizer

hartytp, 2019-08-12:
Sounds like the right priority order.

IIRC there are a few other boards needing fixes as well. Probably best to just check through all the designs in sinara-hw and see what needs doing...

jbqubit, 2019-08-14:
Agreed that Urukul is the highest priority.

hartytp, 2019-08-16:
Would also be good to get a new stabilizer release soon.

hartytp, 2019-08-27:
@gkasprow are any boards ready for review yet?

closed 2019-12-02 by gkasprow