sinara-hw/meta #44: grant proposal

gkasprow, 2019-08-17:
I plan to apply for a grant, roughly 2M EUR, together with WUT and Creotech for development of HW and SW for quantum applications.
Apart from current R&D that takes place or is already more or less defined, what modules would you like to see in the near future?
We can only hire people. Outsourcing is limited to ~10% of the grant, so most of the work including software would have to happen in Poland.

sbourdeauducq, 2019-08-17:
> Outsourcing is limited to ~10% of the grant, most of the work including software would have to happen in Poland.

Things would be so much simpler if politicians weren't constantly playing games like this...

hartytp, 2019-08-17:
Put top priorities are fastino and kasli zynq. Pounder a nice to have.

Finding some way of getting money for
Someone to work for/with m-labs/quartiq on firmware/gateware is also v important.

New modules are fun but v important is also fixing bugs in existing modules (urukul phase sync) and funding design reviews.

gkasprow, 2019-08-17:
The grant will start in one year and finish in 4 years. Fastino, Pounder, Kasli ZynQ is one-year perspective. Note that we still have a grant for commercialization that will end in one year.
The grant will have two parts - research and commercialization. The second mean design reviews, building test benches, improving quality, etc. The first part is new projects that I need to specify for the next 4 years.

hartytp, 2019-08-17:
4 years...give me hundreds of channels of fastino on an asic plz

dhslichter, 2019-08-22:
For us, Kasli Zynq and Shuttler.  I would say the way to go for Shuttler, if you want a kickass research project, would be mixed-signal ASIC that takes in DRTIO in one port and outputs analog waveforms (don't need to be amplified) from DACs on the same IC.  Like Xilinx RFSoC but more channels and less money.

jordens, 2019-08-26:
If you want one killer project that will keep you busy for four years, I'd agree and comprehensively solve the hundred-electrode-challenge. In the mid-term there doesn't seem to be a way around such a device. It's the superset of Fastino, Shuttler, the potential Fastino-ASIC. There is a clear path of incremental/iterative development with many risks and obvious milestones. It covers lots of fields (boards, asics, vacuum, analog/digital, gateware, software). Plenty to talk about.

gkasprow, 2019-08-28:
You would need 25Gbit transceiver for that purpose. I'm not sure if that can work as a dummy serializer with CDR off.

gkasprow, 2019-08-28:
One could even try to run one or more quads quad with 16Gbit transceivers available in Kintex devices with inputs connected in parallel. In this way, one can achieve 64GHz or 128GHz effective sampling rate. The necessary skew could be achieved either using external delay lines or by tuning the sampling clock phase. With such an approach the jitter could be very low, an order of 20ps.

sbourdeauducq, 2019-08-28:
We can probably use the FPGA transceivers for that, though the latency will be higher than 10ns.

sbourdeauducq, 2019-08-28:
In theory they can, but that needs to be tested on devkit of course (and also the clock skew stability).

gkasprow, 2019-09-21:
The grant is almost in place.  We asked for money for RFSOC, Shuttler, and Stamper. Keep the finger crossed! WUT will have the first two years to finish them all.

closed 2019-09-21 by gkasprow

gkasprow, 2019-11-07:
We got it!

hartytp, 2019-11-07:
Amazing!!!

jbqubit, 2019-11-08:
Thank you for being so proactive about applying for grants to do interesting work.