sinara-hw/meta #45: Technosystem purchasing page link is broken

dnadlinger, 2019-09-19:
The TechnoSystem Sinara hardware sales page link at https://github.com/sinara-hw/meta/wiki/Purchasing currently leads to an empty page with invalid SSL certificate. (I've never tried to use it before, so no idea if it ever worked.)

drewrisinger, 2019-09-19:
I think it has been broken as of a few months ago, last time I tried it.

dnadlinger, 2019-09-19:
I've removed the broken link for now, as it seemed to suggest that the TechnoSystem hardware effort is in a much worse state than it actually is. An overview page would definitely be interesting for some users, though.

hartytp, 2019-09-20:
The TS link used to work. Last time I spoke to them (a week or so ago) they were in the process of making a new Sinara shopping page, which is expected to come live soon. For now, just email Pawel.

sbourdeauducq, 2019-09-22:
Soon (ETA sometime in November) there will be a nice system for ordering fully assembled crates from M-Labs and QUARTIQ.
[case -02-desktop.pdf](https://github.com/sinara-hw/meta/files/3639232/case.-02-desktop.pdf)
[case -02-mobile.pdf](https://github.com/sinara-hw/meta/files/3639233/case.-02-mobile.pdf)

dtcallcock, 2019-09-22:
Looks slick!

Would be good to have version numbers on the boards.

dhslichter, 2019-09-23:
I love it!  Definitely want to have board version numbers.  This will be nice.  Will there also be ordering of single boards, or will this still go through Technosystem or Creotech?

closed 2019-10-14 by jordens

dhslichter, 2019-10-16:
@marmeladapk any update on a technosystem ordering page?

hartytp, 2019-10-17:
TS are still working on their new ordering page. For you you need to email them