sinara-hw/meta #46: cryogenic nano-rack

gkasprow, 2019-10-04:
Yesterday we gave the talks about Sinara and Hypersat projects at the quantum technologies workshop in ESTEC ( ESA headquarters). I had a few interesting talks with several scientists and one of them proposed that we develop a modular approach for cryogenic electronics (4K and less).
He was interested in multi-channel DACs, amplifiers, drivers, generally, everything that limits the number of feedthroughs and transmission lines that go down the dilution refrigerator.
So, I'm looking for your opinion and trying to imagine how such a system could look like.

drewrisinger, 2019-10-04:
Context: I worked in a superconducting qubit lab for about a year before working trapped ions at UMD, and part of my work was analyzing feedthrough requirements. Most of my knowledge in this is a few years out of date.

In my opinion, that use-case of reduced feedthroughs from room temp -> 4K is best solved by superconducting electronics at 4K, which convert room-temperature control signals to tons of current bias lines, RF lines, etc. But this sort of problem is CRITICAL to continued superconducting progress, which means that there are lots of groups already looking at this problem (off the top of my head, probably QuTech Delft, Google, ETH Zurich, etc.).

Projections for fridge feedthroughs in the next few years is ~500-1k lines, whereas the number of lines needed scales like ~5x number of qubits (and higher if tunable couplers between qubits are used, which scales O(n^2)).

Off the top of my head, the challenges that a traditional EE faces when designing for cryo are: 
* COTS ICs performing differently than expected at cryo
* Power consumption of CMOS at cryo (cooling power at cryo is limited, and CMOS quickly eats up the cooling power)
* Getting signals through feedthroughs to 4K electronics tends to require attenuation on lines, and might mess with your signals
* Expertise with cryo systems for testing.
* Temperature cycles to cryo can break connectors.

Most of the electronics (minus switches, etc) I've seen working at cryo are custom-fabbed chips, which is probably out of scope for Sinara.

A few papers on cryo electronics off the top of my head: 
Delft Papers on Cryo CMOS from 2016 (read ~2 years ago): 
* https://ieeexplore.ieee.org/document/7838410
* https://dl.acm.org/citation.cfm?doid=2903150.2906828

More recent ones from a quick Arxiv search (unread): 
* https://arxiv.org/ftp/arxiv/papers/1811/1811.09987.pdf
* https://arxiv.org/abs/1908.07685
* https://arxiv.org/abs/1811.11497

dhslichter, 2019-10-23:
I would say in general that developing 4 K electronics is out of scope for Sinara, it's a very involved project with a lot of unknowns.  You would have to do extensive validation because COTS vendors don't test or spec at these temperatures, etc.  Other efforts are being started to work on these topics, I would wait for them to do some of the initial work before we consider entering, not least because nobody using Sinara hardware at present has these requirements.

gkasprow, 2019-11-21:
That's true. I want to ask for a grant, and another group would design and validate the technology. We would only integrate it with ARTIQ

gkasprow, 2019-12-02:
I'm curious about what approach would work for physicists.
Imagine such a situation:
The system consists of tiny, let's say  10x5mm ceramic tiles that would be placed on top of the custom board with spring-loaded contacts. Something like nowadays CPU sockets.
Each ceramic tile has more or less functionality of the EEM modules. They have contact pads on the bottom.
So we have multi-channel DACs, ADCs, IOs, SAWG, IQ modulators, etc. Some with the parallel buses, some with just SPI.
The ceramic tile holds the ASIC and some necessary decoupling caps.
We also have bigger tiles with the controller ASIC which is essentially sequencer, DRTIO, and SPI master.DRTIO would probably be simplified, without CDR, only digital logic due to the problems of CMOS analog circuits with cryogenic temperatures.
The carrier board is application-specific. The client specifies the system, the number of channels, interfaces and places the order. The client can also design its own carrier board. Then the company produces the board using the LTCC process. LTCC boards can be ordered almost as easily as PCBs. Another idea is the multi-layer Alumina PCB. One can also use ordinary PCB but it does not work well in high vacuum.
Then the tiles can be either soldered or mechanically fixed to the carrier board.
Another approach is to use IC dices and bonder. This is used in hybrid chip manufacturing. But it seems to be much more expensive and also the entry barrier is higher.
Planarly assembled ceramic tiles or structures take a lot of space. But they dissipate the heat easily. I don't know how much space we have in such cases. Maybe arranging them vertically would help...
One could imagine 0.3mm ZIF connectors. Such IC die could be mounted on tiny Kapton PCB and then inserted to the vertical ZIF connector. But heat transfer issues would have to be solved. I don't know how much space we have on Z-axis.

dhslichter, 2019-12-02:
I would categorize this as a super-neat idea, and probably the direction that control electronics will have to go in the future, but that we are just not at the stage where this is necessary yet.  I think there will have to be co-design with the ion trap/superconducting chip/whatever your quantum system is, and there are a lot of unanswered questions on the quantum side that will inform the design choices for the "nano-rack".  

A lot of the questions you are asking above @gkasprow are good ones, and the answers will be very much application specific.  In general, though, demonstrating in-vacuum, 4 K electronics of the sort you describe will be a worthwhile and notable result even if you don't couple it to a specific quantum hardware system right away.  

Some others are already working on this (sorry, paywalled):
https://ieeexplore.ieee.org/document/8662480

gkasprow, 2019-12-02:
We are at the planning stage. The development will take a few years. And I hope that in 5 years or so such a system would be needed. It's very easy to get money in Poland for such RnD. There is nearly no competition. My quantum instrumentation grant was competing with innovative car wash station :D. It got two points more in the ranking list.
Thanks for the link. I was at the IC design conference in Crakow a few weeks ago and saw presentations of a few simple circuits running in 4K. This encouraged me to look deeper into this field.

gkasprow, 2019-12-02:
The idea is to develop ASICs that are compatible with ARTIQ and remaining modular hardware. I do not plan to replace existing HW, but build compatible nano-modules that make scaling of the control systems easier.
It does not make much sense to miniaturize all modules. We would put to the vacuum only some of them.

dtcallcock, 2020-02-22:
The latest and greatest from Intel: https://newsroom.intel.com/wp-content/uploads/sites/11/2020/02/Intel-ISSCC-Horse-Ridge-Fact-Sheet.pdf

gkasprow, 2022-11-09:
Btw, we got funding from 2 sources.

gkasprow, 2023-08-09:
first cryo-ASIC is in tapeout process. It will arrive in September and we will make characterisation of primitives in 4K.