sinara-hw/meta #47: 3U PSU

dtcallcock, 2020-01-08:
From @gkasprow 

> Btw, the 3U PSU is under development. One can see some drawings here
> It is cooperation with CERN, we develop a low-cost alternative to COTS solutions.
> One is 230V, 400W
> Another is 18-72V, 200 or 400W. They fully conform with CPCIS including ORing and current sharing. The AC version has an IEC connector on the front panel. It also supports power delivery via a backplane connector. There is an option to produce it without any active circuits, just 12V.
> We can make a version with just DC jacks on the front panel and Molex connectors on the back panel if there is enough interest.

dtcallcock, 2020-01-08:
Hope you don't mind me moving this here as it is very interesting but orthogonal to thread it was in.

> We can make a version with just DC jacks on the front panel and Molex connectors on the back panel if there is enough interest.

Or just a little adaptor PCB that hangs off that CPCIS connector and breaks it out into a bunch of Molex?

drewrisinger, 2020-01-08:
I'm not familiar with CPCIS connectors. Can anyone post an image?

gkasprow, 2020-01-08:
[image]

dnadlinger, 2020-01-08:
Are there some public specs (yet)?

gkasprow, 2020-01-08:
The project is very new.
This is non-rad-hard version of [this project](https://ohwr.org/project/psu-rad-acdc-230v-12v5v-110w/wikis/home)
We need such PSU for a few projects:
- CERN DI/OT. This project is open source and includes Zynq US controller, PSU, crate, backplane, modules. We are making it to be fully compatible with Sinara ecosystem. It is complatible with CPCIS standard.
- industrial White Rabbit switches based on CPCIS, with redundant PSU, removable fan trays and SFP modules. They need both AC and 48V supply
- internal WUT project; we build CPCIS subsystem supplied from 24V DC
- CPCIS version of Sinara project
The repo is [here ](https://github.com/elhep/CPCIS-AC-PSU) and [here](https://github.com/elhep/CPCIS-DC-PSU). We will update the wiki and repo soon.

gkasprow, 2020-01-08:
I updated the specs
https://github.com/elhep/CPCIS-AC-PSU/wiki
https://github.com/elhep/CPCIS-DC-PSU/wiki

dhslichter, 2020-01-10:
Does it need major modifications to be able to run on 120 VAC, or just some component swaps?

gkasprow, 2020-02-15:
update: the mechanical part is finished, both AC and DC  PCBs are nearly done.
There will be another super-simple version with DC jack and Molex connectors for backplane-less use cases.

jordens, 2020-02-15:
Is ce compliance certification part of the roadmap?
(And a reminder: we wanted to do the emi/emc compliance measurements for a typical Sinara system as well)

gkasprow, 2020-02-20:
Yes, I have complete CE test equipment in my lab

hartytp, 2020-02-21:
While we're at it, it would be interested to measure the conducted emissions from Kasli.

gkasprow, 2020-02-21:
This we can do easily. What exacly would you like to measure - what configuration, how ground loops connected, what interfaces, etc.

jordens, 2020-02-21:
@gkasprow what about the preliminary EMTEST with a typical system (Kasli+backplane adapter, 3 DIO-BNC, Sampler. 2 Urukul, Zotino, Grabber, plus a couple IDC-BNC)? I know it's a lot of cables but I am sure this is worth it.

gkasprow, 2020-02-22:
Yes, I remember. So far did not have a chance to collect the required HW in one place. These are so hot products that never stay on the shelves longer than w few days