sinara-hw/meta #48: Aux Power Supply

dtcallcock, 2020-01-10:
In my experience, lab setups rapidly acquire an armada of bench PSUs supplying the same old low voltage DC rails for misc homebrew electronics boxes.  Generally these PSUs are bulky, expensive, have much higher current spec than required, and sometimes have knobs on the front that get accidentally knocked with bad consequences.

It would be nice if there was a simple 3U, 4HP card capable of producing common low-current DC voltage rail combinations.  This board is fairly simple so the main question is probably whether enough people would find it useful and purchase it to make it worth while.

### Inputs
12V - via front panel barrel, rear Molex, or EEM
PoE???

### Outputs
Voltage rail 1 - selectable between 3.3, 5 and 6V - 1A
Voltage rail 2 - selectable between 5, 9, 12, 15, 18 V - 500mA
Voltage rail 3 - selectable between  -5, -9, -12, -15, -18 V - 500mA

Rotary switches to select voltage on each rail independently.  

Ranges above are based on experience and number of options can be increased or decreased as people see fit.  Overall voltage range might need to be limited for simplicity/component availability reasons.

Voltage noise/ripple, tempco and line/load regulation TBC but in line with typical decent bench PSU. 

How should overcurrent be handled?  How about LED lights when spec current exceeded but rail continues to operate (with no promises on spec) until thermal limit exceeded?  Whether this is sensible will depend on component selection.

### Connectors
Banana? - Pretty universal but can probably only fit the minimal 4x on 4HP
M8-4? - Probably my preferred compact connector - could fit 4/5 on panel but could be annoying to break out.

### Indicators
Overcurrent LED for each rail
Thermal protection LED

Voltage indicators would probably take up too much space.  Probably better for the user to label them with a label maker after setting the voltages.

hartytp, 2020-01-11:
Decent supplies like this are always useful if properly designed.

M8-4 are probably fine so long as there is a good source of pig-tail connectors available (not equipment doesn't have banana plugs but rather bear pins/screw terminals so we just want something that can interface with that reliably).

dtcallcock, 2020-01-12:
> M8-4 are probably fine so long as there is a good source of pig-tail connectors available

It looks like they are readily available for ~$10.  Shielded versions are also available for ~$20.

[image]

A close up of a typical cable:

[image]

dtcallcock, 2020-01-12:
To give you an idea of the specs of the kind of triple output PSUs these would replace.  

**Keysight E3630A** - $699

+0-6 V, 2.5 A 
+0-20 V, 0.5 A 
–0-20 V, 0.5 A

Load and line regulation: < 0.01% + 2 mV
Ripple and Noise (20 Hz to 20 MHz): < 350 μVrms, < 1.5 mVpp
Temperature coefficient per C: < 0.02% + 1 mV
Output Drift: Less than 0.1% + 5 mV
Transient response time: < 50 μsec following a change in output current from full load to half load for output to recover within: 15 mV

jbqubit, 2020-01-15:
M8-4 are great. 

>Rotary switches to select voltage on each rail independently.

Switch to select coarse range is good. Addition of a trim pot would be helpful as some devices require wonky voltages. For example, ZFSWA2-63DR+ absolute min/max range is 3-5 V so operating at 4 V is desired. 

My wish would be for a high-power rail that can supply up to 30 V and 100 W (with higher startup current) for RF amplifiers. If the high-power rail also had a barrel connector output on front panel this same module could double as EEM +12V supply. 

Powering the module from AC line makes more sense to me. 

Here's a sample of components from one corner of my lab that I think are representative. 

|Item|Power|
|-----|-----|
|MiniCircuits ZFSWA2-63DR+|+5 V 10 mA|
|RF-Lambda RFSPSTA0218G|+5 V 50 mA, -5 V 50 mA|
|MiniCircuits ZVE-3W-183| +15 V 2.2 A (3 A at startup)|
|MiniCircuits ZHL-4W-422+| +28 V 3.0 A (6 A at startup)|
|MiniCircuits ZFSWA2-63DR+|+4 V 200 mA|
|MiniCircuits ZHL-1-2W+|+24 1.0 A|
|Wenzel quartz oscillator| +15 V 300 mA|
|Lucix DRO oscillator|+15 V 315 mA |
|Microwave Dynamics quartz oscillator | +15 V 630 mA, -5 V __ mA |

dtcallcock, 2020-01-15:
> operating at 4 V is desired.

I added it to the list in the top post.  IME, many 3.3 and 5 V circuits have an LDO onboard so 4 and 6 V are useful to have around.

> trim pot would be helpful

We could select voltages with pots.  My motivation for digital selection is in anticipation of an SMPS + LDO topology.  Using digitally selected rails allows the SMPS and LDO voltages to be separately selected for each voltage to give best noise and dissipation.  Adding a few more voltage options should be cheap.  I imagine @gkasprow has invented this wheel many times before though and may have a better idea.

> My wish would be for a high-power rail that can supply up to 30 V and 100 W (with higher startup current) for RF amplifiers. If the high-power rail also had a barrel connector output on front panel [...]

You are talking about a completely different module here (perhaps one based on #47).  I want a cheap 4HP board uses the +12V we have around as an input and produces some convenient low-current rails at the location they are needed.  Using something with an AC line connector and a bunch of thermal management in places it's not needed is exactly what I'm trying avoid.

gkasprow, 2020-01-15:
Maybe explore potential of the USB-C chargers? One can program voltage between 5 and 20V, current goes up to 5Amps. I tested [a simple solution](https://hackaday.io/project/20424-pd-buddy-sink) where one selects the desired voltage and current. It works like charm with all sources I managed to find including the power banks. If the charger does not manage to deliver suitable power, the power won't turn on. We can add tracking LDO and filters so this would be suitable for various high power amps. One can change the voltage and current by plugging it to the USB and sending a few commands. 

Programmable 3U is another story. It won't be a high power system, certainly not 100W. To limit the power consumption I'd use LDO + tracking SMPS + filters. Programming can be done with a rotary hex switch. We can even add [miniature volt/amper](https://www.amazon.com/MCIGICM-Voltage-2-5V-30V-Digital-Voltmeter/dp/B07Q2RQYPJ/ref=pd_sbs_60_t_0/136-1595395-8936847?_encoding=UTF8&pd_rd_i=B07Q2RQYPJ&pd_rd_r=940d0ce0-ab99-4dfa-aef6-8b02109f89e1&pd_rd_w=LbuY3&pd_rd_wg=VUvgd&pf_rd_p=5cfcfe89-300f-47d2-b1ad-a4e27203a02a&pf_rd_r=RY13HZF47QCPC4WCCZ42&psc=1&refRID=RY13HZF47QCPC4WCCZ42) meter on the panel.

dtcallcock, 2020-01-15:
> Programmable 3U is another story. It won't be a high power system, certainly not 100W. To limit the power consumption I'd use LDO + tracking SMPS + filters. Programming can be done with a rotary hex switch.

Sounds good.

> We can even add miniature volt/amper meter on the panel.

Those look a bit big for 4HP.  If the current meter is just to indicate how close to overcurrent you are then a little 4 LED current 'bar chart' for each channel could suffice?  First two could be green, the 3rd yellow and then the 4th could be red for overcurrent.  It would look well retro.

dtcallcock, 2020-01-15:
> explore potential of the USB-C chargers

This is pretty cool.  Is there really an advantage for us over those 12V supplies we now all now a big box of in our lab though?  Especially as we have POE on most of the more standalone modules too.

gkasprow, 2020-01-16:
If we use the hex switch, then the voltage is defined using 16 steps. Single overload LED is fine.
If we go for potentiometers, the display could be useful. I have 3-digit LED displays that have 15mm length so would fit within the 4HP panel. I saw voltmeters with such displays.

USB-C can go up to 100W and 20V, so can be an advantage for some RF power amps.

dtcallcock, 2020-01-21:
> If we use the hex switch, then the voltage is defined using 16 steps.

I can't think of many applications where this wouldn't give you enough resolution.  If you are more picky than that you are probably going to put nice references and LDOs down on your application board.

> USB-C can go up to 100W and 20V, 

Ah, I see - it allows you to make a compact 100W card because all the lossy voltage conversion is done for you back at the wall wart.  

> so can be an advantage for some RF power amps.

Unfortunately pretty much all of the connectorized Minicircuits high power amps need >24V.

gkasprow, 2020-01-21:
we could combine USB-C, boost and LDO and make it user selectable with hex-switch programmable voltage and current limit. But I'm not sure the standard lab bench supply is not a simpler approach for high power amp.

dtcallcock, 2020-01-21:
Yeah, once you actually need 100W+ it's hard to beat just using one of [these](https://www.rigolna.com/products/dc-power-loads/dp700/).  30V/5A or 50V/3A for $278.  They sell a rack-mount kit too.

michalgaska, 2020-06-22:
I have been designing a power supply. I want to use LT8362 as a DC/DC converter and LDO TPS7A3301 for a negative rail. For positive rails, I am going to use LDO TPS7A47-Q1. Each rail will have a separate DC/DC converter with LDO. 

What do you think about selected devices?
What type of output connector do you want?

Here is the simulation of negative rail with pre-tracking: 
LDO Vout - green curve, DC/DC Vout - blue curve
[image]

[image]

gkasprow, 2020-06-22:
We discussed the power connectors already. Didn't we choose the M8 ones?

gkasprow, 2020-06-22:
What are the output ripples?

gkasprow, 2020-06-22:
@dtcallcock how would you like to have the voltage rails distributed among the M8 connectors?
Should we use 3 identical connectors, wired in some way with all 3 rails each?

gkasprow, 2020-06-22:
@michalgaska think about current limit implemented in such way that it does not overheat the LDO

dtcallcock, 2020-06-22:
> I have been designing a power supply.

Great.  We should move this to its own repository so that we can sensibly break out separate issues.  We just need a name.  It seems like we've been giving lesser boards like this more functional names (eg. PSU_triple) instead of breaking out the Russian lakes or '-er' names?

> We discussed the power connectors already. Didn't we choose the M8 ones?

Yes, M8-4.

> Should we use 3 ... connectors, 

I think we should use as many as will fit on the panel sensibly, leaving room for indicators and the 12V input barrel connector.  Not sure how many that would be but I guessed at least 4?

> wired in some way with all 3 rails each?

Yes, all 3 rails and GND on each connector.  Should the body also be grounded in case people want to use shielded cables?  Is there any kind of pinout convention for these that we should adopt?

> What do you think about selected devices?

Those are nice LDOs that seem well suited for this task.  Their 1A rating seems reasonable.  I do not have much experience designing switching power supplies for low noise and ripple though so I defer to @gkasprow etc.

jbqubit, 2020-06-24:
Yes, the consensus was M8-4. Agree that this ought to graduate to its own repository. How about calling it "AUX_PSU"?