sinara-hw/meta #49: OSA quantum 2.0 conference

gkasprow, 2020-01-12:
I submitted an abstract for [OSA quantum conference](https://www.osa.org/en-us/meetings/topical_meetings/quantum/). Please check the [paper](https://www.dropbox.com/s/g9u4tcenul033x1/ARTIQ%20and%20Sinara_G.Kasprowicz_WUT.pdf?dl=0) and put your comments directly on my dropbox

dtcallcock, 2020-01-14:
Thanks for doing this Greg.

I added some comments that you can take or leave.  Dropbox pdf commenting interface is a bit buggy though.

Submission deadline is 19 Feb if anyone else has comments.

gkasprow, 2020-01-15:
We plan to bring the Sinara HW and make the live demo :)

closed 2020-02-15 by gkasprow

dtcallcock, 2020-07-28:
Are you still presenting at the delayed and now virtual version of this conference?

Btw, there is a new IEEE quantum thing happening with a 1 day workshop on ion trap engineering & control hardware if you are interested.  https://qce.quantum.ieee.org/engineering-trapped-ion-quantum-computers/

Annoyingly, they haven't waived the registration fee like DAMOP and OSA Quantum 2.0 did though (which seems like a no-brainer if you are trying to launch something new).

jbqubit, 2020-07-29:
[Fee](https://qce.quantum.ieee.org/registration-fees/) for the IEEE virtual workshop is $470!

dtcallcock, 2020-07-29:
Apparently this is half price!  Kind of crazy it's so high given the level of corporate sponsorship (they have random AD and Xilinx reps speaking instead of people like Greg) and the fact they are trying to launch a new thing into a crowded marketplace where their competition is free/cheap. 

Anyway, I'll probably suck it up and register.  I am happy to (virtually) take a Sinara poster, represent, and report back anything of interest to us.  Sound like a plan @gkasprow or would you rather do it yourself?

dhslichter, 2020-07-30:
I think the NIST Ions will be out in reasonable force at the IEEE workshop as well....