sinara-hw/meta #50: [RFC] simple power amplifier

jordens, 2020-01-15:
We have an AOM power amplifier application where [Booster](https://github.com/sinara-hw/Booster/wiki) appears overkill (manufacturing/mechanics, complexity, power, bandwidth, gain, size, weight, cost). We are exploring alternative ideas and need:

* P1dB: 30 dBm
* gain (at P1dB): 30 dB +- 1.5 dB
* 1 dB bandwidth: 40-350 MHz (10-450 MHz if possible, or talk about e.g. 80 MHz+-10 MHz narrowband if it dramatically simplifies the design)
* 3U form factor, maybe 4 HP, maybe 2 channels per board
* if possible no fan, no moving parts, convective cooling
* power 12 V or as convenient
* fast output reflection power protection (only if that design is risk-free)
* optional: a µC on board for calibration, monitoring, USB/UART/I2C

Questions:

* Would this be a significant simplification w.r.t. Booster or would it meet the same challenges?
* Are there other interested users and use cases?
* Is there something missing or too much to make this both different from Booster and decidedly useful for applications?

gkasprow, 2020-01-15:
This is more or less the Booster channel itself. We would need to modify the output stage to work with 12V.
Gain is lower, so one preamplifier would be fine. For 30dBm operation, we may find the output stage that does not need negative voltage and sequencing. [PHA-202](https://www.minicircuits.com/pdfs/PHA-202+.pdf) could fit.
It is OK, but we will face temperature stability challenge which in a booster is not an issue due to excessive cooling. The protection circuit is simple and could be copied from Booster.
No firmware, after exceeding the power one has to press the reset button. We can also add connectors with error/reset signals.  The protection threshold can be set by the front panel trimpot. Alternatively, we can have EEM and I2C for control. Booser channel has I2C for calibration and threshold adjustments.
If we go for microcontroller design, we get Booster. 
A flat heatsink can be attached to the board bottom, but with 1W won't be necessary.
So, if you accept worse temperature coefficient, and lack of firmware, this would make sense.
2 channels could fit easily.
The booster can be ordered with not all channels assembled. This significantly lowers the cost.
Such 3U module makes sense if it is very simple and much lower cost and performance than Booster.

hartytp, 2020-01-15:
This feels a lot like a 1 channel version of Booster which would likely face the same issues. The RF design is really the easy part, it's making everything robust which is the tricky part.

jordens, 2020-01-15:
Exactly. This needs to be distinct from Booster.

What drives the temp co? If it is the power amplifier bias current, then `10*log10(1+2*350mA/200µA)=0.005 dB/K` for the PHA-202 would be very much ok with me. Even `0.05 dB/K=1 %/K` total might be fine as long as that is stable and includes the positive feedback.

Pressing a button on interlock and trimpot for threshold is also OK. Or maybe a ~10 minute self-reset if that can be made safe and robust.

Giving up the remote forward/reverse power, temperature, currents/voltage monitoring options is also OK (leave testpoints and goldpins for debugging).

@hartytp What issues do you have in mind?

Looking at Booster the RF design does not seem to be the easy part. See the transient problems and the damages, or the problems with the mechanics/connectors of the high performance RF enclosure, or the challenging bias current control/calibration.
And ditching the digital and monitoring part would remove all "tricky parts" relating to problems with robustness of the software and the control electronics.

jordens, 2020-01-15:
Another question is whether there is space between COTS devices (e.g. 2x [ZHL-1-2W+](https://www.minicircuits.com/pdfs/ZHL-1-2W+.pdf) 560 $ each on a PCB with a power supply module) and Booster.

gkasprow, 2020-01-15:
@jordens the preamp (i.e. [ADL5536](https://www.analog.com/media/en/technical-documentation/data-sheets/ADL5536.pdf) also affects the tempco of the entire amplifier. We can limit it by adding some feedback path, but it may need more stages with additional gain margin. Once we debug the RF problems with Booster, we could re-use parts of the RF chain together with protection features. If we go for lower power (30dBm), no bias settings would be needed. Single EEM with local shields and already used RA SMAs is robust enough. Take into account that such a simplified amplifier with PCB mounted shields would have much higher crosstalk between channels, especially when plugged to neighboring channels.

jordens, 2020-01-15:
The ADL5536 has about -0.005 dB/K, the PHA-202 apparently only -0.002 dB/K. Both negative. Summing we'd be better than 0.01 dB/K absolute worst case, which is fine with me.

Why do you expect worse crosstalk than e.g. on Urukul or Mirny? Urukul has 24 dB gain blocks on each channel, unshielded even. We don't have significantly more gain here.

hartytp, 2020-01-15:
> @hartytp What issues do you have in mind?

Ask me when we've finished debugging Booster. but e.g. our pre-amps seem to have a habit of blowing up. I believe we now understand that (see the thread on Booster) and can fix with some diodes, but these unforeseen things always come up and take time . IME these projects always take an order of magnitude or more time to get working robustly beyond the basic RF design. If you strip out the diagnostics and reduce the power consumption that will certainly help a bit.

> Another question is whether there is space in the spectrum between COTS devices (e.g. 2x ZHL-1-2W+ 560 $ each, low efficiency, large heatsink or ZX60-100VH+ on a PCB with a power supply module) and Booster.

We did think about that (and if we have many more problems with Booster then that's our plan B). But, by the time you've sorted out robust power supplies and mechanics it's still a fair bit of work and additional cost on top of the MCL price. If you're doing that then adding the diagnostics/interlocks on top is useful and should have been relatively low cost and easy to implement. Then you have Booster.

The  ZHL-1-2W+ are 21.6W each. So, with even two of them you've got a non-trivial thermal management issue. If you're like us and running labs with a lot of RF amps the combined heat load adds up so having something more efficient becomes valuable.

jordens, 2020-01-15:
@hartytp All agreed. We should reap all the hard work that went into Booster.

Another observation: From the ~32 Urukul attenuations @dnadlinger [posted](https://github.com/sinara-hw/Booster/issues/335#issuecomment-561401294) only four are below 20 dB and all are 17 dB or more. You could cover most if not all of those channels with 16 dB less gain and 6 dB less P1dB. I see the same behavior in opticlock.

gkasprow, 2020-01-15:
OK, I can give it to one of my students and see what comes out of it:) Are you in a hurry?

dtcallcock, 2020-01-15:
If you could live with +28.7 dBm P1dB we could use a [PHA-13HLN+](https://www.minicircuits.com/pdfs/PHA-13HLN+.pdf).  Whilst it doesn't reuse Booster, it's a simple 1-chip solution so we're basically just copying the eval board.  I was actually planning to post an RFC suggesting that we put a couple of these on a simple 4HP card.

- The gain of ~24dB is pretty well matched to few-dBm sources (and @jordens desires)
- Nice 1.1dB noise figure
- Cheap at $10
- SOT package is easily hand solderable if the user ever needs to replace
- 1-1000 MHz bandwidth
- Dissipation is a manageable <2W

I don't have a very good sense of how robust this amp is and whether things like active protection are really necessary.  Perhaps we could run an eval board at full reverse power and see if bad stuff happens (or just ask an MCL engineer). 

A DNPed pad of a convenient MCL filter package on the input and output might be nice too.

>  ZX60-100VH+ on a PCB

This amp is very handy but the 100MHz max freq limits its use.

gkasprow, 2020-01-15:
TVS is quite robust solution as a protection against reflected power. One needs to make sure it can survive excessive power. We can also make assembly variant with one or two stages.

hartytp, 2020-01-15:
> Another observation: From the ~32 Urukul attenuations @dnadlinger posted only four are below 20 dB and all are 17 dB or more. You could cover most if not all of those channels with 16 dB less gain and 6 dB less P1dB.

Yes. We talked about this in at least a couple of places on the Booster issue tracker. The conclusion each time was that the gain would be quite marginal for us if we scrapped a gain stage (pre-amp) inside Booster -- particularly if we want to keep open the possibility of driving to slightly higher modulator powers than we currently do.

Also, Urukul has a rather high output power. For example for phaser, which doesn't have the pre-amp, the gain is necessary to saturate our AOMs. Arguably it's better design (at least for our use-cases) to put the pre-amps (and hence the power dissipation) inside Booster where it's better managed than to put it on an EEM.

Obviously we can reduce the gain without scrapping a gain stage by adjusting the (already present) attenuators inside Booster, but it's not clear there is much benefit to doing that rather than just using the Urukul pre-amp.

> and 6 dB less P1dB

Yes. We could reduce the P1dB a bit, although I wouldn't be comfortable taking it down a whole 6dB. It's useful to keep a bit of headroom here (particularly since we have other modulators which need a little more juice, and it's nice to have some drive room to compensate for matching/cabling losses, etc.)

Again though, even if we do reduce this a bit I'm not sure it changes the design. The currently have a fairly low cost high-quality single transistor PA stage. I don't think that changes much even if we target, say 3dB less power. A nice thing about Booster is that the bias is controlled digitally, so it's trivial to reduce the bias to save power if desired (characterising the amp at lower bias/gain/P1dB is on my to do list anyway).

hartytp, 2020-01-15:
Essentially though @jordens I do agree with all your points and I wouldn't want to imply Booster is optimal for all use-cases. But, it feels like it does a pretty good job so I'm focusing on getting it working well (not that that should restrict what anyone else does). If the TVSs work as expected, the remaining issues are all sw and I'm confident we can sort that out in a reasonable timeframe one way or another.

hartytp, 2020-01-15:
> OK, I can give it to one of my students and see what comes out of it:) Are you in a hurry?

Well, that gives me flashbacks I didn't need

hartytp, 2020-01-15:
> Whilst it doesn't reuse Booster, it's a simple 1-chip solution so we're basically just copying the eval board.

Again, that's not really so different to Booster. Both PAs are just basically single FET PAs with some passives around for matching, power supplies, thermal management. Booster is fundamentally a really simple device. The power is also a little low for many of our applications.

dtcallcock, 2020-01-15:
> Again, that's not really so different to Booster. 

I don't think that's true - the PHA-13HLN+ doesn't require the external bias and power sequencing circuitry of the NPTB00004A, just an 8V power rail (and also no preamp).

 > The power is also a little low for many of our applications.

I thought that a 750mW amplifier would fit the 'Booster is overkill niche' nicely. However it sounds like people want something that is basically as powerful as Booster, but just 1ch and without all the bells and whistles.  That's also fine, just wanted to be clear.

gkasprow, 2020-01-15:
For some time I am thinking about building a simple interface that could be attached to a single Booster channel and make essentially 1-channel Booster, supplied i.e. from USB-C with console access only. But let's make Booster working without issues first.

jordens, 2020-01-16:
That ~28 dBm P1dB (at ~100 MHz) is on the too low side for me. I'd really want those 2 dB more. Otherwise the PHA-202 has the same easy powering requirements/price/etc but would need a small preamp. But the actual part choice is not critical to me. I know this is well in the CATV upstream/downstream amplifier territory and there should be plenty of options to choose from, hopefully also ones with long product lifetimes. Let's see later what we can robustly achieve in terms of available chips.

@gkasprow Not in a hurry. A year until working devices would be OK for me. Would this be something for Michal at HUB in March maybe?

gkasprow, 2020-01-16:
Old CATV amps have often the non-linear responses. I played with a few of them and they do not pass low level signals. This is probably done on purpose to behave as a noise gate. Analog TV uses shallow AM modulation so they don't care about linearity so much.
Michal is not RF specialist, I want him to play wit ARTIQ because we miss such competences.

jordens, 2020-01-18:
Even in the non catv linear amplifiers there are several really cool options. Qorvo: https://www.qorvo.com/products/p/TQP7M9105
https://www.qorvo.com/products/p/TQP7M9106
This one would be funnny. Would have to test whether it works at lower frequencies: https://www.qorvo.com/products/p/RF6886

jordens, 2020-01-18:
And a bunch of interesting devices vom wantcom. https://www.wantcominc.com/WHM_series_PA.htm (thanks Nils).

gkasprow, 2020-01-18:
the Vom devices need negative gate voltage and sequencing circuit. Not a big problem but complicates the design.

gkasprow, 2020-01-18:
I can buy devkit and check the lower frequency. Choose 2 or 3 the most interesting amps.

jordens, 2020-01-24:
I'd play with the two qorvo amps (TQP7M9105/6). Wide band match will probably be tricky. The guys at PTB said that the second one was hard to get working properly.

gkasprow, 2020-01-29:
I ordered the TQP7M9105 devkit.

jordens, 2020-02-04:
I think PTB will play with the RF6886 a bit.

Beyond the usual trick to go to class D (which one could also look at to great efficiency gain), I stumbled over some cool newer ideas and many pointers to older ones on wide band matching (up to 0.6 relative bandwidth) and maximizing efficiency in class AB (more or less) amplifiers in this [nice Ampleon AN](https://www.ampleon.com/documents/white-paper/AMP-WP-2016-0909.pdf) on Doherty amplifiers and ultra-wideband matching and [some FBH work](https://ieeexplore.ieee.org/document/8910788) with resonant "outphasing" and class D.

Looking at these options is a very nice but also time consuming project to get right. If we don't do any of these tricks however, the large power back-off in practical applications completely kills efficiency. Then again, an inefficiency at 1 W RF is not that much of a problem from the systems design perspective. More of an issue for Booster, where, looking at the Booster survey data, it seems to be typically operated at around 10% overall efficiency.

gkasprow, 2020-02-05:
If you want to go for higher efficiencies, there are [ICs and modules](https://www.maximintegrated.com/en/products/comms/wireless-rf/SC1894.html/tb_tab3) from Maxim that do adaptive linearization. Essentially you insert it between input and output coupler and it does the magic itself, increasing the efficiency. I'm not sure if it is wort the effort in this application because such block can also create the new physics in some experiments:)

jordens, 2020-02-05:
Nice! I didn't know that existed. This may be extremely interesting when used to un-distort AOM or microwave electrode drive chains, especially when used with this device in-loop: place it between the PA and a photodetector after the AOM or a monitoring tap after the trap respectively.

gkasprow, 2020-02-05:
I'm going to use it in my design so I will have more info soon.

gkasprow, 2021-04-16:
What about using [QPB8808SR?. ](https://www.qorvo.com/products/p/QPB8808) There is also a cheaper, pin-compatible version with lower output power ([QPA8801](https://www.qorvo.com/products/p/QPA8801)). It's very low distortion amplifier, and it would make it a good candidate for applications requiring low harmonic content.
It has bias regulation, so one can trade efficiency over linearity.
That could be a nice supplement to Booster.
I ordered a devkit, will give it a try.
If offers 33dBm IP3 and 50 – 1218 MHz Bandwidth

jordens, 2021-04-16:
They both look really interesting!
33 dBm P1dB and OIP3 of 50 dBm (c.f. ~36 dBm for Booster) at 6 W draw.
A bit annoying that it's CATV 75R, but maybe it doesn't matter.

gkasprow, 2021-04-16:
I know it. But input impedance doesn't matter since the preamp will be millimeters away.
The output impedance is not that critical since it's the load that matters.
These amps are accepting all kinds of loads.
I'm testing intensively various power doublers recently for other applications and they are good for amplifying complex, multitone signals.

jordens, 2021-04-16:
Why are they called power doubler if they have a gain of ~ 20 dB?

gkasprow, 2021-04-16:
They have a double power stage - push pull. Double cascode

dtcallcock, 2021-04-16:
>The output impedance is not that critical since it's the load that matters.

It needs an output balun anyway so can't you use that to get to 50Ohm if you really want it matchy matchy?

gkasprow, 2021-04-16:
>It needs an output balun anyway so can't you use that to get to 50Ohm if you really want it matchy matchy?

Yes, I can and I will :)

gkasprow, 2021-04-16:
I think it would be sufficient to replace the input and output baluns with 1:1.5 impedance ratio ones. I ordered such ones and will give them a try.

gkasprow, 2021-04-16:
how could we call the new module? Something like LINPA ( LINear Power Amplifier)? Or maybe you have a better name related to the typical application.

dtcallcock, 2021-04-16:
Boostette.

jordens, 2021-04-19:
I would not want to encode features or specification into the name. E.g. "linear" would conflict with a more linear version of Booster. "Sipa" for "simple power amplifier"? OTOH @dtcallcock 's suggestion (ironic or not, I don't know) sounds ~OK to me, even if it's not a battery charger:

[image]

dhslichter, 2021-04-23:
I'd prefer a non-gendered diminutive "boostlet"?  Or how about LPA? ("Little power amplifier")  Or SmolPA.

dtcallcock, 2021-04-23:
Lil Boosty

jordens, 2021-04-24:
If you are worried about gender bias when naming a device you should definitely be able to see and avoid the size and age bias encoded in diminutives like  "smol" or "lil" and the manifest cultural bias against non-english names that prevailed a couple years ago.

That leaves "Sipa" as the only suggestion avoiding possible biases. It's opaque and has no size, age, language, or sexual biases.

I seriously wonder what the non-gendered and non-stigmatizing alternative to a hermaphroditic connector is.

gkasprow, 2021-04-24:
I received the QPA8801 devkit. I powered it and discovered that it's just a heater. It does not amplify. So I replaced the IC. Still no result. I soldered in QPB8808 - still, no amplification. Then I discovered that devkit is missing a bias inductor to GND(L2). Without it, IC died silently after powering up. So, I soldered a 1uH inductor and replaced the IC with QPB8808, and modified R1 =910 and R2=18k. It started working.
[image]

gkasprow, 2021-04-24:
First, I measured S11 with the [original](https://cdn.macom.com/datasheets/MABA-009210-CT1760.pdf) 75 Ohm balun. It looks surprisingly well. The S11 is below -13dB from 14MHz up to 900MHz
[image]

Then I replaced the input balun with [1:1.5 one](https://www.minicircuits.com/pdfs/ADT1.5-122+.pdf)
it improves S11 at low frequency (20-130MHz) but then it makes it worse. When I add some pF capacitance it behaves slightly better.
[image]

Then I assembled back the original 1:1 75Ohm balun and added autotransformer [1:1.5](https://www.minicircuits.com/pdfs/ADT1.5-17.pdf). It gives the best S11 - below -20dB from 20MHz up to 468MHz
[image]

gkasprow, 2021-04-24:
The next thing I measured was S22. Original 1:1 75Ohm balun:
[image]

Then I replaced the output balun with 1.5:1 [one ](https://www.minicircuits.com/pdfs/ADT1.5-122+.pdf). The S22 got 4dB better 
[image]

Then I added the same combination of baluns as in the case of S11 measurements - a 1:1 75Ohm balun + 1.5:1 transformer. The situation didn't change much
[image]

gkasprow, 2021-04-24:
The S21 was taken with a 20dB attenuator + 3m of cables. Input and output is equipped with baluns + 1:1.5 transformers.
The 3dB BW is from 10MHz to more than 1GHz
[image]

gkasprow, 2021-04-24:
The next thing was harmonic and gain measurement in various output stage configurations. I used a precise 20dB output attenuator.
Original 75Ohm balun. Pin=11.1dB, f=150MHz.  G=18.9dB
[image]

With 1.5:1 output balun.
Pin=12.1dBm so the gain dropped by 1dB. G=17.9dB
[image]

With 75Ohm 1:1 balun + 1.5:1 transformer. Input power is 11.3dB. G=18.3dB
[image]

The harmonic distortion doesn't change at 1W output power in various output circuit configurations.

gkasprow, 2021-04-24:
If someone wants to play with raw data, they are here. The files can be viewed with free [Keysight software]
(https://www.keysight.com/zz/en/lib/software-detail/computer-software/fieldfox-data-link-software-2085717.html)
[QPB8808.zip](https://github.com/sinara-hw/meta/files/6370604/QPB8808.zip)

gkasprow, 2021-04-24:
Reduction of bias current to 300mA drastically increases 3-rd harmonic content at max power also decreasing the gain.
It simply means that clipping comes earlier.
I=300mA, Pin=15.5dB, 20dB attenuator
[image]

At lower power, it does not look bad.
I=300mA, Pin=2.6dBm 20dB attenuator
[image]

gkasprow, 2021-04-24:
The lower power mode (Ib=300mA) is useful for output power up to 24dBm. If someone wants more, 500mA current must be enabled. The output power higher than 33dB activates some thermal protection which causes the power to drop to 28dBm. After lowering the input power and waiting several seconds the amplifier work again correctly.

gkasprow, 2021-04-24:
To conclude:
- we will use the preamp to ramp up the gain, so the S11 is not that important. We can use the same balun as at the output to limit the BOM. S11 is still much better than -10dB
- we will use a 1.5:1 output balun because it gives better S22 at the cost of a 1dB drop of gain. But we have preamp anyway.
- preamp assembly variants would be needed to cover 24dB and 40dB gain. We can also use two SPDT switches to disable part of preamp

dnadlinger, 2021-04-24:
Clearly the only acceptable name is Boosty McBoostface.

jordens, 2021-04-25:
@gkasprow you were worried about those catv amps squelching for low input powers. Could you test with 10us pulses of low duty cycle?

jordens, 2021-04-25:
@gkasprow What 1.5:1 output balun is that? And if you have the data: taking all together, what P1dB (or thermal shutdown power if that's earlier) did you get excluding cable losses?

gkasprow, 2021-04-26:
This particular amp is linear even at very low power. It's foreseen for digital CATV when multiple tones need to be transmitted reliably. The ones squelching for low input power were designed for analog TV with AM modulation. It was P1dB that occurred earlier.  I had to pump over 18dBm of power to get a thermal shutdown.
I used ADT1.5-122+. Right, they are specified for 0.5W power. I probably saturated one.
However S parameters were taken at much lower power.

gkasprow, 2021-04-26:
the internal LF generator in SMB100A has limited rise time. I have to get some high-speed solid-state RF switch...
[image]

gkasprow, 2021-04-26:
To measure the P1dB I will bring my Agilent power meter. SA is not good at that.

gkasprow, 2021-04-26:
@jordens I did some observation and I'm not that sure about P1dB I wrote above. That was probably tested with a cold heatsink.
 I don't see noticeable power drop up to 32dBm. At 33dBm I see a 0.2dB drop. At 33.5dBM I see 1dB of drop. At 34...35dBm it enters power limit depending on how hot the heatsink is. So it looks like P1dB is just before the thermal shutdown.

gkasprow, 2021-04-26:
I observed only the first harmonic using SSA.

[14 more replies]