sinara-hw/meta #51: Wiki: Board revs

dtcallcock, 2020-02-18:
There has recently been a lot of minor point board revisions.  There so far isn't really a systematic way of dealing way of keeping track of the following in the wiki and figuring anything out involves a trawl through issues.

- Change log
- What board revisions were actually produced and when (I get the impression a few revisions only had order 1 prototype made and can pretty much be ignored).
- What fixes can be applied to earlier revisions to bring them in line with newer revisions
- Photos of the older boards
- Old test data and specs

Perhaps a revisions section at the bottom of each page with a table?

jbqubit, 2020-02-18:
For Urukul its on the wiki. https://github.com/sinara-hw/urukul/wiki#errata

I rather like the ARTIQ [RELEASE_NOTES.rst](https://github.com/m-labs/artiq/blob/master/RELEASE_NOTES.rst) approach since the notes are in the repository.

dtcallcock, 2020-02-18:
> For Urukul its on the wiki.

Sure, I'm not saying there isn't ad-hoc documentation for a few boards.  I'm saying I don't see a nice systematic template that can be applied across the wiki.