sinara-hw/meta #52: [RFC] mechanical designs

gkasprow, 2020-05-05:
In the Sinara project, we focused on electronics. But the physical lab is much more than that.
We can also design, produce, and distribute in the same way also mechanical and mechatronic equipment.
For example, I have a 3D printer farm -12 printers that at the moment are producing 24/7 adapters for medical staff, but in a few weeks can be used for other purposes.
I'm pretty sure there are gadgets in the typical lab that can be made of ABS, PETG, or other 3D-printable material.
For the moment the only use of 3D printing technology are SMA isolators.
We can also do much more complex stuff like the Booster amplifier enclosure. We can integrate stepper motors, actuators, optics. In CTI we produce scientific-grade CCD cameras so we can use the resources to design something that is needed and not easy to buy.

jordens, 2020-05-05:
It's best to decide this on a case-by-case basis. The front panel prototypes or SMA sleeves are likely a perfect use case. But in the past when I needed a mechanical part, 3D printed parts either lacked the strength, surface quality, dimensional accuracy, porosity, thermal, electrical, vacuum properties etc. There were exceptions though (simple enclosures, fixtures and "holders" of all kinds).
What on Booster would you do with a 3D printer? How do you get thermal and RF conductivity?

jordens, 2020-05-05:
I could imagine that with **very** careful design (thermal, mechanical stability) and material selection some kind of library of optomechanical parts would be very useful if it works. The typical use case of "cage system", lens and mirror holders, translation, adjustments.

hartytp, 2020-05-05:
Interesting. The standard issue for this is finding something that is sufficiently specific/niche that Thorlabs etc don't already offer it cheaply, but not so specific that there are enough users to make it worthy of a Sinara project rather than one labs' in-house design.

sbourdeauducq, 2020-05-05:
Is there an established procedure for testing such mounts, e.g. measuring influence of temperature, resistance to vibration, etc.?
Thorlabs don't exactly offer those cheaply but the Chinese do (e.g. $28 for a basic kinematic mount for 25.4mm optics).

gkasprow, 2020-05-05:
It's not only about 3D printing. This was just an example. The booster enclosure was machined.

jordens, 2020-05-05:
The goal would need to be to offer something **better** than Thorlabs (in the same way that Sinara competes and beats other existing mainstream vendors). Doing the same thing and higher price is stupid.
For optomechanics, replacing modularity at assembly-time by modularity at design-time would be one idea. Instead of hunting down the right length of cage system rod, you'd assemble the mechanics for your beam telescope on your computer and then print it saving grub screws.
Would obviously need a comprehensive analysis and proposal.

gkasprow, 2020-05-05:
Recently I produced tens of detector holders that keep the right position between scintillator, fiber, and  SIPM detector assembly. I also printed the entire scintillator electronic enclosure. We need over a thousand of them for the entire detector. With my printer farm, I will do that in a week. Here are more details https://github.com/elhep/SiPM_AFE/issues/3

hartytp, 2020-05-05:
Cool!

dtcallcock, 2020-05-05:
Offering some nice photodiodes would be a good starting point for extending Sinara into the optics realm.  What's out there isn't great and is somewhat overpriced.  There is also the opportunity to combine the photodiode with a fiber collimator, polarizing beamsplitter, and a pickoff in a compact mechanical package.  This would allow for plug and play integration into a SU-servo system.  

@dnadlinger has a nice design along these lines.

dtcallcock, 2020-05-06:
Sort of related, the [MIT group](http://web.mit.edu/~cua/www/quanta/) is making a tool to design optical subsystem breadboards.  You can then machine them from aluminum using your favorite maker tool/service.

[https://www.openopticsbench.org/](https://www.openopticsbench.org/)

It's still a work in progress but Ike has a student working on a release version.

The example is an AOM double pass, but they've made a bunch of other stuff like an atomic absorption lock, an injection-locked diode laser, etc.

Based on OpenSCAD, python, and FreeCAD apparently.

It uses Thorlabs parts but cuts costs a lot as you don't need all those posts, bases, spacers, and clamps.  I'm sure parts from @sbourdeauducq's  favorite cheapo Chinese vendor could be added too.

[image]

airwoodix, 2020-05-06:
Along the lines of @dtcallcock: is there a way to offer photodiode heads like Thorlabs [S120C](https://www.thorlabs.com/thorproduct.cfm?partnumber=S120C) but 10-ish times cheaper?

No need for certified calibration, just a relatively large area (> 2x2 mm²) photodiode and a  wide enough dynamic range for monitoring purposes. Maybe something around [logpd](https://github.com/jordens/logpd)? Ideally with an ad-hoc Sinara module to get large channel counts (analog output from the photodiodes sounds easier to manage, but Sampler channels are too expensive).

Sounds more like an electronics project though, so a bit off topic here I guess.

dtcallcock, 2020-05-06:
>analog output from the photodiodes sounds easier to manage

There have been vague ideas for a general digital peripheral card since the beginning of Sinara but nothing solid (at least I couldn't find the old issues where this was discussed).  I think the idea would be to combine an SPI bus and power together on some commodity connector (eSATAp?).  Then there would be a Sinara card with eg. 4 of these on the front panel that one could plug devices into (photodiodes, temp sensors, switched RF attenuators, etc).  In this scenario, the photodiode would have an SPI ADC onboard. 

> but Sampler channels are too expensive

This was one reason I pushed for CPU ADC access on Stabilizer (3× ADCs with 16-bit max. resolution, up to 3.6 MSPS, 36-ch multiplexer).

One could also imagine making a nice dedicated analog input/datalogging card around the STM32 that reuses a lot of Stabilizer/Thermostat work.

gkasprow, 2020-05-06:
We already have an RJ45 DIO module. Why not use photodiode with RJ45 and SPI over LVDS? Another issue is the power supply. Maybe worth exploring PoE-like solution...

dtcallcock, 2020-05-06:
> Another issue is the power supply

Sure.  If the [Aux PSU card](https://github.com/sinara-hw/meta/issues/48) was produced then you would just put that and the RJ45_DIO side-by-side and run an RJ45 and M8-4 cable to each peripheral.  Obviously not as neat as a 1-card/1-cable* solution but on the other hand do with want to create another little 'mini-ecosystem'?  

*We could also make a custom cable with an RJ45 + M8-4 fork on one end and some more compact single connector on the other.

gkasprow, 2020-05-06:
For the photodiode, you need 3-wire SPI. Just deliver the clock, CS, and data comes out.  Or, use bidir data line like in the 3-wire SPIs. then one pair could be used for the power supply. Do you need some configuration? Gain/BW switching? Another idea is to use clock as a power line. The clock can be AC-coupled and carry DC voltage as well. Two coils make the separation.

gkasprow, 2020-05-06:
Yet another option is USB-C cable. It has enough pins to support both USB, config, and high sped data lines. I'm using them to connect silicon multiplier diodes in my experiment. The AFE has regulated power supply, high-speed TIA, LVPECL calibrator, STM32 MCU for management. One can plug it to standard PC to upgrade the firmware when plugged to dedicated USB-C connector it works as a dual HS detection head. I use CAN as a manaement IFC because I will have over a thousand of such detectors.

gkasprow, 2020-05-06:
https://github.com/elhep/SiPM_AFE/issues/1

dtcallcock, 2020-05-06:
> Yet another option is USB-C cable.

It's pretty cool that it works both as USB and as a custom interface.

> when plugged to dedicated USB-C connector it works as a dual HS detection head

What would an EEM to drive these kind of things look like?

> STM32 MCU for management

What size/cost/power consumption are you looking at on the chip you'd need for this?

dtcallcock, 2020-05-06:
> But in the past when I needed a mechanical part, 3D printed parts either lacked the strength, surface quality, dimensional accuracy, porosity, thermal, electrical, vacuum properties etc. There were exceptions though (simple enclosures, fixtures and "holders" of all kinds).

I too was pretty unimpressed with this stuff circa a couple of years ago.  The latest generation of 3D printing [tools we've got here](https://tsa.uoregon.edu/equipment) can print things like carbon fiber, 99.95% fill titanium, 3D PCBs with embedded components, stuff with sub-micron features etc.  Not sure any of it's useful to me yet but I'm starting to feel like it's not just for making little plastic holders any more.

gkasprow, 2020-05-06:
A lot changed last years. Even with low-cost printers, one can do quality prints. Here are some examples I did recently
[image]
[image]

jordens, 2020-06-06:
For enclosures like booster I'd think that sheet metal and a bit of thinking would be perfect in all regards.
https://youtu.be/RopgrECLSJc for some inspiration. The other videos are an extremely good intro to most things mechanical with maximum information density for the average physics PhD student.

hartytp, 2020-06-07:
Nice video @jordens 

Agreed, Booster is massively over designed mechanically/thermally and there is plenty of room for simplifying and reducing costs if anyone has the bandwidth.

gkasprow, 2021-08-18:
interesting 3D project 
https://journals.aps.org/prxquantum/abstract/10.1103/PRXQuantum.2.030326
https://twitter.com/PRX_Quantum/status/1425852083085922309

dtcallcock, 2021-08-18:
Seems like a total gimmick. Why wouldn't you just machine that stuff from aluminium - it'd work much better (>10x the modulus, less than half the CTE, way smaller tolerances, only 2x the density).

gkasprow, 2021-08-18:
Totally agree. Soft Aluminium - capable CNC can be bought fot a few hundreds of $.  Recently I bought used professional grade CNC capable of working in hard steel for 3k EUR.