sinara-hw/meta #54: [external board]  Colorlight 5A-75E/B

jordens, 2020-06-17:
Just a note on an interesting cheap board I was pointed to, that could be used to handle the need for "lots of digital IO".

These are two boards that more or less match Banker. They are output-only, and instead of EEM they speak 1000Base-T Ethernet, but the have an ECP5 with transceivers. Dirt cheap.

https://www.aliexpress.com/item/32332336247.html
https://www.aliexpress.com/item/32331955683.html
https://github.com/q3k/chubby75

gkasprow, 2020-06-17:
It's cheaper than I would pay for assembly:)

gkasprow, 2020-06-17:
I saw this board already. They are used to control large LED screens (video walls) . Several such LED tiles are daisy-chained by 1Gbit Ethernet

sbourdeauducq, 2020-06-18:
It's ECP5 and seems to have a SDRAM, it could also be used as the cheapest ARTIQ core device possible :) Plus, no Vivado required.

gkasprow, 2021-01-23:
It could be good as an embedded system with ARTIQ controlling some IOs, maybe some slow Arduino boards with sensors, DACs ADCs etc. I do not see it as a Kasli replacement.

gkasprow, 2021-11-09:
Recently I played a bit with ultra-low-cost FPGA devkit from EFINIX. They are available from Digikey. They are similar to ICE40, but are optimized for video processing (MIPI IP cores). And, the environment supports HW debugger. An interesting alternative to many FPGAs with 50 weeks lead time.