sinara-hw/meta #55: Sinara Paper

dtcallcock, 2020-06-19:


gkasprow, 2020-06-19:
That's a great idea. I will contribute.

cjbe, 2020-06-19:
I completely agree that this would be a really useful paper to write. In my mind this should also be suitable as the first reference / overview we can point new users to in order to get a feel for the Artiq/Sinara ecosystem.

I have already had a significant number of people emailing me after seeing the Artiq software citation in a few of our papers with confused questions about what hardware Artiq can run on.

dtcallcock, 2020-06-19:
I've invited the usual suspects to the Overleaf.

Overleaf has an inline commenting system so probably best to discuss minor issues there than here. 

I will keep adding things as time permits.  Happy to triage other people's contributions too.

gkasprow, 2020-06-19:
I added Dorota to the project. She is physicist and also studies IT. She will help us with writing papers.

dhslichter, 2020-06-19:
@dtcallcock I like this idea!  I suggested the same thing to @gkasprow and others about a month ago (albeit with smaller scope), encouraging them to submit the paper to the new IEEE Transactions on Quantum Engineering, an open-source journal of which (full disclosure) I am an assistant editor.  I think it would be a good venue; obviously one could also go for something like RSI.  One advantage of IEEE TQE is that it is open access (publication fees are waived for submissions at present), and that it will enable us to reach more engineers than RSI, I think.  This could be good for encouraging folks in engineering with an interest in quantum to join in the effort on hardware/software/gateware development.

dtcallcock, 2020-06-19:
> (albeit with smaller scope)

I don't necessarily want to dictate the scope.  Some of the 'sections' I have could end up just being a few sentences.  IMHO, just getting it done is more important that completeness.  It's a moving target and we have the wiki to fill in the gaps.

> One advantage of IEEE TQE is that it is open access...and publication fees are waived

RSI is open if you pay $3500, but free is better (though perhaps someone here is at an institution where they can use a central pot to pay the charge).  I don't have strong feelings beyond noting that RSI is clearly the default option.  As far as I can tell, there are no journal-specific requirements that mean we have to decide before writing anyway.

gkasprow, 2020-06-19:
If there is a need to pay for publications, I have a few k$ for such activities in my research grant.

dhslichter, 2020-06-19:
Agreed that it's better to get something out describing in broad terms, rather than having to be an exhaustive study.  Each one of these boards could be a stand-alone paper, really.  The point is more to describe how the system works, and the features it has, and the ways it can be extended.  Part of this will be about interfacing with ARTIQ too....there's a paper that needs to be written about ARTIQ alone too, but that is another separate challenge.  

I agree that RSI is the "default" option, but I would strongly suggest IEEE TQE be considered, although because it is new I understand if people are wary.  Since it is an IEEE journal, I think it would let us reach a broader audience in the end, especially among engineers (I don't think that many electrical/computer engineers read RSI?).

gkasprow, 2020-06-20:
We started preparing papers for individual boards of families of boards already

gkasprow, 2020-06-20:
[This](https://www.dropbox.com/s/b28zvq5w0qsn1rg/Sampler_article_for_IJET_jrn.gk02.pdf?dl=0) is a paper about Sampler. We plan to write about all other modules in similar way

dtcallcock, 2020-06-20:
> This is a paper about Sampler. We plan to write about all other modules in similar way

Oh wow, I didn't realize.  Now I feel like I was doubting you had this all in hand!

An overarching Sinara paper would clearly still be useful.  For example, there are a few chunks of your Sampler paper that could move into the Sinara paper so you can just cite them instead of copy-pasting across 20 individual board papers.  It can also do the heavy-lifting on citing prior work by other groups.

> Since it is an IEEE journal, I think it would let us reach a broader audience in the end, especially among engineers (I don't think that many electrical/computer engineers read RSI?).

If Greg is putting out papers in heavy-duty EE journals, perhaps the converse is true and the Sinara paper should be targeted more at physicists.

dhslichter, 2020-06-20:
@gkasprow it would be good if you would ask the authors before submitting -- for example, I don't think it is appropriate for me to be an author on a paper about Sampler, since I don't think I contributed very much to it.  It's also important to be able to see what my name is being attached to, so I can make sure I am comfortable with the material being presented.

gkasprow, 2020-06-21:
@dhslichter 
- it was not submitted yet
- I posted it on GitHub, but now I can't find where it is :) I should have opened a dedicated issue.
- I asked about the list of authors
- there was discussion about this paper, it is already after a few iterations.

dhslichter, 2020-06-23:
Sorry, I have been very bad at catching github notifications -- just too many alert emails for me from too many repositories.  In any event, I think that it would be appropriate to remove me from the Sampler author list, because I don't think I contributed to that effort at the authorship level.

dhslichter, 2020-06-24:
On the topic of journals: IEEE TQE has a special section on classical control electronics right now, which would be a perfect fit for this.  Here's a link:

https://tqe.ieee.org/cls_ctl_cfp/

DorotaNowicka, 2020-06-30:
Some time ago I started working on Urukul article in IJET format, similar to that of @gkasprow about Sampler. Should I keep working on it?

dtcallcock, 2020-06-30:
> Some time ago I started working on Urukul article in IJET format, similar to that of @gkasprow about Sampler. Should I keep working on it?

I don't think an overarching Sinara paper should in any way detract from single board papers like this.  It should make it easier to write single-board papers as there will be a bit less repetitive background and explanation required - the main Sinara paper can just be cited.

To see how they might work together, I mocked up a Sampler section (on the Overleaf too).  It has a single paragraph board description, front panel photo, board photo, and simplified block diagram.  Do people think more, less, or different details are required?

[Sampler Section](https://github.com/sinara-hw/meta/files/4854854/Sinara-2.pdf)

gkasprow, 2020-07-02:
This was my original idea. Write a lot of papers dedicated to modules or groups of modules (DIO) and then another paper which cites the dedicated ones. So yes, @DorotaNowicka keep writing.

dhslichter, 2020-08-01:
No progress on the overleaf -- is writing taking place somewhere else?

dtcallcock, 2020-08-02:
> No progress on the overleaf

I stopped as I think I have put enough ideas down that there needs to be some feedback on the structure before continuing.  In particular we have to decide [how much detail to go into on each board](https://github.com/sinara-hw/meta/issues/55#issuecomment-652105097).  

> is writing taking place somewhere else?

I don't think so.  If @DorotaNowicka and @gkasprow aren't comfortable with LaTeX/Overleaf we can switch to another platform.

gkasprow, 2020-08-12:
we are comfortable with Latex. Overleaf is great tool. We are just buying license for it.