sinara-hw/meta #57: IEEE International Conference on Quantum Computing and Engineering (QCE20)

dtcallcock, 2020-08-11:
(As I mentioned here)[https://github.com/sinara-hw/meta/issues/49], I'm going to take a (virtual) poster to (IEEE International Conference on Quantum Computing and Engineering (QCE20))[https://qce.quantum.ieee.org/engineering-trapped-ion-quantum-computers/].

Here is a preliminary abstract and author list.  I took the author list from (Greg's OSA Quantum 2.0 abstract submission)[https://www.dropbox.com/s/g9u4tcenul033x1/ARTIQ%20and%20Sinara_G.Kasprowicz_WUT.pdf?dl=0].  Let me know if this is up to date or if people need adding @gkasprow etc.  If anyone wants changes, let me know in the next couple of days.  I'll submit a draft poster for comments closer to the conference.

Sinara: An Open Hardware Ecosystem for Quantum Physics

Sinara is a modular, open-source measurement and control hardware ecosystem dedicated to quantum applications that require hard real-time performance. It is based on industrial standards and consists of over 50 modules. The hardware is controlled and managed by the ARTIQ, open-source software that provides nanosecond timing resolution and sub-microsecond latency via a high-level programming language.

Grzegorz Kasprowicz
Paweł Kulik
Michal Gaska
Tomasz Przywozki
Jakub Jarosinski
Joseph W. Britton
Thomas Harty
Chris Balance
Weida Zhang
David Nadlinger
Daniel Slichter
David Allcock
Sébastien Bourdeauducq
Robert Jördens
Krzysztof Pozniak

jordens, 2020-08-11:
Thanks, LGTM!
FYI, there is also some panel discussion on Oct 16 where I was asked to be on.

dhslichter, 2020-08-12:
> hard real-time performance

Perhaps "deterministic timing performance" is more clear for the uninitiated?  

> over 50 modules

over 50 scalable card-based modules built to perform a variety of analog (dc to microwave) and digital input and output tasks with precision timing.  

> managed by the ARTIQ, open-source software that provides

managed by the ARTIQ open-source software platform, which provides

sbourdeauducq, 2020-08-12:
> Perhaps "deterministic timing performance" is more clear for the uninitiated?

I would suggest "deterministic high-resolution timing" which is descriptive and accurate.

dtcallcock, 2020-08-12:
Thanks.  I added some Oxford commas and spelt Chris's name right too.

gkasprow, 2020-08-12:
Please add: Mikolaj Sowinski, he is SW lead at the WUT

dtcallcock, 2020-10-12:
[ARTIQ_Sinara_draft_sma.pdf](https://github.com/sinara-hw/meta/files/5362774/ARTIQ_Sinara_draft_sma.pdf)

Here's a poster draft.  Most of it's fairly uncontroversial specs taken from the wiki.  I added an FAQ to try and cover some of the questions we get asked a lot though.  I'm sure you all have a slightly different perspective though.  White space needs moving around a bit but I'll wait to see what changes people want before doing that.

Let me know by the end of tomorrow (Mon Oct 12th) if there's anything any of you want changing @gkasprow @hartytp @jbqubit @jordens @dhslichter @sbourdeauducq @cjbe @dnadlinger @WeiDaZhang @marmeladapk @kaolpr @michallgaska .  Sorry about the late notice - this one kind of slipped during the pandemic fun that we've been having recently.

sbourdeauducq, 2020-10-12:
How did you forget Phaser? :)

dtcallcock, 2020-10-12:
Amateur hour.

Fixed: [ARTIQ_Sinara_draft_small.pdf](https://github.com/sinara-hw/meta/files/5363122/ARTIQ_Sinara_draft_small.pdf)

sbourdeauducq, 2020-10-12:
Nitpicking about terminology: the "ARTIQ Server" should be called "ARTIQ controller", which then clashes with "FPGA controller".
Kasli is called "Processor" on m-labs.hk, which isn't a well-established name nor one that I like very much. And the "FPGA" part isn't entirely true since the advent of Zynq-based core devices where a large part of the processing takes place in the PS. Maybe we should start consistently calling those "ARTIQ motherboards"?

sbourdeauducq, 2020-10-12:
Maybe add a link to the ARTIQ homepage https://m-labs.hk/artiq/

sbourdeauducq, 2020-10-12:
The picture of the crate is somewhat dating now, but I don't think we can refresh it before the deadline.

hartytp, 2020-10-12:
> tions that require deterministic high-resolution timing. It is based on industrial standards and consists of over 50 scalable card-based modules built toperform a variety of analog 

50?

jordens, 2020-10-12:
@dtcallcock 

* Please associate me with `QUARTIQ GmbH and M-Labs Ltd`.
* Capitalize it as QUARTIQ.
* `and we’ll take it from there` doesn't match practice (and doesn't work). Maybe `and we'll work with you`.

@hartytp 

There are 57 repos in `sinara-hw`. Minus some software/mechanics/meta/web.

hartytp, 2020-10-12:
> There are 57 repos in sinara-hw. Minus some software/mechanics/meta/web, plus variants.

I'm not overly fussed, but the "over 50" feels a bit high. It depends exactly what you're counting, but e.g. sinara-hw has three different thermostat repos, mothballed Booster mech/firmware, etc if you remove those the web etc we're under 50. I personally also wouldn't count the 5 banker breakouts as card-based modules.

gkasprow, 2020-10-12:
there are some related designs in my group repository, mainly  CPCIS/DIOT 
https://github.com/elhep/

dnadlinger, 2020-10-12:
@dtcallcock 

> Let me know by the end of tomorrow (Mon Oct 12th) if there's anything any of you want changing 

Driver appears to have the same text for the compliance ranges as Zapper; I think 250 mA/1.5 A ∑as discussed.

gkasprow, 2020-10-12:
There is also Almazny, 8-channel Thermostat, Piezo driver mezzanine for Stabilizer and individually isolated 8/16channel DIO
Otherwise, it's fine.

hartytp, 2020-10-12:
> 8-channel Thermostat

What is the plan for that? Same design as thermostat but more channels, or looking to make some changes?

jordens, 2020-10-12:
It's for ai-artiq. Off topic here.

gkasprow, 2020-10-12:
@hartytp it's the same as Thermostat, but with 8 input channels to enable multi-point measurement. It will be an EEM card.

dnadlinger, 2020-10-12:
More typos:

 * "To sync yo experiments" (Planned Hardware -> Line Trigger)
 * "Superconducting" should be lower-case (Metlino/Sayma/RFSoc)

dtcallcock, 2020-10-12:
> Kasli is called "Processor" on m-labs.hk, which isn't a well-established name nor one that I like very much. And the "FPGA" part isn't entirely true since the advent of Zynq-based core devices where [...]

And they are also called 'carriers' in a lot of the early docs.  DI/OT calls them 'system boards' and describes them as 'crate controllers'.  How about we just copy that in anticipation of the switch over?

A minor issue here is that we have cards with ethernet that can drive EEMs (Stabilizer, Humpback), but can't be ARTIQ core devices (or at least can't be in the DRTIO tree).  Until those use-cases are better developed though, it's not really a detail someone new to Sinara needs to know.

marmeladapk, 2020-10-12:
@dtcallcock Everyone but Greg is mentioned in alphabetical order.

> FPGA Controller(eg. Kasli)

Missing period (e.g.?)

You used an out of date WUT logo, which is illegible here. [Here's the new one (we have it for ~2 years)](https://promocja.pw.edu.pl/content/download/1458/8446/file/wersja-EN.zip). I can also supply it in vector form.

Please check if this CTI logo has higher quality than the one used.
[image]

> Two commercial vendors (Creotech and TechnoSystem) can supply bare tested boards

Maybe "can supply tested modules"? Bare tested boards sounds like the components are not populated.

> uTCA

What do you think about mentioning Sayma and Metlino here? v2 hardware is ready

>  Supported as part of SUServo

Maybe expand this to Sampler-Urukul Servo? I don't think that anyone would guess that from the short name.

> Booster Remote control

Extra space here before word "Remote".

dtcallcock, 2020-10-12:
> which is illegible here.

Sorry, I should have mentioned that this is a low res version to save the monster download.  Submitted poter will be full res.

> I can also supply it in vector form.

That's always best.  Please email or upload here @marmeladapk .

> Everyone but Greg is mentioned in alphabetical order.

It is typical in the AMO field to put the big boss PI last.  If people prefer strictly alphabetical I can do that (@gkasprow ?).  Perhaps as more people start presenting Sinara work at conferences, some of these conventions can live in a guide on the wiki so we don't have to discuss every time?

Btw, I would have put myself first even if I wasn't first alphabetically.  I find not having the presenting author at a conference first to be really confusing for other attendees (and the organizers!).

gkasprow, 2020-10-12:
I don't have a preference here.

sbourdeauducq, 2020-10-13:
> And they are also called 'carriers' in a lot of the early docs.

OK. Let's call it "carrier" then. Also this will solve the problem with the VHDCI carrier, which isn't really a motherboard or a controller, and hasn't been called any other name so far.

> DI/OT calls them 'system boards'

Too vague.

> and describes them as 'crate controllers'.

As I mentioned "controller" is also the ARTIQ software thing.

> You used an out of date WUT logo, which is illegible here

Should it be the WUT or the ISE logo anyway?

dtcallcock, 2020-10-13:
> Let's call it "carrier" then.

'Carrier' or 'FPGA Carrier'?  

I don't particularly like it because I thought it came from those boards that 'carry' FMC cards which doesn't seem relevant to Kasli.  But if it's unique and it doesn't confuse people then I suppose it works.

sbourdeauducq, 2020-10-13:
No "FPGA". One could also have a Zynq carrier, a VHDCI carrier, a MCU carrier (Humpback), an x86 carrier (for crunching numbers faster than Zynq), etc.

sbourdeauducq, 2020-10-13:
> I don't particularly like it because I thought it came from those boards that 'carry' FMC cards which doesn't seem relevant to Kasli.

Kasli "carries" EEMs, even if not in the literal sense of the term. And the EEM standard, electrically, is very much a simpler FMC.

dtcallcock, 2020-10-13:
Thanks all for the feedback. Here's the final version:

[ARTIQ_Sinara_final_small.pdf](https://github.com/sinara-hw/meta/files/5368750/ARTIQ_Sinara_final_small.pdf)

I'll let you know if it generates any interest and what kind of questions I get.

>but the "over 50" feels a bit high

Well I left it in the end as it's in the published abstract.  If anyone doubts us I'll just reel out random Russian lake names until they go away.

> The picture of the crate is somewhat dating now, but I don't think we can refresh it before the deadline.

Ditto on the Artiq screenshot (Artiq 2.0??). Hopefully this will be a bit of a living document and can be steadily improved.  Anyone want to take it to any of the other 50 online quantum conferences that have sprung up this year?

gkasprow, 2020-10-13:
> Should it be the WUT or the ISE logo anyway?

we use the WUT logo on all schematics, let's be consistent here.

marmeladapk, 2020-10-13:
@dtcallcock 

> It is typical in the AMO field to put the big boss PI last.

Ok, good to know.

>  Please email or upload here

I sent you .eps and .ai to dallcock@uoregon.edu

Something happened in summary of Kasli: "Clock recovery and distribution" - last three letters are shifted (unless it's an intended pun).

dtcallcock, 2020-10-14:
Interesting talk from Analog Devices.  Apparently they are aware that the gap between precision DACs (ie. Fastino) which are a bit too slow, and fast DACs (ie. Shuttler) which don't have great noisy/drift/accuracy is pretty annoying for us.  They plan on releasing a new DAC early next year that addresses this.  So that's exciting.

dhslichter, 2020-10-14:
> Ditto on the Artiq screenshot (Artiq 2.0??).

Yes, that screenshot is ancient!  Time for a new one....

sbourdeauducq, 2020-10-14:
That's actually ARTIQ 1.2, but the GUI has not changed much since. There would be [ndscan](https://github.com/OxfordIonTrapGroup/ndscan) but it's not finished/merged yet.

The hardware has better front panels, new modules, and significantly different versions of modules such as Kasli 2.0.

closed 2020-10-16 by dtcallcock