sinara-hw/meta #59: Sinara Poster

dtcallcock, 2020-10-16:
[Final full resolution poster (dropbox)](https://www.dropbox.com/s/bcz667b8qvowla3/ARTIQ_Sinara_final.pdf?dl=0) from [IEEE International Conference on Quantum Computing and Engineering (QCE20)](https://github.com/sinara-hw/meta/issues/57).

[Low resolution version (GitHub)](https://github.com/sinara-hw/meta/files/5368750/ARTIQ_Sinara_final_small.pdf)

Creating an issue to track mistakes and additions as people spot them.  Hopefully this poster can live on to go to other conferences and be a useful resource to give people new to Sinara until there is a real paper.

Things to fix for next time:

- [ ] Add new crate shot with recent front panels and Kasli 2.0
- [ ] Add new Artiq screenshot
- [ ] Creotech and TechnoSytem logos aren't vectorized
- [ ] Something happened in summary of Kasli: "Clock recovery and distribution" - last three letters are shifted
- [ ] Kasli-SoC is a 'controller', not a 'carrier'
- [ ] Add link to CERN DI/OT project

dtcallcock, 2020-10-16:
- [ ] Who's logo should actually go on here?  Presumably people who have funded/contributed some relevant hardware/gateware/software.  Should we make it clear that that's why they are on there and this isn't just a list of people who bought crates.  Is anyone missing (UW-Madison, the new Oxford Hub @cjbe, ...)?

sbourdeauducq, 2020-10-17:
Yes, the logos on the poster should be updated. https://m-labs.hk/experiment-control/funding/ would be a good reference.

dtcallcock, 2021-07-19:
[QCE21](https://qce.quantum.ieee.org) poster deadline is Aug 2.  Considering taking this again this year.  Anyone else going?

dtcallcock, 2021-07-20:
[EQTC](https://www.eqtc.org) deadline is also Aug 2.  I will not attend as middle-of-the-night virtual conferences aren't my thing but could be a good opportunity to promote Sinara.

dhslichter, 2021-07-20:
I will probably attend virtual Quantum Week, would be game to help present a poster if desired.

gkasprow, 2021-07-20:
I will talk with my students :)

gkasprow, 2021-07-21:
OK, Dorota (our student) already submitted an abstract related to Sinara. Once she comes back from the holidays she will prepare the poster and consult it with the community.

gkasprow, 2021-07-21:
Btw, here is one SInara related paper
https://indico.cern.ch/event/981823/contributions/4295432/attachments/2250078/3816845/poster_445.pdf

sbourdeauducq, 2021-07-21:
Why did you move to the Apex PA95 ICs? In addition to being expensive, they definitely do not have 1MHz bandwidth as advertised on the poster.

dtcallcock, 2021-07-22:
> she will prepare the poster and consult it with the community.

Great!  The poster linked above was made in Adobe Illustrator so she can load the pdf in that and edit or copy things from it should she want to.

jordens, 2021-07-22:
I got contacted regarding a ARTIQ/Sinara talk at the "Advancing the performance of engineered trapped-ion quantum systems" workshop. I'd like some suggestions and nominations/volunteers for that.

dhslichter, 2021-07-22:
@jordens some of the ARTIQ day presenters, perhaps?

DorotaNowicka, 2021-08-09:
Hi everyone! In agreement with Greg Kasprowicz, I submitted a poster about Sayma and Phaser to the QCE21: https://www.dropbox.com/s/8v5iy0ttmifsusk/posterQCE21v2.pdf?dl=0
The deadline for all updates is Sep 6.
I will be grateful for any comments and suggestions, especially please make sure that all authors are listed.

gkasprow, 2021-08-09:
@DorotaNowicka we have much better photos of Sayma AWG. Please ask @marmeladapk about them.

gkasprow, 2021-08-09:
>Why did you move to the Apex PA95 ICs? In addition to being expensive, they definitely do not have 1MHz bandwidth as advertised on the poster.

@sbourdeauducq well, they have a small signal BW of 1MHz, I measured it personally. To obtain 1MHz you need a pre-driver that generates +/-20V due to GBP limitation. Then you are limited by SR. I moved to Apex because the discrete amplifier had serious issues and was unable to work stable for more than one channel due to very poor PSRR (<0dB). To make them work you need a dedicated pair of HV LDOs for every channel!
 The APEX chips are not that expensive if you buy directly at the source.

dtcallcock, 2021-08-09:
Thanks @DorotaNowicka. I made comments in Dropbox.  

I think one thing that's missing is a brief higher-level discussion of why there are two SAWG cards. I think the form factor is a big part of it and is unclear on the poster at the moment.  uTCA provides more board space, cooling, power, management, and data bandwidth, but comes with a big complexity and cost overhead.

sbourdeauducq, 2021-08-09:
> they have a small signal BW of 1MHz, I measured it personally. To obtain 1MHz you need a pre-driver that generates +/-20V due to GBP limitation. Then you are limited by SR.

By just saying "bandwidth" it makes it sound like you could have a 400Vpp sine wave output at, say, 10Hz and then by increasing the input frequency to 1MHz and keeping the input amplitude constant you'd have a 280Vpp (3dB) sine wave output. This is not what the Apex chip does. According to your own test, instead you get 13Vpp and a lot of distortion:
[image]

gkasprow, 2021-08-16:
@sbourdeauducq this is exactly how BW in opamps is defined. Take any opamp datasheet and see how it's defined.
I didn't write "full power" bandwidth. The amp full power BW is limited both by SR and small-signal BW.
SR is limited by the load capacitance.
So I will simply add SR parameter .
The HV amp module has ability to replace amps with ones having higher SR and much higher power BW. One would need to provide external HV due to higher Iq

gkasprow, 2021-08-16:
I updated the wiki

DorotaNowicka, 2021-08-23:
Thanks for the comments @dtcallcock . I made all the corrections  (if I missed any, let me know, please!): https://www.dropbox.com/s/hs63bxkq5hq8opq/posterQCE21v3.pdf?dl=0
The only thing I haven't changed is JESB204B and JESD204B in Sayma clock generation block, now it's on a poster like on the Wiki page. Is it wrong?

marmeladapk, 2021-08-24:
@DorotaNowicka hey, I added some comments to the second version.