sinara-hw/meta #62: AD9166 Vector Signal Generator

dtcallcock, 2020-12-17:
Cool new silicon: https://www.analog.com/media/en/technical-documentation/data-sheets/ad9166.pdf

Not dug into it too much, but could offer a one-chip solution for much of the Phaser functionality.

dtcallcock, 2020-12-17:
You make a good case for why it probably wouldn't make sense to redesign Phaser around this chip.  Moving issue to Meta and leaving open though is it clearly has unique capabilities relevant to Sinara.

dhslichter, 2020-12-23:
I would say that this chip is how one would build ARTIQ hardware for superconducting qubit applications, if not using an RFSoC.  If one made a Sayma RTM targeted to sc qubits, that could be interesting.  A major consideration with sc qubits, though, is round-trip latency, which is generally problematic with JESD.  The datasheet is also a bit cagey about the ability to achieve true deterministic latency, which is essential (what part of "within several device clock cycles" is "deterministic"?).  
[image]

dtcallcock, 2020-12-23:
Without digging too deeply into capabilities, at $4k/40W for 8-channels, how does it compare to the ZU27DR you are using on RFSoC-AMC @gkasprow? 

> A major consideration with sc qubits, though, is round-trip latency, which is generally problematic with JESD.

Yes, I understood that JESD was a no-go for SC qubits.

gkasprow, 2021-01-23:
RFSOC is 10k for 8 channels. They upgraded it to 10GS/s recently.

gkasprow, 2022-11-09:
We have a use case for this chip in the ion trap community. Just a better version of the Phaser with much lower phase noise.
The plan is to buy a devkit, plug it into CERN DIOT 3U Kintex carrier and give it a try.

gkasprow, 2022-11-13:
I think AD9164 would be better. It offers far better phase noise.
Correct me if I'm wrong, but the only difference I found between AD9164 and AD9166 is an embedded wideband amplifier with single-ended output.
AD9166 is advertised as a vector signal generator, but the AD916xx family has exactly the same digital data paths, including NCO and IQ  upconverter.

gkasprow, 2022-12-09:
I ordered a devkit, and my student will compare the phase noise and other parameters of the AD chip and the RFSOC.