sinara-hw/meta #63: Circuitmaker

sbourdeauducq, 2021-01-23:
Installing and running (instructions compiled from various online sources):
```shell
nix-shell -p wine winetricks
WINEARCH=win32 WINEPREFIX=/home/sb/.wine-circuitmaker winetricks vd=1920x1080 -q gdiplus corefonts riched20 mdac28 msxml6 dotnet40
WINEPREFIX=/home/sb/.wine-circuitmaker winecfg
# go to the "Libraries" tab, find "msado15", click "Add", make sure it's marked as "native, builtin"
WINEPREFIX=~/.wine-circuitmaker wine ~/.wine-circuitmaker/drive_c/Program\ Files/Altium/CM/DXP.EXE
```

* tested on NixOS 20.09.2654.646f528cce6
* vd=1920x1080 works around the orange rectangle masking the login dialog box during installation
* the msado15 override works around the access violation in IntegratedLibrary.DLL during Circuitmaker startup

This makes Circuitmaker work as intended without requiring Windoze, and it can open schematic files faster and with fewer limitations than the Altium online viewer.

However:
* the biggest problem is it cannot read Altium Designer PCB files at all, unless they are exported by Altium Designer into older formats (see e.g. https://circuitmaker.com/forum/posts/211895).
* It seems that it can only save schematics and PCBs into the Altium cloud servers. I don't know how easy it is to spoof the server and if the format is compatible with Altium Designer.

sbourdeauducq, 2021-01-23:
Circuitstudio also works in Wine and is marginally less bad - it can save its design files on your computer. But it still cannot read Altium Designer PCB files.

jordens, 2021-01-23:
Very cool!
Another useful tool would be a little script to generate viewer html pages for all projects and `releases` en masse with gha and a corresponding

sbourdeauducq, 2021-01-23:
> Another useful tool would be a little script to generate viewer html pages

The Altium Viewer links expire after 48 hours, and saving the HTML page does not work easily (and likely has licensing issues).

jordens, 2021-01-23:
No. The iframe pages work forever and they explicitly are meant for this. I talked to them.

sbourdeauducq, 2021-01-23:
Kicad import has some problems but looks promising
[image]

gkasprow, 2021-01-23:
The Circuitstudio is a free tool with many limiatations. They use same libs and sch format but PCB is different. Essentially, the content is still the same but the file container keeps it in differently named directories :)
There is another,[ low cost Altium Circuit Studio ](https://octopart.com/search?q=CircuitStudio&currency=USD&specs=0)tool that seem to not have that many limitations. AFAIK it supports native PCB files. I played with it once it was introduced and AFAIR there was an option to import PcbDoc files from Altium designer.
CERN has KICAD version of their Alitium libraries, so once you import Altium design to Kicad, can substitute libraries with centrally managed ones. But as I wrote, Kicad is far less productive than Altium (I'd compare it with Circuit Studio) and in case of complex boards it's cheaper to buy Altium license than to pay extra money to engineers :)