sinara-hw/meta #64: EMC / EMI / Crate grounding

kaolpr, 2021-02-26:
What is current strategy for EMC / EMI and crate grounding for Sinara?

I believe the crate should be connected to PE. Panels must be a electrically connected with a crate, so they are an element of the shielding. Then all modules' connectors should be isolated from the panel and panel should be connected to modules GND with RC circuit. This is fulfilled in case of *some* of them (e.g. Urukul or Sampler). 

However, e.g. Kasli has no connection to panel mounting holes, but SFP cage and USB connector are connected directly to GND. Similarly with Fastino and Zotino where there is no isolation between connector and panel.

gkasprow, 2021-02-26:
SFPs have no internal connection to the chassis, so we can isolate them from Kasli GND
The same with USB, there is usually no electrical connection between USB shield and GND, at least in the cable.

pathfinder49, 2021-03-01:
I have spent some time investigating grounding configurations for combinations of Kasli and EEMs. Personally the most reliable grounding configuration for a kasli crate (given the front panels are connected to board ground) is as follows:
- Make sure the kasli-crate is isolated from PE. If you mount your crate in a Rack with an earth strap, you can use [Humfrees](https://uk.farnell.com/adam-hall-hardware/5635blk/humfrees-rack-washers/dp/3401048) to break the electrical connection.
- Make sure there is a low impedance path to your reference ground directly from the kasli-crate. You should **not** rely on the signal cables for this. I've seen significant EMI improvements by using a ground strap directly to the reference ground.
- Power kasli from a class II DC supply (like the ALM65US12C2-8). Above a few 100 kHz there is capacitive coupling across the transformer in the supply. If you are sensitive in this frequency range, I would highly recommend using a common mode current choke in series with the class II supply.
- Use common mode current chokes on the signal cables from the kasli-crate. Depending on your frequency range and sensitivity, ferrites can work well here. The key is to make the common mode impedance large compared to the ground strap. (Of cause, you should still try to keep the loop area with the ground strap small)

As you mention Fastino, I presume you've already seen the recommendations from https://github.com/sinara-hw/Fastino/issues/56. The measurements there are largely what inspired the configuration above.

pathfinder49, 2021-03-01:
> I believe the crate should be connected to PE

Given that existing EEMs have their front panel connected to the board ground, I think it could lead to headaches trying to switch now. If you have a mixture of board ground and PE front panels in the same crate, you will bypass the 12 V input filter on kasli. This would also allow (noisy) PE to couple into the board round and by extension the EEM outputs.

pathfinder49, 2021-03-01:
@gkasprow If I could get editing access, I'm happy to contribute an EMI/EMC/grounding section to the wiki (probably [here](https://github.com/sinara-hw/meta/wiki/EEM))

gkasprow, 2021-03-01:
done

kaolpr, 2021-12-16:
I don't feel like we've achieved a broad agreement on how to deal with this issue. Though @pathfinder49 solution is well considered, it looks a bit like a workaround of project shortcomings. Question is - do we want to fix them (trying not to make things harder as we go) or we're happy with the current state?

gkasprow, 2021-12-17:
I think we should do the homework and sketch the EMI source diagram to know where the real issue is, what are the ground loops, which CMCs help, and which harm.
I hope I will finally get my own SSA which will help identify the issues. The procurement started in March and they published it today :D