sinara-hw/meta #65: Wavemeter EEM

sbourdeauducq, 2021-05-14:
During the ARTIQ day, there was a short discussion about the possible development of a laser wavemeter EEM. Compared to competing solutions, it could be more open, not require Windows, have nicer firmware features (e.g. MQTT), be better integrated with the EEM ecosystem (form factor and direct connection to e.g. Stabilizer), and hopefully cost less.

The most popular wavemeter technology today is the Fizeau-Snyder type, which is compact and fast, works with pulsed lasers and has no moving parts, but unfortunately requires custom high-precision optomechanics which are difficult to prototype, manufacture and characterize, and cost a lot of money. Depending on the required accuracy, they might also require vacuum and/or temperature stabilization.

The operating principle of these wavemeters is described in these old papers:
* http://www.mathnet.ru/php/archive.phtml?wshow=paper&jrnid=qe&paperid=10622&option_lang=eng (in Russian)
* http://www.mathnet.ru/php/archive.phtml?wshow=paper&jrnid=qe&paperid=12199&option_lang=eng

These are also relevant:
* https://core.ac.uk/download/pdf/42821983.pdf
* https://lightmachinery.com/optical-design-center/more-optical-design-tools/fizeau-designer/
* https://lightmachinery.com/optical-design-center/library/users-guides/fizeau-designer-users-guide/ and the references, in particular https://sci-hub.st/10.1063/1.3525102
* https://sci-hub.st/10.1364/OL.6.000610
* https://sci-hub.st/10.1007/3-540-51558-5_12

Potential options for the optical block:
* buy it from [Moglabs](https://www.moglabs.com/fizeau-wavemeter) (if possible) and integrate it with our electronics.
* design it ourselves and manufacture it e.g. at [LightMachinery](https://lightmachinery.com/). Obviously this is more risky and requires a larger investment.
* http://dple.by might also make them.

sbourdeauducq, 2021-05-14:
I realized the second link is also in Russian. Here's a translation:
[image]
[image]

gkasprow, 2021-05-14:
The question is: would there be a market for such a device? Let's assume we would produce it in low quantity, tens of pieces, so the price would be high. Probably similar to the COTS one.
This is a measurement device, people expect good accuracy and reliability. Would they pay for the open-source design only because it's open? The money is not a big deal, if there is a business opportunity :)

dtcallcock, 2021-05-14:
If you are happy with 200MHz then what's wrong with just using the MOGLabs one? It has an Ethernet port and doesn't require host software so that would appear to eliminate the main pain point with the High Finesse ones. MQTT sounds like the kind of thing they might consider adding of it is important to you. If you want a different form factor and stabilizer integration they will sell you an OEM version with SPI interface. IME they are very reasonable people to work with.

dtcallcock, 2021-05-14:
If you want a 1MHz wavemeter that you can lock your lasers to then that's a different kettle of fish. You have to ask why no competition to the WS8 has come out in 10 years. This suggests to me that either it is very hard or the market is not large enough for a competitor. Hard to imagine the R&D alone costing less than a few $M if you are not already set up to do similar things. 

One idea though. The high finesse wavemeters are electronically very simple I think. Basically a USB webcam and a temperature controller. Would it be possible to reverse engineer and write host software that doesn't suck? This has the advantage that everyone already has the optomech installed in their lab!

sbourdeauducq, 2021-05-14:
> Would it be possible to reverse engineer and write host software that doesn't suck?

I think so, though the reverse engineering work would still require access to one of these beasts and probably other expensive things like tunable single-frequency lasers.

hartytp, 2021-05-14:
> Would it be possible to reverse engineer and write host software that doesn't suck?

HF seem to have to compile different versions of the driver for each wavemeter they ship. I assume the hardware changes pretty frequently in ways that the host side needs to know about. IME writing code for one wavemeter does not mean that it will work for all wavemeters. This feels like a rabbit hole. Also, the cost of doing that development is probably more than buying every used a windows box...

dtcallcock, 2021-05-14:
I didn't say it was a good idea!

> probably other expensive things like tunable single-frequency lasers.

The provided software will let you export the raw interferometer data.  So any old lab could send you data and you could write the analysis routines offline.   

You will need hardware for the USB driver bit though and as @hartytp points out, there appears to have been quite a few under-the-hood changes over the years. 

Anyway, don't you have a formal, if tenuous, connection via M-Labs->QUARTIQ->Opticlock->Toptica->High Finesse.  Given how much everyone complains about their software, you'd think they'd be open to someone helping them make it better.  If nothing else, perhaps they would be willing to provide some super-pared down binary windows/linux driver that avoids all the GUI and cruft?  Probably a long shot but if you don't ask...