sinara-hw/meta #66: low cost EEM - FMC carrier

gkasprow, 2021-05-22:
There seem to be a need to have a simple, low-cost native EEM FMC carrier.
Essentially, a Phaser with an analog part replaced by the FMC connector.
The first application would be Shuttler and Stamper. We also have a number of interesting open-source boards - like a DDS generator, high-speed ADCS, delay modules, etc. One can also use many dev boards from ADI/Texas.
We also have a CameraLink input/output board, analog video card (HD-CVI, HD-TVI), 32 channel TTL IO, etc.
The [CERN Peripheral board ](https://ohwr.org/project/diot-pfc-ku/wikis/home)with Ultra Scale FPGA  is an overkill for many applications and it would be expensive at the beginning until we hit the step pricing.

The initial spec:
- Artix 100T (same as in Kasli/Phaser)
- FMC HPC connector
- VADJ fixed at 2.5V
- one MGT routed to the FMC 
- one MGT routed to CPCIS adapter connector, 
- two MGTs routed to SATA (or to FMC)
- 2x EEM connector
- no SDRAM (we need 4 banks to route HPC pins)
- power supply as in Phaser
- front panel with FMC cutout and maybe USB or SFP if that makes sense

I have funding so can proceed when there is enough interest.

cjbe, 2021-05-22:
This would definitely be useful to me!

Most of my applications require several MGTs to the FMC, so not full HPC-FMC, but more than LPC-FMC, hence my preference would be connecting to the FMC over a SATA connector.

As an explicit useful, I am interested in connecting up a 2 lane Coaxpress interface, using (for example) [this](https://kayainstruments.com/fmc-ii-coaxpress-12g-card/) FMC

gkasprow, 2021-05-23:
The question is if such an approach would be much more expensive than a dedicated EEM board.
The cost of the FMC connector and additional PCB area would be neglected by a much higher volume of the carrier board than yet another version of DAC for shuttling. So I think the cost of such a solution would be similar providing that we use the carrier board in more than just one project.

gkasprow, 2021-05-23:
@dhslichter

gkasprow, 2021-05-23:
> As an explicit useful, I am interested in connecting up a 2 lane Coaxpress interface, using (for example) this FMC

I would give the assembly options as we do in the case of MTCA FMC carriers.
You can connect all MGTs to FMC or some of them to other connectors

gkasprow, 2021-05-23:
Actually, we can keep the SDRAM. We have enough pins.

jordens, 2021-05-23:
Keeping that bit of sdram would be very valuable.
And if possible make it so that the bank that's unavailable on the smaller fpgas in that same package is used for either sdram or the HPC pins something else that can be isolated functionally.

gkasprow, 2021-05-23:
this bank is too small for SDRAM.
It has just 17 LVDS lines.
Too small for HA or HB pins
Just enough to host half of LA bank within the clocking region
FMC standard is quite restrictive about CC/data signals placement

gkasprow, 2021-05-23:
Actually, we can split the HA and HB banks!
In such case when we install 50T FPGA, the first half of the HA bank wound be not connected
[image]
The Shuttler and Stamper were designed in a way that takes these constraints into account

dhslichter, 2021-05-25:
@gkasprow sorry was out of the office -- this carrier would be nice to have (as you know I have requested it previously!)  For Shuttler it obviously makes it easier to integrate immediately with existing EEM-based systems.  It would also be great to allow ARTIQ to take advantage of a lot of existing FMC cards of various sorts from CERN OHL or many other sources by making this piece of hardware "glue".  Obviously there is gateware to be written too, but hopefully that can be relatively independent of the hardware as long as the EEM FMC carrier board is designed correctly.

dhslichter, 2021-05-25:
Is there a reason to have VADJ be fixed at 2.5 V?  It would enable more flexibility (potentially) to have this be able to be set to 3.3 V or 1.8 V for some boards, no?  Is the complexity associated with this too high?

marmeladapk, 2021-05-26:
Artix FPGAs support LVDS only on banks with 2.5V supply.

dhslichter, 2021-05-26:
But some FMC cards may use LVCMOS signals rather than LVDS.  Those signals may be 3.3 V LVCMOS or 1.8 V LVCMOS, depending on the chips.

dhslichter, 2021-05-26:
> Artix 100T (same as in Kasli/Phaser) - HPC version
Artix 50T - LPC version (some HA pins not connected)

These are cheap FPGAs, the price difference is less than $100 between them at retail price (and less with Xilinx volume discount).  Would it not make sense to just use the larger FPGA on both boards?  I/O is not the only consideration, it may also be useful to have more FPGA resources (as in the 100T) even in the case of an LPC FMC.  Also, using just one FPGA variant means faster to get to lower step pricing.

gkasprow, 2021-06-11:
Yes, we can mount 100T as default.
All our open source FMC boards work with 1.8 and 2.5V LVCMOS signaling.
Some Xilinx devkits have VADJ fixed at 2.5V so the majority of FMC boards should work with such voltage.

gkasprow, 2021-06-12:
However, we have 1.8V and 3.3V rails anyway so we can simply install a jumper that would select the desired voltage.
All Artix banks work with 3.3V (they are HR-type)
However, one of the banks is used for both FMC and EEM signals which means that HA signals must have a 2.5V supply.
Another bank is used also for configuration, which means we have to add a translator.

gkasprow, 2021-06-12:
Please list potential FMC boards that could be used with this carrier. Many of them are not 100% FMC compliant

dhslichter, 2021-06-14:
As far as I am concerned, Shuttler is the only target FMC board I currently care about.  So if 3.3V VADJ compatibility is a problem, let's not worry about it.

jordens, 2021-06-14:
@gkasprow 

* Your TDC/ADC FMC
* FmcAdc100M14b4cha
* Some coaxpress mezzanine with maximum compatibility in terms of https
* maybe one of those 12 gs/s DAC fmcs If that can be made to work.

dhslichter, 2021-06-15:
If we get the -100T in a larger package we should be able to route more GTPs to the FMC.

gkasprow, 2021-06-17:
> If a new FMC needs to be designed that requires 3.3V logic for some chips, we can just put the translators on the FMC card.

This is what I do in case of new cards. All new FPGA families starting from Kintex have LVDS in HPC banks. So they are limited to 1.8V anyway. So the translators are a must.

dhslichter, 2021-06-18:
@gkasprow what is the timescale for implementing this board?  We will need them for Shuttler and so are interested in having the design done ASAP.  What other hardware questions need to be answered at present?

gkasprow, 2021-06-18:
The shuttler can be tested on any similar Artix board like AFC. Filip is finishing new revision of AFCK, his next task is this board. He starts in 2 weeks. It will probably take him 2 months to do schematics and layout (we have holidays).

dhslichter, 2021-06-18:
The Shuttler smoke tests can be done on any similar board; my main question is about being able to buy carrier boards to run things in the lab.  Things like thermal drifts, noise performance, etc are important but can't really be characterized properly until we are running in the designated environment.  For example, to what extent do we need/want temperature stabilization, how bad is electrical noise from Kasli SMPS, etc.  

Anyway, end of August for schematics and layout seems reasonable, but I just wanted to know soon so I can plan.

gkasprow, 2021-06-24:
@marmeladapk is assigned to this project.

marmeladapk, 2021-07-07:
Please check https://github.com/sinara-hw/EEM_FMC_Carrier/discussions, I'll make a few threads there to talk about schematics.

closed 2021-07-07 by marmeladapk