sinara-hw/meta #69: DC-coupled AWG

gkasprow, 2021-10-16:
I have already a few use cases for general purpose, low-cost AWG.
One could use Phaser for that purpose, but most of these applications require DC coupling and output swing of +/-5V or so.
Essentially, we need to replace top-bench AWG with modules that are more tightly integrated with ARTIQ.
I see two approaches:
- take Phaser, get rid of the upconverter, RF amplifier, and add a high-speed DC-coupled amplifier at its output.
- use the FMC approach, just place the DAC (one used in Phaser) and amplifiers. It seems more feasible because that approach works both in MTCA and EEM, as well as on devkits. It would be a lower cost. Another approach is the possibility to use it with higher-grade FPGAs when higher DAC speed is needed (ZynQ US CPCIS carrier).

I'd like to hear some thoughts and possible use-cases.

dtcallcock, 2021-10-17:
I like the idea of keeping momentum behind Phaser development instead of fragmenting it.

jordens, 2021-10-17:
FMC

dhslichter, 2021-10-19:
Another vote for FMC.