sinara-hw/meta #70: IDC-MCX

jordens, 2021-11-03:
This is a proposal for a very simple new break-out board: analog 2xIDC to 16xMCX.
Electrically similar to IDC-BNC but mechanics similar to [DIO-MCX](https://github.com/sinara-hw/DIO_MCX/).
For Fastino and Zotino to enable higher density setups.
Whether the resulting cable density is desirable is another question.

gkasprow, 2021-11-03:
yes, connecting all 32 channels of Zotino to BNCs consumes a lot of panel space.

gkasprow, 2021-11-03:
the easiest way would be to use a single PCB and vertical MCX connector. But the EMC lips and RC filters won't be possible here
[image]
Another approach is dual-level PCB like in DIO-MCX, slightly more expensive but without compromises

jordens, 2021-11-03:
Cool idea. I don't think we really need the CMC and RC filters on there. But the EMC lip would be desirable.
If you do a single 8 HP (or 12 HP) panel for 32 channels, can it be made to work with a single PCB?

gkasprow, 2021-11-03:
32 channels should fit on 8HP with EMC lip, but it could be difficult to unplug some connectors selectively using standard fingers :)

gkasprow, 2021-11-03:
that could work
[image]
[image]

gkasprow, 2021-11-03:
to include the CMCs we would need to go for 12HP or dual PCB approach

gkasprow, 2021-11-03:
on the other side, what is the benefit of having a 12HP solution since 4x SMA gives 16HP...

gkasprow, 2021-11-03:
we can also think of an 8HP mezzanine sitting on top of Zotin/Fastino which could work without IDC cables at all.

jordens, 2021-11-03:
Let's leave the CMC s out. We removed them from fasting and from the other IDC boards already.
If you can.make it work without the IDC cables, even better. I lack the Imagination what that would look like with a single PCB. ;)

gkasprow, 2021-11-03:
well, I didn't say it would be a single PCB :)

gkasprow, 2021-11-07:
This is the idea of the Zotino/Fastino mezzanine
[image]

One can also solder in the IDC connectors and CMC filters and use it stand-alone
[image]

panel
[image]