sinara-hw/meta #71: open source HW for SC quantum

gkasprow, 2021-11-17:
https://aip.scitation.org/doi/10.1063/5.0055906
https://gitlab.com/LBL-QubiC/rfmixingmodules
https://github.com/openquantumhardware/qick

dtcallcock, 2021-11-17:
The QICK thing is described in a paper too: https://arxiv.org/abs/2110.00557

We got a shout out.  Wish we had a better citation than M-Labs website so we could get out citation point (=prizes!).

>The ion trap community has developed the FPGA-based ARTIQ system [19] for control of their systems,
albeit on slower timescales than those required for solid-state qubits, which typically have faster gate speeds (and decoher- ence rates).

jordens, 2021-11-17:
From a business perspective papers just don't have good ROI.

dtcallcock, 2021-11-17:
Potential customers read the arXiv.

jordens, 2021-11-17:
Potential customers also watch cat videos and we're not making those.
As you rightly note citation points gets you prizes, not sales.

dhslichter, 2021-11-17:
@dtcallcock I think if you want this paper, you are going to have to do the heavy lifting.  I imagine that @jordens and @sbourdeauducq could be convinced to provide some technical content in specific areas where it is relevant, but all the other boilerplate has to be handled by someone else.  Frankly I'd rather have them working on contracts I have with them rather than paper writing :)

gkasprow, 2021-11-17:
Today I had a chat with the Creotech guys. They are writing a lot of reports for the funding agencies. Just translate, make it short and publish.

sbourdeauducq, 2021-11-18:
> Wish we had a better citation than M-Labs website

M-Labs is the new Elsevier.

We have some preliminary code to run bare metal Rust on Ultrascale+/RFSoC. Can develop further and/or port ARTIQ if there is interest.

gkasprow, 2021-11-18:
We are finishing the design of the stand-alone RFSOC AWG + we already have a native CPCIS UltraScale carrier board manufactured. So there is an interest, at least from polish side :)