sinara-hw/meta #72: Sinara CE certification

nkrackow, 2022-05-03:
@gkasprow @akaminska As discussed recently, we would like to get going with the CE certification/EMC testing. Did you start with thinking about how to coordinate this at Creotech yet? Like which representative system to test etc. https://github.com/sinara-hw/meta/issues/47

nkrackow, 2022-06-03:
@gkasprow @akaminska Ping ;)

gkasprow, 2024-07-19:
We did a lot of EMI/EMC testing in various configurations
So far only the Booster experienced issues with ESD and was fixed. Existing units can be easily fixed.
Various configurations of Kasli and EEM modules passed EMC tests without issues.

gkasprow, 2025-07-11:
We finally finished CE certification of Sinara HW. A lot of details matter- how you route the cables in the crate, how you assemble the crate, mount the grounding tabs, insert the ferrite beads etc. We did a few iterations to pass the EMC tests in the certified lab. There is no simple rule to follow, one needs to think about it as certification of the assembly process, not module production.
It means there is no way we can certify EEM modules. The entire crate needs It. We tested several popular configurations. It was much more difficult than we previously thought. I believe DIOT approach will make it far easier.