sinara-hw/meta #74: EEM isolator

gkasprow, 2022-07-31:
ADI released a high-speed [LVDS isolator.](https://www.analog.com/en/products/adn4656.html?ADICID=EMAL_WW_P628000_MIX-NL-PN_1125&deliveryName=DM25392#product-overview)
So, if there is an interest in isolating the EEM board from Kasli, there is a potential way to do this.
Of course, this adds some extra delay, but much lower than another Kasli.

dtcallcock, 2022-07-31:
How would this work?  Existing [VHDCI_Carrier](https://github.com/sinara-hw/VHDCI_Carrier/wiki) in the Kasli crate and then this new EEM in the satellite crate (where the new EEM would have a VHDCI front panel connector, 12V power barrels, and 4 downstream EEM connectors).  Or something else?

gkasprow, 2022-07-31:
An adapter that you plug between Kasli and the EEM board. It could also be integrated as a DIOT (CPCIS) to EEM adapter.