sinara-hw/meta #76: [RFC] Zotino/Fastino break out

jordens, 2022-12-21:
In most cases it is best to integrate the HD68 connector into the application and not use 32 BNC cables to connect Fastino/Zotino to the experiment. Yet people do it anyways at least for prototyping, debugging, or testing.
We've had the need a couple times already for a simple HD68-BNC break-out. The EDGE series of modules doesn't have a HD68 connection (that might be one solution). The HD68-IDC+4xIDC-BNC+front-panels+case is surprisingly costly. Overall it's probably much easier to make a straight 1xHD68 to 32xBNC break out with minimal mechanical complexity:

* BNC connectors as in https://github.com/sinara-hw/Banker_EDGE_2_BNC
* A HD68 connector preferrably perpendicular to the board (for simpler mechanical design) like the BNC connectors
* No front panel, make the PCB the panel (as in the EDGE series).
* Find a simple case that the PCB can slide or be screwed into. It's only task is to cover the PCB underside. Or alternatively just mount the PCB onto a simple strengthener (preferrably COTS though and not custom).
* Use a standard board size that fits standard cases. Add common mounting holes.

@gkasprow What do you think?

gkasprow, 2022-12-21:
OK. I'm curious why all these people use these old large connectors instead of mcx :)

gkasprow, 2022-12-21:
There is already 32-channel [MCX adapter](https://github.com/sinara-hw/IDC_MCX_Adapter/wiki) for Fastino/Zotino that can be mounted on top of the Fastino board as a mezzanine

jordens, 2022-12-21:
The problem with the IDC-MCX is that it hasn't been manufactured yet (AFAIK), that is mechanically relatively complex (two boards and a link between) and that it (compared to this hypothetical HD68-BNC) requires a panel plus a crate plus IDC cables plus the HD68-IDC adapter: the goal is to use the HD68 cable to get away from the rack but then break out in proximity to the experiment.
I'd also prefer MCX but the dominant legacy cable is BNC or SMA. Maybe MCX/BNC as a variant?

jordens, 2022-12-21:
I'm not complaining about the IDC-MCX adapter at all, especially since I instigated it at least in part. Just trying to best respond to practical use cases.